volkszaehler/vzlogger #96: vzlogger.conf "timeout" parameter is unregarded in MeterD0

skre, 2015-01-02:
If I set my "timeout" to 30 the vzlooger still keeps the fixed 10s timeout

logfile:
[Jan 01 23:49:00][d0]   nothing received for more than 10 seconds
[Jan 01 23:49:00][d0]   read timed out!, context: 0, bytes read: 0, last byte 0x7f

mbehr1, 2015-01-12:
The timeout parameter is just for the local http
(  "local": {
        "enabled": false,   // should we start the local HTTPd for serving live readings?
        "port": 8080,       // the TCP port for the local HTTPd
        "index": true,      // should we provide a index listing of available channels if no UUID was requested?
        "timeout": 30,      // timeout for long polling comet requests, 0 disables comet, in seconds)

Anyhow I'd like to add a parameter for MeterD0 but I'd like to wait until #109 has been merged. Then I can check whether a timeout/delay for baud rate switching is needed as well and would add this at the same time.

mbehr1, 2015-01-17:
new configurable timeout added with PR#111.

mbehr1, 2015-02-14:
Issue can be closed, or?

closed 2015-02-16 by andig