volkszaehler/vzlogger #102: Fehler beim Cross-compilen unter OpenWRT

autohausnielsen, 2015-01-08:
Ich nutze den YPORT+ Node (mit Erweiterung) von Udo um meinen Stromzähler mit dem USB-Lesekopf und den Gaszähler mit S0 auszuwerten.
Der YPORT+ läuft unter OpenWrt, vzlogger übersetze ich mit der OpenWrt Buildroot. Das hat bisher problemlos funktioniert. Seit kurzem bricht der Übersetzungsprozess mit Fehlern ab.

## Compile Log:
>vzlogger.o: In function `config_parse_cli(int, char**, Config_Options*)':
vzlogger.cpp:(.text+0x394): undefined reference to `g_GIT_SHALONG'
threads.o: In function `logging_thread(void*)':
threads.cpp:(.text+0x6f8): undefined reference to `vz::api::Null::Null(std::tr1::shared_ptr<Channel>, std::list<Option, std::allocator<Option> >)'
MeterMap.o: In function `MeterMap::registration()':
MeterMap.cpp:(.text+0x362): undefined reference to `vz::api::Null::Null(std::tr1::shared_ptr<Channel>, std::list<Option, std::allocator<Option> >)'
collect2: error: ld returned 1 exit status
make[5]: *** [vzlogger] Error 1
make[5]: Leaving directory `/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-1d272282c0948c329d54fd705ab189ca60d4c4d3/src'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-1d272282c0948c329d54fd705ab189ca60d4c4d3'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-1d272282c0948c329d54fd705ab189ca60d4c4d3'
make[2]: *** [/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-1d272282c0948c329d54fd705ab189ca60d4c4d3/.built] Error 2
make[2]: Leaving directory `/home/dt/custom-feed/vzlogger'
make[1]: *** [package/feeds/custom/vzlogger/compile] Error 2
make[1]: Leaving directory `/home/dt/udoopenwrt'
make: *** [package/vzlogger/compile] Fehler 2

Es hängt anscheinend an folgenden Änderungen:
9a3a8c: Added null api for not pushing readings to any con...
3e59e4: add git version info

Das compilieren geht, wenn ich:
a) das null api in threads.cpp und MeterMap.cpp auskommentire 
b) in vzlogger.cpp das  # include gitSha1.h  und g_GIT_SHALONG auskommentiere

andig, 2015-01-08:
> threads.cpp:(.text+0x6f8): undefined reference tovz::api::Null::Null(std::tr1::shared_ptr, std::list >)'
> MeterMap.cpp:(.text+0x362): undefined reference tovz::api::Null::Null(std::tr1::shared_ptr, std::list >)'

Mit welchem Kommando startest Du das Build?

Viele Grüße,
Andreas

r00t-, 2015-01-08:
vermutlich benutzt du autoconf und nicht cmake?

=> wahrscheinlich am besten, wenn wir autoconf rauswerfen. parallel pflegen wird nicht lohnen.

r00t-, 2015-01-08:
(also, die neuen source-files fuer das neue zeug wurde in cmake eingepflegt, aber nicht in autoconf)

andig, 2015-01-08:
Gibts cmake für OpenWRT? Cmake als Voraussetzung ist schon länger dokumentiert...

r00t-, 2015-01-08:
es geht ja um cross-kompilieren FUER openwrt, nicht AUF openwrt.
die interessantere frage ist, wie man mit cmake cross-kompiliert.

andig, 2015-01-08:
So http://www.cmake.org/cmake/help/v3.0/manual/cmake-toolchains.7.html ?

autohausnielsen, 2015-01-08:
"cmake" ist in der Toolchain von OpenWrt vorhanden, siehe:  http://wiki.openwrt.org/about/toolchain

@andig: Den Buildprozess für vzlogger starte ich über "make package/vzlogger/compile V=s" auf einem Ubuntu 14.04 LTS. 

Komplettes Log

>dt@dt-VirtualBox:~/udoopenwrt$ make package/vzlogger/compile V=s 
Collecting package info: doneing...
make[1]: Entering directory `/home/dt/udoopenwrt'
make[2]: Entering directory `/home/dt/udoopenwrt/package/libs/toolchain'
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
WARNING: skipping libssp -- package not selected
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libstdcpp" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libthread-db" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
WARNING: skipping libgfortran -- package not selected
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgo" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
WARNING: skipping ldd -- package not selected
WARNING: skipping ldconfig -- package not selected
make[2]: Leaving directory `/home/dt/udoopenwrt/package/libs/toolchain'
make[2]: Entering directory `/home/dt/udoopenwrt/feeds/packages/libs/libmicrohttpd'
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libmicrohttpd.default.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libmicrohttpd.default.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libmicrohttpd.default.install.clean; fi; echo "libmicrohttpd" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libmicrohttpd.default.install
make[2]: Leaving directory `/home/dt/udoopenwrt/feeds/packages/libs/libmicrohttpd'
make[2]: Entering directory `/home/dt/udoopenwrt/package/libs/ncurses'
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "terminfo" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install
if [ -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "libncurses" >> /home/dt/udoopenwrt/staging_dir/target-mips_34kc_uClibc-0.9.3 [... cut ...]
make[3]: Leaving directory `/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-ffc6b5c02bd6b07865a37d092f9c47c6778cb696'
make[2]: *** [/home/dt/udoopenwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/vzlogger-ffc6b5c02bd6b07865a37d092f9c47c6778cb696/.built] Error 2
make[2]: Leaving directory `/home/dt/custom-feed/vzlogger'
make[1]: *** [package/feeds/custom/vzlogger/compile] Error 2
make[1]: Leaving directory `/home/dt/udoopenwrt'
make: *** [package/vzlogger/compile] Fehler 2

andig, 2015-01-08:
Bin keine endlosen Logs- dafür gibts pastebin. Aber wenn Du cmake hast nutze es- so soll der vzlogger laut Wiki ja auch kompiliert werden. Könntest Du das mal testen?

r00t-, 2015-01-08:
see also: https://github.com/volkszaehler/vzlogger/pull/105

r00t-, 2015-01-08:
@autohausnielsen:
woher kommt "package/vzlogger/compile"?
ist nicht unsers, koennen wir daher erstmal nicht fixen.

autohausnielsen, 2015-01-10:
@r00t: "make package/vzlogger/compile V=s" ist der Aufruf, um das Paket vzlogger mit "make" in der OpenWrt Toolchain zu compilieren. V=s schaltet Verbose ein.

r00t-, 2015-01-13:
@autohausnielsen:
die frage war nicht, was das kommando bedeutet,
sondern woher dieser build-setup, bzw. das script/makefile kommt.
soweit ich das sehe, ist das <b>NICHT VON UNS</b>,
damit koennen wir daran auch nichts aendern.

das beste was ich bisher gefunden habe, ist:
https://dev.openwrt.org/browser/branches/buildroot-ng/openwrt/package?rev=4946&order=name
aber dort gibt es vzlogger nicht.

- rausfinden woher das kommt
- dem maintainer sagen, dass er auf cmake umstellen sollte

siehe auch https://github.com/volkszaehler/vzlogger/issues/105

r00t-, 2015-01-19:
"Subject: [vz-dev] Cross-compilen für OpenWRT, #102
Anbei ein Paket um vzlogger für OpenWRT zu kompilieren.
Gruß
Udo"
-- http://demo.volkszaehler.org/pipermail/volkszaehler-dev/2015-January/004137.html

closed 2015-03-20 by andig

r00t-, 2015-03-21:
this should not really be closed.
ideally the openwrt build swtup should be included in the repo.

holzingerm, 2017-11-10:
Hallo zusammen,
Ich habe mal versucht vzlogger für meinem Router (Turris Omnia, fork von OpenWRT) zum Laufen zu bekommen
https://github.com/holzingerm/vzlogger.git
Mit deinem Makefile und ein paar Anpassungen konnte ich es kompilieren, jedoch scheint noch ein Problem zu bestehen welches ich bis jetzt nicht identifizieren konnte.
gibt es außer den angegebenen Abhängigkeiten im Makefile (+libmbus) noch weitere die ich bis übersehen habe?
kompiliert man vzlogger für ein locales system (raspbian oder ubuntu-server x64) scheint alles zu laufen.
in der Cross-Compile Umgebung erzeugt er jedoch ein binary das erschreckend klein ist und sich nicht ausführen lässt.

@r00t- 
hättest du noch einen Tip an was es liegen könnte?

PS: dabei ist mir aufgefallen das überall auf libmbus-0.8.0 oder höher verwiesen wird.
Diese version funktioniert leider mit dem aktuellen GIT stand nicht mehr da vzlogger bereits funktionen verwendet die erst später hinzugefügt wurden, man benütigt auch hier einen aktuellen GIT Stand von libmbus

franzflasch, 2022-09-05:
Hi! 
Falls es noch jemanden interessiert: Ich habe ein openwrt-package für vzlogger erstellt:
https://github.com/franzflasch/vzlogger-openwrt

Habs auf meinem kleinen GL-MT300N-V2 minirouter zum Laufen gebracht. Bin danach erst drauf gekommen, dass bei meinem Smartmeter die IrDA Schnittstelle garnicht verwendet werden kann.

Evtl ist es ja für jemand anderes von Nutzen.