volkszaehler/vzlogger #103: Update wiki to use forked libsml

andig, 2015-01-08:
Just a reminder...

andig, 2015-01-08:
Places:

  - http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version
  - http://wiki.volkszaehler.org/vzlogger-installation (skript anoassen- wo ist das Skript eigentlich???)

r00t-, 2015-01-08:
the "install script" was contributed by "Wolfgang Fahl",
http://volkszaehler.org/pipermail/volkszaehler-dev/2014-May/003592.html
http://wiki.volkszaehler.org/vzlogger-installation?do=revisions

personally i see no point and wouldn't maintain it.
i'll send him an email.

r00t-, 2015-01-08:
http://demo.volkszaehler.org/pipermail/volkszaehler-dev/2015-January/004078.html

andig, 2015-02-17:
Looking at http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version the libsml ref is updated. 

Should we now delete the install script and close this issue?

mbehr1, 2015-02-17:
fine for me.

> 
> Looking at http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version <http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version> the libsml ref is updated.
> 
> Should we now delete the install script and close this issue?

Gruß

Matthias

andig, 2015-02-18:
> Should we now delete the install script and close this issue?

Done.

closed 2015-02-18 by andig

r00t-, 2015-02-19:
i wouldn't (have) delete(d) the script from the wiki?
why is marking it as obsolete not enough?
that won't encourage the author to make further contributions.
note that he did even reply to my inquiry, see the mailing list thread.

if we were to delete every entry in the wiki that is not immediately relevant for vz.org,
the leftovers would be rather dull...

(@justinotherguy, what's your opinion?)

andig, 2015-02-20:
> why is marking it as obsolete not enough?

imho we have too much "old stuff that used to work" in the wiki and our site is too confusing for new users (e.g. the "c" version etc). Instead of a historic graveyard we should have a delicious menu.

r00t-, 2015-02-20:
i don't think deletion with history removal is applicable to anything but spam.
i'd at worst have put it here: http://wiki.volkszaehler.org/obsolete

justinotherguy, 2015-03-01:
my 2 cents:
we do need an overhaul for the structure in the wiki; with that we should find a decent place where we can put old stuff; this should be labeled very clearly.
In cases like the c-version of vzlogger, where we something to - more or less - replace it I would agree on removing it (not from the wiki history, but at least in a way that avoids people from using it by accident).
For all other cases ("stuff that used to work" or "is not being maintained by us (anymore)") I'm fine with some "obsolete" area in the wiki.