volkszaehler/vzlogger #104: Compile fails due to libcurl4-gnutls-dev

andig, 2015-01-08:
Since #97 build fails unless `libcurl4-gnutls-dev` is installed. A prereq check at least would be helpful.

andig, 2015-01-08:
@mbehr1 could you take a look?

andig, 2015-01-08:
Solved by `rm CMakeCache.txt` as already documented by @mbehr1 in #97

closed 2015-01-08 by andig

r00t-, 2015-01-08:
habt ihr auch den fix fuer:
```
/usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++0x -fpermissive -Wno-ignored-qualifiers    CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o  -o vzlogger -rdynamic protocols/libproto.a libvz.a api/libvz-api.a -Wl,-Bstatic -ljson -Wl,-Bdynamic -lrt -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -Wl,-Bdynamic -lssl -lcrypto -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -lssl -lcrypto 
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-daemon.o): undefined reference to symbol 'gnutls_certificate_set_retrieve_function2@@GNUTLS_DEBIAN_0_3_0_0'
//usr/lib/x86_64-linux-gnu/libgnutls-deb0.so.28: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

mbehr1, 2015-01-08:
Wie kommt man da hin? Setup?

(btw: bei hingen solche Themen oft noch daran, dass CMakeCache.txt gelöscht werden musste)

> habt ihr auch den fix fuer:
>
> /usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter [...]

Gruß

Matthias Behr

andig, 2015-01-09:
Wie wär's denn wenn wir für Dummies wie mich ein `make clean` Target einbauen welches den cache löscht?

milefork, 2015-01-20:
+1 ´rm CMakeCache.txt´ löst das Problem bei mir auch. Sollte in das ´make clean´ Target mit aufgenommen werden.