volkszaehler/vzlogger #107: build fail - json search problem ?

devZer0, 2015-01-10:
Compiling vzlogger from source according to http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version#building_from_source fails for me on openSUSE 13.2 (i586) with the following error:

linux-xlde:~/vzlogger/vzlogger # cmake .
-- The C compiler identification is GNU 4.8.3
-- The CXX compiler identification is GNU 4.8.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
Compiling for target ''
-- checking if -Wno-ignored-qualifiers works
-- Performing Test W_NO_IGNORED_QUALIFIERS
-- Performing Test W_NO_IGNORED_QUALIFIERS - Success
-- FindSml check
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for module 'sml>=0.9'
--   package 'sml>=0.9' not found
-- SML_HOME env is not set, setting it to /usr/local
-- Looking for sml in /usr/local
-- FindMicrohttpd check
-- checking for module 'microhttpd>=0.9'
--   package 'microhttpd>=0.9' not found
-- MICROHTTPD_HOME env is not set, setting it to /usr/local
-- Looking for microhttpd in /usr/local
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- FindJson check
-- checking for module 'json>=0.9'
--   found json, version 0.10
-- JSON_HOME env is not set, setting it to /usr/local
-- Looking for json in /usr/local
Jsoon search: '/usr/local/include;/usr/local/include;/usr/local/include;/usr/include'
CMake Error at CMakeLists.txt:85 (message):
  libjson ist required.

  Install json or call cmake -DJSON_HOME=path_to_json_install

-- Configuring incomplete, errors occurred!
See also "/root/vzlogger/vzlogger/CMakeFiles/CMakeOutput.log".
See also "/root/vzlogger/vzlogger/CMakeFiles/CMakeError.log".

using autogen.sh, configure.sh, make works - at least with the source before commit, 9a3a8c5a3f096242b826c0edbeceb9c2f5c32b87 - as that one seems to introduce the following problem:

g++  -g -O2 -lpthread -lm -lstdc++ -ljson -lcurl -lssl -lcrypto   -o vzlogger vzlogger.o Channel.o Config_Options.o threads.o Buffer.o Meter.o ltqnorm.o Obis.o Options.o Reading.o exception.o local.o MeterMap.o MeterS0.o MeterD0.o MeterSML.o MeterFluksoV2.o MeterFile.o MeterExec.o MeterRandom.o Volkszaehler.o MySmartGrid.o CurlIF.o CurlCallback.o CurlResponse.o -lsml -luuid  -lmicrohttpd
threads.o: In function `logging_thread(void*)':
/root/vzlogger/vzlogger/src/threads.cpp:197: undefined reference to `vz::api::Null::Null(std::tr1::shared_ptr<Channel>, std::list<Option, std::allocator<Option> >)'
MeterMap.o: In function `MeterMap::registration()':
/root/vzlogger/vzlogger/src/MeterMap.cpp:132: undefined reference to `vz::api::Null::Null(std::tr1::shared_ptr<Channel>, std::list<Option, std::allocator<Option> >)'

autohausnielsen, 2015-01-10:
>using autogen.sh, configure.sh, make works - at least with the source before commit, 9a3a8c5 - as that one seems to introduce the following problem:
>threads.o: In function logging_thread(void*)': /root/vzlogger/vzlogger/src/threads.cpp:197: undefined reference tovz::api::Null::Null(std::tr1::shared_ptr, std::list >)'
MeterMap.o: In function MeterMap::registration()': /root/vzlogger/vzlogger/src/MeterMap.cpp:132: undefined reference tovz::api::Null::Null(std::tr1::shared_ptr, std::list >)'

Same problem as in Issue #102

andig, 2015-01-10:
Just to be sure- `rm CMakeCache.txt` versucht?

devZer0, 2015-01-10:
after fresh git clone, there is no such file

mbehr1, 2015-01-12:
Can you check which version of libjson is installed.
Where is the header file json.h located? Doesn't seem to be in 
/usr/local/include;/usr/local/include;/usr/local/include;/usr/include
Can you double check?

Sineos, 2015-01-31:
I had the same issue

The json-c that was installed via apt-get was not found regardless of -DJSON_HOME
The json-c pulled from git fails on autotools
The json-c tarball (v0.12) that has autools already done fails to build because in the Makefile -Werror is set and some variable is unused. Deleting Werror makes the tarball build and pointing vzlogger to this build finally worked for me.

mbehr1, 2015-02-19:
on travis-ci json-c is installed via:
```
# install json-c
sudo apt-get install build-essential
sudo apt-get install wget
sudo apt-get install libtool
git clone -b json-c-0.12 https://github.com/json-c/json-c
cd json-c
sh autogen.sh
./configure
make
sudo make install
cd ..
```
can you try this?

mbehr1, 2015-02-19:
Falls sh autogen.sh fehlschlägt mit ...autoreconf missing...
```
sudo apt-get install dh-autoreconf
```

andig, 2015-03-01:
Is this still an issue? Otherwise suggesting to close as invalid.

closed 2015-03-20 by andig