volkszaehler/vzlogger #110: support s0 on raspberry pi GPIOs

r00t-, 2015-01-13:
WIP here: https://github.com/r00t-/vzlogger/tree/raspis0
see also: "request for comments: s0 meter, supporting different interfaces" https://github.com/volkszaehler/vzlogger/issues/85

mbehr1, 2015-02-23:
pls take a look at https://github.com/mbehr1/vzlogger/tree/s0_gpio as well. Was tested already ok by Udo. Quite a simple change to enable gpio support. But needs external configuration of gpio ports before start of vzlogger.

r00t-, 2015-02-24:
department of redundancy department?

- personally i much prefer having the config self-contained, instead of relying on external scripts
- my code has some other fixes too, like not consume two pulses per reading
- plus the WIP on #85, will try to push that for review sometime soon

mbehr1, 2015-02-24:
Yes, I agree.

I’d prefer a more complete solution as well. 
(There’ve been some questions/discussions on the mailing list and there was no answer/reply/comment from your side.)

Basically I thought that the current meterS0 code would already work with gpios. During test I noticed that gpios need blocking reads so I quickly changed it…

Waiting for your changes for review.

> department of redundancy department?
>
> personally i much prefer having the config self-contained, instead of relying on external scripts
> my code has some other fixes too, like not consume two pulses [...]

Gruß
Matthias

gitka, 2015-03-06:
Tried to test S0-support on Udos Addon-board from https://github.com/mbehr1/vzlogger/tree/s0_gpio for local HTTP-support, but don't get a result or error-message.
[mtr1] Meter configured, enabled.
[chn2] New channel initialized (uuid=...f95918 api=NULL id=Impulse)
local http-output:
{ "version": "0.4.0", "generator": "vzlogger", "data": [ { "uuid": "7836dd20-00d5-11e0-bab1-856ed5f95918", "last": 0, "interval": -1, "protocol": "s0" } ] }
Gruß
Karlheinz

mbehr1, 2015-03-06:
Ist das mit vollem Loglevel?

Conf Datei?

Gruß
Matthias 

Sent from a mobile device.

gitka, 2015-03-07:
Hi,

> Ist das mit vollem Loglevel?
das war nur auszugsweise bei verbosity=10
Mit 15 u. ausgeschaltetem ersten Meter (d0), sieht das so aus:
[Mar 07 18:24:08]       Opened logfile /home/pi/vzlogger.log
[Mar 07 18:24:08][]     ===> Start meters
[Mar 07 18:24:08][mtr0] Meter for protocol 'd0' is disabled. Skipping.
[Mar 07 18:24:08][mtr1] Meter connection established
[Mar 07 18:24:08][mtr1] Meter thread started
[Mar 07 18:24:08][mtr1] Meter is opened. Starting channels.
[Mar 07 18:24:08][http] Starting local interface HTTPd on port 8080
[Mar 07 18:24:08][]     Startup done.
[Mar 07 18:24:08][mtr1] Number of readers: 2
[Mar 07 18:24:08][mtr1] Config.daemon: 0
[Mar 07 18:24:08][mtr1] Config.local: 1
[Mar 07 18:24:08][mtr1] Got 0 new readings from meter:
"Got 0 new readings" ...  innerhalb einer Minute 35 MB logfile
> Conf Datei?
Auszug vzlogger.conf:
...    "enabled" : true , // true, Beginn S0 Impulse
     "skip": false,
     "protocol" : "s0",
     "aggtime": 60, // aggregate all signals for x seconds.
     "device": "sys/class/gpio/gpio17/value",
     "resolution": 2000,
     "channel": {
         "api" : "NULL" ,
         "uuid" : "7836dd20-00d5-11e0-bab1-856ed5f95918",
         "identifier" : "Impulse", /* Impulszähler */
         "aggmode": "SUM",
         } ...

/etc/rclocal :
echo 17 > /sys/class/gpio/export && echo "rising" > 
/sys/class/gpio/gpio17/edge

Gruß
Karlheinz

mbehr1, 2015-03-07:
Innerhalb einer Min. 35MB?

Was steht denn da drin? Kannst du kurzen Auszug schicken?

> Hi,
>
> das war nur auszugsweise bei verbosity=10
> Mit 15 u. ausgeschaltetem ersten Meter (d0), sieht das so aus:
> [Mar 07 18:24:08] Opened logfile /home/pi/vzlogger.log
> [Mar 07 18:24:08][] ===> Start [...]

Gruß

Matthias Behr

gitka, 2015-03-07:
> Innerhalb einer Min. 35MB?
>
> Was steht denn da drin? Kannst du kurzen Auszug schicken?
>
wie gesagt, ständig diese Zeile:
[Mar 07 18:24:08][mtr1] Got 0 new readings from meter:

Wie kann ich prüfen, ob die Impulse wirklich auf GIO17/value ankommen?

mbehr1, 2015-03-07:
"device": "sys/class/gpio/gpio17/value“,

da fehlt ein „/„ -> „/sys/class/gpio/…“

> 
> 
> > Innerhalb einer Min. 35MB?
> >
> > Was steht denn da drin? Kannst du kurzen Auszug schicken?
> >
> wie gesagt, ständig diese Zeile:
> [Mar 07 18:24:08][mtr1] Got 0 new readings from meter:
> 
> Wie kann ich prüfen, ob die Impulse wirklich auf GIO17/value ankommen?

Gruß

Matthias

andig, 2015-03-08:
Moin,

 Ist es evtl sinnvoll eine regex zur Prüfung des Devices einzubauen die bei Bedarf zumindest eine Warnung ausgibt?

Viele Grüße, Andreas

mbehr1, 2015-03-08:
Ja, macht Sinn. Kann ich hinzufügen. Allerdings soll der Code eigentlich eh nicht aufgenommen werden. Ich warte noch auf die eigentliche Lösung von r00t- mit generischem Ansatz.

> 
> Moin,
> 
> Ist es evtl sinnvoll eine regex zur Prüfung des Devices einzubauen die bei Bedarf zumindest eine Warnung ausgibt?
> 
> Viele Grüße, Andreas
> 
> 
> 
> 
> > 
> > "device": "sys/class/gpio/gpio17/value“,
> > 
> > da fehlt ein „/„ -> „/sys/class/gpio/…“
> > 
> > 
> > > 
> > > 
> > > > Innerhalb einer Min. 35MB?
> > > >
> > > > Was steht denn da drin? Kannst du kurzen Auszug schicken?
> > > >
> > > wie gesagt, ständig diese Zeile:
> > > [Mar 07 18:24:08][mtr1] Got 0 new readings from meter:
> > > 
> > > Wie kann ich prüfen, ob die Impulse wirklich auf GIO17/value ankommen?
> > > 
> > 
> > Gruß
> > 
> > Matthias 

Gruß

Matthias Behr

gitka, 2015-03-08:
> Ja, macht Sinn. Kann ich hinzufügen. Allerdings soll der Code 
> eigentlich eh nicht aufgenommen werden. Ich warte noch auf die 
> eigentliche Lösung von r00t- mit generischem Ansatz.
>
Ja, eine Prüfung auf Vorhandensein des files/port wäre hilfreich. Ich 
frag mich nur, wie generisch die Lösung von root aussieht?

Anscheinend ist in meinen Einstellungen noch etwas falsch - ich erhalte 
noch kein reading.
[chn2] Buffer dump (size=0): {}

In der Kombination
"daemon":false
"local":"enabled":false

sollte vzlogger im Vordergrund laufen. Hier erhalte ich folgende Meldung:
[Mar 08 13:49:55][chn2] Start logging thread for NULL-api. Running as 
daemon: no
terminate called after throwing an instance of 'vz::OptionNotFoundException'
   what():  Option 'middleware' not found
[Mar 08 13:49:55][mtr1] Got 0 new readings from meter:
Aborted

OK- Es läuft nur dann ohne Fehler, wenn für "middleware" etwas 
eingetragen ist - auch wenn "api"=NULL

Trotzdem noch keine readings

r00t-, 2015-03-08:
i suggest to keep the structure part in the other thread:
> sorry for the delay, pushed my unfinished work here:
> https://github.com/r00t-/vzlogger/tree/bases0
> 
> please have a look, comment, use, or discard...
-- https://github.com/volkszaehler/vzlogger/issues/85#issuecomment-77751155

(please note that the code in my raspis0 branch is still working as-is,
and probably more complete than mbehr1's)

mbehr1, 2015-03-08:
Eine Prüfung auf Existenz des Files gibt es eigentlich schon:
	if (fd < 0) {
		print(log_error, "open(%s): %s", "", _device.c_str(), strerror(errno));
		return ERR;
	}

bekommst du die Warnung? (mit dem alten „sys/…“)?

Wie konfigurierst du den Pin bevor du vzlogger startest?

@Udo: kannst du die Settings noch mal schicken?

> Ja, eine Prüfung auf Vorhandensein des files/port wäre hilfreich. Ich
> frag mich nur, wie generisch die Lösung von root aussieht?
>
> Anscheinend ist in meinen Einstellungen noch etwas falsch - ich [...]

Gruß

Matthias

gitka, 2015-03-16:
> Eine Prüfung auf Existenz des Files gibt es eigentlich schon:
> if (fd < 0) {
> print(log_error, "open(%s): %s", "", _device.c_str(), strerror(errno));
> return ERR;
> }
>
> bekommst du die Warnung? (mit dem alten „sys/…“)?
>
Nein, leider keine Warnung.
> Wie konfigurierst du den Pin bevor du vzlogger startest?
s. s0vz <https://github.com/w3llschmidt/s0vz>: /etc/rclocal  :
echo 17 > /sys/class/gpio/export && echo "rising" > 
/sys/class/gpio/gpio17/edge

  In welchem Zweig wird nun weiterentwickelt - mbehr oder root?

UdoSchake, 2015-03-28:
die '/etc/vzlogger_gpio.conf' sollte folgendes beinhalten:
 "meters": [{
            "enabled": true,               // disabled meters will be ignored
            "skip": false,                  // if enabled, errors when opening meter will lead to meter being ignored
            "protocol": "s0",
            "mode": "gpio",
            "aggtime": 5,                 // aggregate all signals and give one update to middleware every 5 seconds
            "aggfixedinterval": true,       // round all timestamps to middleware to nearest aggtime
            "device": "/sys/class/gpio/gpio4/value",
            "channel": {
                "api": "volkszaehler",      // default middleware api: volkszaehler.org
                "identifier": "Impulse",    // s0 meter knows "Impulse" and "Power"
                "uuid": "deine UUID",
                "middleware": "http://127.0.0.1/middleware.php",
                "aggmode": "SUM"            // add all s0 intervals in the aggregation
                       }
        },

und die '/etc/rc.local' für den dazu gehörigen GPIO-Pin:
echo 4 > /sys/class/gpio/export && echo "rising" > /sys/class/gpio/gpio4/edge

Gruß
Udo

r00t-, 2015-05-16:
gpio is generic and not raspi specific, meter should be named accordingly.
https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

mbehr1, 2015-06-06:
see #169 with a solution for this topic.

mbehr1, 2015-07-12:
can we close this issue?

closed 2015-07-12 by andig