volkszaehler/vzlogger #112: Duplicate Key Entry

milefork, 2015-01-20:
Hi, 

vzlogger claims about a Duplicate Primary Key Entry. See http://pastebin.com/v31nLdVR how this can be solved?

andig, 2015-01-20:
Schau mal bitte was in der composer.json der Middleware steht?

Viele Grüße, Andreas

milefork, 2015-01-20:
welche? Die im root? Wenn ja: http://pastebin.com/khcf6NLj

andig, 2015-01-21:
Kannst Du bitte auch die `composer.lock` noch posten? Aktuelle Version vom vzlogger? Bitte Datei `src/Volkszaehler.cpp` aus vzlogger ebenfalls posten.

milefork, 2015-01-21:
Ja gestern erst neu kompiliert -> b3d5bd915b0270ef0aa819367d19ee571318d106 . 
Auch volkszähler.org ist auf dem Stand von gestern : `290f4c803a7b6042cc0632f9b90d1db1cd0141a9`
`composer.lock` -> http://pastebin.com/rji2wKAw
An die `scr/Volkszaehler.cpp` komme ich gerade nicht ran, sollte aber zu 99% die aus dem Commit von gestern sein.

andig, 2015-01-21:
Verstehe ich nicht. http://pastebin.com/v31nLdVR zeigt eine `DBALException`. http://pastebin.com/rji2wKAw zeigt DBAL 2.5.1, die habe ich ebenfalls installiert. 

Bei mir gibt es bei duplicate key eine andere Exception die im vzlogger auch korrekt behandelt wird: `UniqueConstraintViolationException`.

Zu testen einfach mal ein Duplikat anlegen (link 2x im Browser aufrufen): 
[http://localhost/middleware.php/data/<uuid-zum-testen->.json?operation=add&value=1&ts=1](http://localhost/middleware.php/data/<uuid-zum-testen->.json?operation=add&value=1&ts=1)

Welche Exception kommt dann?

Bist Du sicher dass Du gegen die Middleware loggst deren Parameter Du hier gepostet hast?

milefork, 2015-01-21:
Oh ******* , bin mit meinem Server umgezogen, und hatte URL redirects eingebaut, die eigentlich auf meinen neuen Server zeigen, die haben aber anscheinend nicht so funktioniert wie gedacht. Habe gerade die URLs angepasst und jetzt läuft alles. Sorry für die Umstände :/

closed 2015-01-21 by milefork