volkszaehler/vzlogger #120: Suggestion: JSON logger

andig, 2015-02-10:
We already have a log file parser which is essentially:

  - a file transport to read the file line by line plus
  - a pattern matcher for that line (which seems to be broken but alas)

A JSON logger would consist of:

  - an HTTP(S) transport
  - a JSON parser
  - a search expression

Logfile and JSON parser might/should use the same infrastructure and config settings in that regard.

Features I would be looking for are:

   - search expression for target value:
     - properties: `parent.child`
     - arrays: `parent[1]`
     - arrays with access from end: `parent[-1]`
     - child property matching for arrays: parent.channels[@uuid=abc]
  - option to skip duplicates if found value == last value

The latter option might also make general sense for other meters.

What do you think?

andig, 2015-02-23:
I've created a prototype for the parsing implementation here:

https://github.com/volkszaehler/volkszaehler.org/pull/252

r00t-, 2015-02-23:
maybe the load-data-from-http part could be abstracted to also parse formats other than json?
i.e. xml/html via xpath, or plain regular expressions.
for example to replace "htmlvz": http://wiki.volkszaehler.org/howto/wechselrichter_kostal_piko

andig, 2015-03-19:
> for example to replace "htmlvz"

Alternatively, we could add this functionality to the jsonclient of Volkszaehler. We already have HTTP GET and could add CSS selectors instead of JSON parsing.

I'm beginning to think that vzlogger might indeed not be the best place to put this functionality as it can be easier realized in scripting. Reading e.g. log files via blocking IO may be a different case as it's not suitable for a cron-based approach?

andig, 2016-06-25:
VZ meeting: can already be implemented using jsonclient together with vzclient.

closed 2016-06-25 by andig