volkszaehler/vzlogger #126: fix debian-build to use cmake instead of automake

r00t-, 2015-02-19:
see also https://github.com/volkszaehler/vzlogger/pull/105

https://github.com/volkszaehler/vzlogger/blob/master/debian/

andig, 2015-03-01:
Versteh ich nicht. Wäre es nicht sinniger debian-build zu entfernen da wir

  a) keine Packages brauchen und
  b) ohnehin kein Maintainer da ist

?

andig, 2015-03-01:
What is this issue really about opposed to https://github.com/volkszaehler/vzlogger/pull/105 ? Is it for the open follow-on issues of https://github.com/volkszaehler/vzlogger/pull/105 ? Then it should likely be more specific on what needs solving.

r00t-, 2015-03-01:
as i wrote there, #105 does not contain actually changes to the debin stuff.
as such, the debian build will still try to run autotools.
(i made an attempt to change that, no idea where i left that, but the changes were trivial anyway)

let's just merge #105, and then either discuss to remove /debian (i would not),
or continue here.

justinotherguy, 2015-03-01:
my 2 cents:
although we are not going to provide debian packages anymore I'd also like to keep the infrastructure on board to build them (as long as we don't have trouble to keep it in a working condition)

andig, 2015-03-20:
#105 is merged. Anybody wants to take on debian build? Otherwise I'd rather remove than have it non-working for sake of clarity and sanity.

cmorty, 2016-01-17:
OT: @andig I'm willing to look into Debian or rather Ubuntu packaging. I've registered the project at launchpad and just added my first pull-request #234. If there is a release-branch I'd set things up for lauchpad to automatically build nightlies and releases based on theses branches. 
This commitment would be on push basis: I'll not follow the project, but someone has to assign a bug to me (I already have too many projects to follow another one ;) ).

andig, 2016-01-27:
ping @justinotherguy and @r00t- packaging is not my home turf...

r00t-, 2016-01-27:
@cmorty:
does the debian build works for you (besides the missing build-deps?)
my status was that the control file in the repository does not even get to start building...
see: https://github.com/volkszaehler/vzlogger/pull/105#issuecomment-69676003
i did not know how to fix that back then, have no time for working on it now, so very welcome if you can fix it!

cmorty, 2016-02-01:
Building now works at LP, which is pretty picky.  - see https://github.com/volkszaehler/vzlogger/pull/234#issuecomment-178103873 Should work on Debian, too.

andig, 2016-05-05:
Closing in favor of https://github.com/volkszaehler/vzlogger/pull/234

closed 2016-05-05 by andig