volkszaehler/vzlogger #127: near impossible to build on debian sid

r00t-, 2015-02-25:
i think it is since #97 was merged, that i could never get vzlogger to build cleanly on my debian sid systems,
and it's been taking any fun out of developing.
i am running into a few issues:

- for inexplicable reason, all depedancies of libcurl are linked statically.  
  is there any reason for this?
  it means one has to install some 20 -dev packages of libraries to have the static libs around, 
  and which are not direct dependancies or curl-dev, because they are not needed for dynamic against it.
  i most of the time eneded up manuall removing all static declarations from the linker invocation
- /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-daemon.o): undefined reference to symbol 'gnutls_certificate_set_retrieve_function2@@GNUTLS_DEBIAN_0_3_0_0'
//usr/lib/x86_64-linux-gnu/libgnutls-deb0.so.28: error adding symbols: DSO missing from command line
- it somehow seems that curl and microhttpd are linking two different versions of gnutls?

mbehr1, 2015-02-25:
Can you post your cmake and make output?

> i think it is since #97 <https://github.com/volkszaehler/vzlogger/pull/97> was merged, that i could never get vzlogger to build cleanly on my debian sid systems,
> and it's been taking any fun out of [...]

Gruß

Matthias

mbehr1, 2015-02-25:
one my system there is no static linkage of libcurl:
```
ldd src/vzlogger
	linux-gate.so.1 =>  (0xb76f3000)
	librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xb76c9000)
	libgcrypt.so.11 => /lib/i386-linux-gnu/libgcrypt.so.11 (0xb7642000)
	libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb7625000)
	libuuid.so.1 => /lib/i386-linux-gnu/libuuid.so.1 (0xb761f000)
	libcurl.so.4 => /usr/lib/i386-linux-gnu/libcurl.so.4 (0xb75b7000)
...
```
can you post an ldd output and indicate the wrong ones?

r00t-, 2015-03-08:
please be aware that statically linked libs will not show up in ldd for obvious reasons.
the issue is/was, that cmake would generate a lot of `-Wl,-Bstatic` in the final linker invokation.
right now i can't reproduce that with current head,
so maybe it was fixed.

the issue that remains is:
```
/usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++0x -fpermissive -Wno-ignored-qualifiers    CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o  -o vzlogger -rdynamic protocols/libproto.a libvz.a api/libvz-api.a -Wl,-Bstatic -ljson-c -Wl,-Bdynamic -lrt -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -Wl,-Bdynamic -lssl -lcrypto -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -lssl -lcrypto 
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-daemon.o): undefined reference to symbol 'gnutls_certificate_set_retrieve_function2@@GNUTLS_DEBIAN_0_3_0_0'
//usr/lib/x86_64-linux-gnu/libgnutls-deb0.so.28: error adding symbols: DSO missing from command line
```
which can be fixed by manually adding /usr/lib/x86_64-linux-gnu/libgnutls-deb0.so.28 to the link command.
but note that "-Wl,-Bstatic -lgnutls" is already linked, so it seems where are linking two versions if it...

this may very well be debian's bug and not of ours,
it just has me rather frustrated in any case...

mbehr1, 2015-03-13:
Those errors are typically (or quite often) due to wrong order for the parameter given to the linker. 

> please be aware that statically linked libs will not show up in ldd for obvious reasons.
> the issue is/was, that cmake would generate a lot of -Wl,-Bstatic in the final linker invokation.
> right now i [...]

Gruß

Matthias

andig, 2015-03-20:
@r00t- still an issue?

closed 2015-11-10 by andig