volkszaehler/vzlogger #128: Add commit date to version hash

andig, 2015-02-25:
vzlogger -v shows commit hash. Unfortunately commit hash does not give direct indication how old the build or rather the commit is.

Would it be possible to add the commit date as well?

mbehr1, 2015-02-25:
I’ll take a look. You’d prefer commit date and not build date, right?

> 
> vzlogger -v shows commit hash. Unfortunately commit hash does not give direct indication how old the build or rather the commit is.
> 
> Would it be possible to add the commit date as well?

Gruß
Matthias

nyphis, 2015-02-25:
Why not displaying both?

Btw.: With my current (up to date) build the vzlogger states
```
# vzlogger -V
0.4.0
 based on git version: heads/master-0-g077a180c35
```
which has fairly little to do with the current commit hash `79166a9fc0f7fa826909c83cd78268bf81b5db58` ... or am I wrong?

mbehr1, 2015-02-25:
can you post a
```
git log -n 2
```
output?

nyphis, 2015-02-25:
```
# git log -n 2
commit 077a180c3519c220d6131c3e01cef121c2fbf9b9
Merge: b80e6a8 79166a9
Author: root <root@PowerPi.fritz.box>
Date:   Tue Feb 24 17:38:28 2015 +0100

    Merge branch 'master' of https://github.com/volkszaehler/vzlogger

commit 79166a9fc0f7fa826909c83cd78268bf81b5db58
Merge: 02b9e5c 0f7719f
Author: andig <cpuidle@gmx.de>
Date:   Tue Feb 24 07:22:51 2015 +0100

    Merge pull request #124 from mbehr1/fix_LuG_E350

    fix and unit tests for Landis u Gyr E350
```

mbehr1, 2015-02-25:
so it fits. 077... is the last commit. (heads/master-g077...)

mbehr1, 2015-02-25:
I added PR#129 to add last commit date to -V. I think build date is not really useful (but could be added later simply by adding a __DATE__ to printf.

mbehr1, 2015-02-25:
see here an example output:
```
src/vzlogger -V
0.4.0
 based on git version: heads/commit_date-0-g79166a9fc0-dirty
 last commit date: Tue, 24 Feb 2015 07:22:51 +0100
```

andig, 2015-02-26:
Great, thank you!

closed 2015-02-26 by andig

justinotherguy, 2015-02-26:
nice one, Matthias :-) :+1: