volkszaehler/vzlogger #130: Feature request: config-file editor "make menuconfig" like

mbehr1, 2015-02-27:
As more and more people struggle with the complexity of the configuration file I suggest to add a config-file editor like linux kernel "make menuconfig", raspi-config, opensips/makemenuconfig,...

The editor should support:
  - taking the etc/vzlogger.conf as a template (incl. help text from comments)
  - asking the customer which options to choose in a structure way (global options, adding meters? (add->d0->ask options)
  - writing only the needed/configured options into a conf file
  - reading of existing conf files and using this as preset input
  - ...

Using this approach we could make the etc/vzlogger.conf much more complex/unreadable by adding more detailed comments, descriptions,... (instead of trying to keep the documentation in sync in the wiki or source or conf file...) and generate a specific config file easy to read, maintain and understand.

What do you think about it? suggestions/comments welcome/requested.

Is anybody aware of such an editor or some other tool/source to start with? (opensips/makemenuconfig might be a starting point providing a simple ncurses based implementation, json parsing/stuff to be added,...)

andig, 2015-03-03:
Excellent idea but imho not entirely necessary right now. Might even- bonus- start from type of meter to connect instead of from asking the parameters (Elster xyz...). 

> Is anybody aware of such an editor or some other tool/source to start with? 

Nope...

>  json parsing/stuff to be added

I guess that's the tricky part that gets complex. Maybe just have it as one-way or "initial" config tool? If json doesn't comply with/ or has tool-generated MD5 signature you'd get a warning of "overwriting" the config.

skre, 2015-03-15:
I like the idea. Perhaps we could collect some typical configuration for meters out of the wiki. After an "auto-config" everybody would be free to adjust his/her config-file - what would be much easier.

dave-watts, 2015-03-19:
As a start, can the config file not just be better documented with a repository for typical example applications?

mbehr1, 2015-03-22:
How about generating a json-schema for our json-conf-file? There are many editors and validators for json-schema. And references to other schemas (e.g. for meterOCR conf file) would be feasible as well.
I'll provide an example the next days.

mbehr1, 2015-03-23:
see here an example: 
http://jeremydorn.com/json-editor/?schema=N4IgJAzgxgFgpgWwIYgFwhgF0wB1QenwCsIB7AOwFpp5kA6UgJwHN8ATRpAM00oAYALPhqIkAYhAAaEAEs2aEFJCYAnjjgLSAIyJwomJTkal1jTDLgQ0oRnEyMV12fPT5b9x9NXqFM8pjhmOEYlNksoRhkccwoFAAlSAHcAAgAbCmZkzFJkiFS4OBxkrTtEgvJkriQZfLZk2wBHAFdLTAhJZL9cvQo2KwBfaTYkRFjUUDkFdhGEWK81DXQtUlJ8pHJQuCqm1IN0exbN6EjomTGQAFFyJC185OHRitmwkEGQADdg5YgZVSdJ1yfRjfX6eZQLXz+QLBJRwchNBBoADafEkAFZJABGVGYtEAXSG4ROMQ26D4yQAvGlSMwAPrBYyMZJManMSiJJCMch+ZgdNGU1m0vxcUgdbEC9JssJaJq85K4iU02lcPytTbbXZoPhvSX/FwgfC6+Y+dAQew8o4RKIkhQ4JCYGBZHKSyo1DQ60hQJCpPVTdJen3GxYgbS6fSGYymcyWJxwm61X2uf3e/Bx25weRBhTLVZwdaW4lnUkgCAwUg7OplXKYTmYLLwakB5JxAAqLYACnURUyIMF3jy0jJPvU82weRAAPyvaQ4Jh7Cb6w2elOzsxKbzBvwBIIhQkavYADj4R8Jx2tRYUDrgyRbAGF28lV3Xu/Xr8nUs2253p7JyGEAB6Jga774H4AHrhCSwrGsGynlapznKW5apJW15GKQ/ZhMkSCdH+cD/oOZoDqQXDYe81SpPG16wOs5BwKkECdKR5A5AAqqxACSAAiyQcoxjQtGaGZTm85gIHA5bzs4frLqk+BiRJTQGFm6BbtCIRvDKXBcDC4zSUmsn4FpOm7uCJq/tuMJ7kgOx7JQmJwYW5xlik6TkJk2TFE02nBCOSBju5jEvlejberhASMFUUBwB0XS9lAvSMQAFDIpHMLY9q+XwACUzJMi5yTIOQKhZE0OD5IxnnuE0XKPr5NHkHRH6pckAA85JJZQADMArVbVPWYGVFXZa8bwCTItguEiIBpgmeL9GNcDNBNGbIiADyzLBIC6vNQAA==&value=N4IgTgpgLmCeIC4AMAaEATAhhAtgewDtEYBXCNANwjACM8BnASyngQFY0AbPAc0RBBc8AY0ydEoCAUw1OEdIgBmY+hAC+aoA&theme=bootstrap2&iconlib=fontawesome4&object_layout=normal&show_errors=interaction&disable_edit_json
"JSON Output" should be a valid conf file. Hint: the properties button can enable/hide optional settings.
Schema definitions can be put in multiple files or areas within one file to make it more readable.
The schema used looks like that:
```
{
  "$schema": "http://json-schema.org/draft-04/schema#",
  "id": "",
  "type": "object",
  "properties": {
    "retry": {
      "id": "/retry",
      "type": "integer",
      "description": "How long to sleep between failed requests, in seconds"
    },
    "daemon": {
      "id": "/daemon",
      "type": "boolean",
      "default": "true",
      "description": "Enable daemon mode"
    },
    "verbosity": {
      "id": "/verbosity",
      "type": "integer",
      "enum":[0, 5, 10, 15], "description":"log_error or log_warning",
      "default": 0,
      "description": "0 = log_error or log-warning, 5 = log_info, 10 = log-debug, 15 = log_finest"
    },
    "log": {
      "id": "/log",
      "type": "string",
      "description": "path to log file"
    },
    "local": {
      "id": "/local",
      "type": "object",
      "properties": {
        "enabled": {
          "id": "/local/enabled",
          "type": "boolean",
          "description": "should we start the local HTTPd for serving live readings?"
        },
        "port": {
          "id": "/local/port",
          "type": "integer",
          "default": 8080,
          "description": "the TCP port for the local HTTPd"
        },
        "index": {
          "id": "/local/index",
          "type": "boolean",
          "description": "should we provide a index listing of available channels if no UUID was requested?"
        },
        "timeout": {
          "id": "/local/timeout",
          "type": "integer"
        },
        "buffer": {
          "id": "/local/buffer",
          "type": "integer",
          "default": -1,
          "description": "how long to buffer readings for the local interface, in seconds (if greater 0) or how many tuples to return per channel if <0 (-3 = return 3 tuples)"
        }
      },
"required": [ "enabled" ]
    }
  },
  "required": [
    "daemon",
    "log"
  ]
}
```

mbehr1, 2015-03-23:
See here an example with the possibility to add meters (till now just with enabled? option):
http://jeremydorn.com/json-editor/?schema=N4IgJAzgxgFgpgWwIYgFwhgF0wB1QenwCsIB7AOwFpp5kA6UgJwHN8ATRpAM00oAYALPhqIkAYhAAaEGzhcAluXmZ5FCGlAAbUlCSaNITAE8ccNCFIAjInCiYpIHI1KnGKuOtSg45JJc1wbAbyQej42rqa+D5+AUHSxqbmlqSkAUjkDrLQjPI4KhTmEDCkAK6abAAEAO5wlRCYSG6VmPCVEXqVABIAKj0AClVcTPVwjABuiszt8uN1jHBIbFMQAPwgAL7SOEz2XiAh5uE6evg7bg6JZuiKmHDMY1lySOV7ABx8H9LZULn5qpl0K06j0AML9SrnTCVYaMFptDqabp9QabaSKWQAD2CoRAx0i+AxcGxCRM1xAKTSi0y3w8vzyBUBIGKZQqNTqTlIk1klSQlSJmJmDSmlVIXF54yQ8k0sTqsAy5Dgmgg/PF5FIlQAqpqAJIAERqSBVCwAjqUPHc2OstoZ5Ag4GU9qBDmFEfgVPbHZcyeZbvdHjbLKUuFxHvsXXi3UGQ49SUkbuQ7g9GJsbabSvIFqEANogGL+QIgAC6NvtdxT+yu5isNjsDk5rncnm8vgLoVAVfQlPSNJkdL+jPM+YCorhywgsvh8hVZbGdEqeunsqqs7h1WlSMsdXkzHVWboTy4L00eyPyrgGzTcDNmcLqFzw8LJekq+bhh96CanCMDgQih1dwIJ4ACM6KAZ42agGACxcOYYjsHIijKACED4KumxFtISzLIyegAYgb7QXIcEIQoSiMqh6GXjanYWNYtj2NsziNvIHgGAsmCMD+4a4vgHFcd68YHIm/oprSOQMgC5hdKQ1TtBQ0yYBqEABHAOCVFumC1D4MJSnElTphaECSPy5CjFAFBsOoNpsEgiCFDxRy2fZva0d21KHseeyGIw5pPBJ/wOSAACirYjs5CAUJUkWyGiIBzIwKQQMo3HOrxCVJSlgnkn6yYOD4pQIGg2Z8JIACskjAaVwFlZhfYBYO6B8JUAC88nMAA+mMzhwiM2jMJQ1RNEo5DMCZZWte1HWKMMJlVZN/WULIQZjZUNULaQnVkRanmvGgfA2v1OJHEdcbkg0uSjf59KBUyOBIK0LQav1MLSmYh0nPo+xEbB6DwbIZHIWo+J6HFr4GD9JEA0hFFoXA5bWVeN5ZsVMh2ZFvZHSWQAAA&value=N4IgTgpgLmCeIC4AMAaEATAhhAtgewDtEYBXCNANwjACM8BnASyngQFY0AbPAc0RAD0FTGAHceQgF7ie1AHTiQXPAGNMnRKAgFMNThHSIAZuvoQAvmhzRq9RAG0tOvQeOmLAXXNA&theme=bootstrap2&iconlib=fontawesome4&object_layout=normal&show_errors=interaction&ajax

andig, 2015-03-24:
> http://jeremydorn.com/json-editor/?schema=N4IgJAzgxgFgpgWwIYgFwhgF0wB1QenwCsIB7AOwFpp5kA6UgJwHN8ATRpAM00oAYALPhqIkAYhAAaEGzhcAluXmZ5FCGlAAbUlCSaNITAE8ccNCFIAjInCiYpIHI1KnGKuOtSg45JJc1wbAbyQej42rqa+D5+AUHSxqbmlqSkAUjkDrLQjPI4KhTmEDCkAK6abAAEAO5wlRCYSG6VmPCVEXqVABIAKj0AClVcTPVwjABuiszt8uN1jHBIbFMQAPwgAL7SOEz2XiAh5uE6evg7bg6JZuiKmHDMY1lySOV7ABx8H9LZULn5qpl0K06j0AML9SrnTCVYaMFptDqabp9QabaSKWQAD2CoRAx0i+AxcGxCRM1xAKTSi0y3w8vzyBUBIGKZQqNTqTlIk1klSQlSJmJmDSmlVIXF54yQ8k0sTqsAy5Dgmgg/PF5FIlQAqpqAJIAERqSBVCwAjqUPHc2OstoZ5Ag 4GU9qBDmFEfgVPbHZcyeZbvdHjbLKUuFxHvsXXi3UGQ49SUkbuQ7g9GJsbabSvIFqEANogGL+QIgAC6NvtdxT+yu5isNjsDk5rncnm8vgLoVAVfQlPSNJkdL+jPM+YCorhywgsvh8hVZbGdEqeunsqqs7h1WlSMsdXkzHVWboTy4L00eyPyrgGzTcDNmcLqFzw8LJekq+bhh96CanCMDgQih1dwIJ4ACM6KAZ42agGACxcOYYjsHIijKACED4KumxFtISzLIyegAYgb7QXIcEIQoSiMqh6GXjanYWNYtj2NsziNvIHgGAsmCMD+4a4vgHFcd68YHIm/oprSOQMgC5hdKQ1TtBQ0yYBqEABHAOCVFumC1D4MJSnElTphaECSPy5CjFAFBsOoNpsEgiCFDxRy2fZva0d21KHseeyGIw5pPBJ/wOSAACirYjs5CAUJUkWyGiIBzIwKQQMo3HOrxCVJSlgnkn6yYOD4pQIGg2Z8JIACskjAaVwFlZhfYBYO6B8JUAC88nMAA+mMzhwiM2jMJQ1RNEo5DMCZZWte1HWKMMJlVZN/WULIQZjZUNULaQnVkRanmvGgfA2v1OJHEdcbkg0uSjf59KBUyOBIK0LQav1MLSmYh0nPo+xEbB6DwbIZHIWo+J6HFr4GD9JEA0hFFoXA5bWVeN5ZsVMh2ZFvZHSWQAAA&value=N4IgTgpgLmCeIC4AMAaEATAhhAtgewDtEYBXCNANwjACM8BnASyngQFY0AbPAc0RAD0FTGAHceQgF7ie1AHTiQXPAGNMnRKAgFMNThHSIAZuvoQAvmhzRq9RAG0tOvQeOmLAXXNA&theme=bootstrap2&iconlib=fontawesome4&object_layout=normal&show_errors=interaction&ajax

Fancy. Can we integrate this an an online helper instead as separate binary?

mbehr1, 2015-03-24:
Yes. That's the idea. We could either host it somewhere on volkszaehler.org or provide the html-pages/-scripts for installation on the local webserver.

andig, 2015-03-24:
Excellent. Depending on what it takes it might even be hostable in the
github pages.

mbehr1, 2015-03-24:
@andig : the code for the editor is here: https://github.com/jdorn/json-editor (MIT license). Can you (or some other volunteer ;-) check whats needed for an installation and where it would fit best?
A nice solution would be:
 - web server where the editor is hosted.
 - Schema already linked/preloaded to our repo (/etc/vzlogger.conf.json)
 - possibility to upload an existing .conf file and download the created/modified one.

I'd work on the schema files.

andig, 2015-03-25:
I'll take a look.

andig, 2015-03-25:
We could do this on github.io. See http://andig.github.io/vzlogger/ for an example. This page lives in the gh-pages branch.

We'll need the schema as separate file (preferably as part of the gh-pages branch to not clutter the master) and add some logic to load schema and default vzlogger.conf. If you have a working schema please let me know for an initial commit to vzlogger.

We can further link the editor to a volkszaehler.org subdomain (e.g. conf.volkszaehler.org or better volkszaehler.org/vzloggerconf) if @justinotherguy wants to set one up.

mbehr1, 2015-03-25:
Perfect. I’ll prepare the schema file.
(For testing a button/possibility to upload a schema file would be handy.)

> We could do this on github.io. See http://andig.github.io/vzlogger/ <http://andig.github.io/vzlogger/> for an example. This page lives in the gh-pages branch.
>
> We'll need the schema as separate file [...]

Gruß

Matthias

mbehr1, 2015-03-25:
See here an updated schema file with support for meter (s0 and d0) and channel config:
http://jeremydorn.com/json-editor/?schema=N4IgJAzgxgFgpgWwIYgFwhgF0wB1QenwCsIB7AOwFpp5kA6UgJwHN8ATRpAM00oAYALPhqIkAYhAAaEGzhcAluXmZ5FCGlAAbUlCSaNITAE8ccNCFIAjInCiYpIHI1KnGKuOtSg45JJc1wbAbyQej42rqa+D5+AUHSxqbmlqSkAUjkDrLQjPI4KhTmEDCkAK6abAAEAO5wlRCYSG6VmPCVEXqVABIAKj0AClVcTPVwjABuiszt8uN1jHBIbFMQAPwgAL7SOEz2XiAh5uE6evg7bg6JZuiKmHDMY1lySOV7ABx8H9LZULn5qpl0K06j0AML9SrnTCVYaMFptDqabp9QabaSKWQAD2CoRAx0i+AxcGxCRM1xAKTSi0y3w8vzyBUBIGKZQqNTqTlIk1klSQlSJmJmDSmlVIXF54yQ8k0sTqsAy5Dgmgg/PF5FIlQAqpqAJIAERqSBVCwAjqUPHc2OstoZ5Ag4GU9qBDmFEfgVPbHZcyeZbvdHjbLKUuFxHvsXXi3UGQ49SUkbuQ7g9GJsbabSvIFqEANogGL+QIgAC6Nvl5EV+n2V3MVhsdkuygC5jLFYcnNc7k8oFKGdCoGr6AauXIzCeOQZAPMPZCovFrXkKpbSpapGqTTYKuBlQQITYATXCzRB1kifkCjD/Z9g8ww9HtPH/0KNxPKnPcLF8IXlSXSK4zgQ8J1PadyMHQlQAKJ0MwYEAIz8KgMF0G8dB8EeSA4PIBgDsyN5TA4PilAgaC5uMaQANYQAAXkgcAwAEKbSAgRgQMgbjMLk8QgOQ5T6EWtJcC8mh7CApGaBR1G0fRY70o+TI7mwe5wAedTofIK6VJYdSlBAgRgXqzyvJuGqieJNF0WMDAsGhzDMAgpCyFhV44be+HcURqC5kg4x3iAyAksyhEOOqirFveMmMuYACyyh3JotTNFw5pwgAMnAC6YKUI6SJUkUAIIABowkllQAFpmfR2UAMqapFRVjJUoJlImsYyAZQnmMFZg2mwpQ4Jo8i6HcXaGE5frJg4O5KAggWoHw/GCXsc2tQ+EXoLIAmvJUfCVAAFDp5BVD1fUDUgdyVJKmjmhAACU2UAHzbQAvKMB2VEd/WDXUF1XaK5CaEYlSLLAlQADzvSdQ13aMUAUBuYG5XYsynXUFD/TCIx+BAYEw5lIHnXoP07WVkljA0NEHXA128q96rQrC/IIDg5Q6RAACEqbSOmmaFh5IDTpxIQ+K+8iPCWjFwCBjnxhY1i2PYCSNuSwEte2YydgY+ZxFL5KUukNLLeFk7oJrKNwssECyp+KrK6BlR6guspVDbNTSkimn8sw6pZnQTwbe1qACcqcA2no2jVBAZF5NrySpHr0l/KtBziibbDZWMziMCq1TwOQoqmEoI7bhL9XVK77SLFUmAas7mkivIntMIW82vGggc6TafoTHo0cJkmLU/AnRutTKAOKNDKqaZgtQ+JUZpjCLRmAUXkvN/7lAwSHNketcl7S2N/d0oPT4gEgNkLMwyO8poSIQPXvjKtTVTMLMKOKpUvVsJfVfbru+5NHUcA5iMABiDU+zBt5Qx0jDA66hV57HXpvZgChMSBE7hdHuFJY7UnjhOY+zhMpVFDi0O0FokCM0XvJRSyl1KKn/g0XkW8SG+wWq3PQ7dpA/mGmABYXBzBiHYHIRQygAQQHwJwjmIAuZZmImLXyxdGCVVQlWRW5hFEaSNIEZeLVQ4AHleEeVANwuQfCBEKCUIyURNs0SgFVm4BeBhORVxhpWXe5Ihx4WkARdyuYICoVkbISYUAd4jWlu4kcODZLmACQNOoW5nZfmgYqOwmiq5gUgtBSoAjxjumMJqSqAAhVCGw0xwDNNzHMjhnBOLSE8QJZgSyyJtnqJR/YVHoGaeonSTt5EOF0fo7MhieEmPWkIix+ArFbBsc4Ds9j9iOJ0DUqsTkwk+S8cRGQfjuqAJiRglZETE7RKCUveJi4KBJMtCuNJUEwJZJyUYPJhSjwlAaLs3C4SwpHyZHkXkCl3xzjaM7HaZ5Ki0zetzOwaNEly0CNdI8PVGYAH0FBNiWaEt5PkB64KZDgU6MB8DIpiPaWcS8CXozhFuHSuROjwpwEeJm18dImleS5D5WLzBVzYBqUurQ3ptXsCHKAFFSnMo8QbT57L7KkGYS3dALwq5HksC8DgyMNZuWIgAVjmgAdnVZIGCME5owQAMwCD1ZqyQAAmPgc0jXWskAANjtTBK1hqPhzQtQIO1Ag3WSAAJyOsNb6l1kgjVvE9XNdVWqA16pguq4NFrbXhr4q1P2exbVLUxZEoEkqFVKs4HcBFCwliqpmtmc1OqY2GpNWa91dr00OqdcGmCPqPVep9f6p1Qa61hrtZG6N+q412oTYIa1yb1osNQPWsVbLs2crpU0ZQRgS3eJAG8cgMEHBavXZuuAG7pBatIBusdfLzBat3fsoeHKpU2jXMoBFEAjDkCgMu9ZPhOJil4ce1NNYQwXuPleo8Ra2AIu3l6VF5J94MRTROg1rKs2GBzYGPNyMEVlgeAi2QI8MGQelf7DNh8Z0IbncUjYsjxHgfME0TgS70R3AQMNCgcA9HEVcTWWW9YFaYBRSAH8bZplq1md2XsIr3nTvg/zYl85FwwAVMuKuB4NzHN/kpf+R5BanjfCJjFBH4PqeFvVD8UnvwyfLMuP8pAAJxPkVcjJcE+AISQihNCGEtOuVLSJciVFyotSYixJomB2KHGkNxa+oVoMyo82JLzJMoOZsTpQv+CxeQYXUu7bSuk7Z8sXiZaL5lQJMFHIguyDkKPXhZXmNVvMvI+T8g4CAgVguMbC3Foe0VsBKninTYqqV0qZWYNlPKhVEr1WJnlqqNU6pwkarjA+370CdThb1D6yNhrYRw4xIR013L4bm/hlaQ9x2bW2ntN9b0lsQy+gTDwt1KgPUqM9fah1zufXxpdDwv00ZAxgKDcGn0ICQNsLDLGlQEYqElGdVGAN6aY2xk1PG313tE285nRob7KaPxBaQOmIw7T0pZuzYpnNSkZmkbzfmDg9NnhFimBphO5EgVW05KjSAaO+UUDqOjngYK0cQAxxUzGDG2i4+SNRiqulaKg9hWscs+MuAEx4VVjsMG62wXBxOJtRRmwdgWK2Eu9La7iBLl218NJ1Dvo3NgPs4GsKDiHa+q4I5R1K5gqkGQ/1fOTr4AsqdAaMAzlnHOecfAimdqXE36RK7V3kabuuDdpHW4Dmw4O6JmpdxcSEiDzVxpq4O0qFn/Jc5QInhLaeuc565He9/KzK9wtrw3tIMB29sNZ4Pvt4+YDz6XyIbfT2bCMfPzmL9LSOBP5nW/gllTSXAFjBAY3khAPoEblw/A+vJ8bLINQan9BzuVdu5z3gpqhCTfbzJuQ9SE/qHf1oQsehc/7TL5t+wnjJmKxcKGegfhIzzEiLES/pU6gbR5lnFXNPFKsfFNlaQ6kQCxMDltkjlq96oEkzloVI8bMbkAk7kHkikSkylSdcwgCalICdkxZWlhdzAOkxdNErE4xyRpcONKk5c7EFd9gU5lcsE98YCh4NcRhzZLYjMbZ9cLZvcjcw83Yzc49dIH9E9bcG97dw5I5aUd92D9YWtj5gUU404/cmAA8Z45cC5phQ8y4I9z9o9a5C5zd49a89g25k8Dgt9u5nd1tODj5MN88x4i9Tcp44AZ5y8F51IEDYsT1UAEEG9GF79HCW9Ai28mQO97gu8Tce974VQMgn4X4h934R8v5q5lNqFp9gFQY784AF8gcpCQi18kF5AUFlh7D09sJd8VCdNE58FXoiET9Ggz9x8cj/4aFFgb9oRCipCbDSw/9lQDAjF9EQBP9BFv81Bf9ZNRjACqkFlajll0U3MV02AIDWooDnc9l98mRDlYkAVo8kDTNklUCIJrlMkMDsB7kCkilpBnknQM8ig1j9jfQcAfkOBJNji8YgU1QscwUFgIUAYoVzjYVupCIcAkVpRglsI9jnDsVcV8VYTfAiVDM2hSV6YKV55qUoS6UeJGVoDVCmQr0XYeVDt2oBUhUmVdi3jESJU50E8T5Sh5UkNShlU7gX0PJy1dV9Uq1TVY1a0bU7V+0m0W1w1JBvU7UO1A1g1Q1JS+0nVY141E1R0E8p0STGTr1pBFUOT804BC0K5uSy1tU+TYNjVBTzV5TRTG0nUJS20ZT+0u0bUe0I0o1lTB13U1S+Av0J1NTGjL1ENtgF1jATTV1t191IyQAz090YzD1msgiYzz13jZ0dSQBb1MB71H1n0WCwCKt31f0/SIsP13dtTAMK4QMSEwNWNe5/RAi5tYMGS0zc19SUM0NDTXCl0Ii+4GyJ09tDZ/1EMcCSceZsxac6cpd2N5YGCZlmDQAFgbxuznRcR8BFzgFvQ95Iiyz0AuhVx2gKBphv4IAAg4BPjJ5S8YQpRDd0wLQIBsp3DAcYE4UaI7JAQVyjhP5EBCgaCY5XcGi5tDBGBzQdyQBwIvcAg3pXyKBtx7IuppAgEUhb4wzwxVzELSBkLWc1ttzQDS0I09VDV1Vj1ojzAnoDzmAEV04RgRhtBmBKADx9Dsp1V7tyKEVFBhhsoDUWLaLKBZAgx+tKhY1uLSAKKzELQpC+AbRaKcQjhpLfyytRUtT0AcUeVv5aKYRYSjxEQxj39JjTFRkf9tKbQbY39jEP99KZjyBLF5EACRzyl1kvy3yHBpKSwgAAAA==&value=N4IgTgpgLmCeIC4AMAaEATAhhAtgewDtEYBXCNANwjACM8BnASynmTQBs8BzREEDvAGNM7RKAgFMNdhHSIAZiPoQAvmhzRq9RAG1xk6bIVLyIEZwDu9ANaMADsfbK0jAlGoURiALQBGNJhcXFCMGj7+ZkHyjAAesq7uYJ6iCIpOpoIAFpgEBBBOuqAkJIxyCGbeNN6C3ujeEPwgpRIh0dS8vt5ICL4AdAAcvUggakUlZSDFpQBMjc1ujG1gvDMjALpodmB4UEJ4KSD0w2joEBSMgg3lAPSnFNdQLACqAMoAQsMqaypAAAA==&theme=bootstrap2&iconlib=fontawesome4&object_layout=normal&show_errors=interaction

andig, 2015-03-26:
Updated `gh-pages`. If you switch to that branch you can:

  - edit the schema in the editor itself
  - then update schema.json which is committed as well

This seems buggy:

[image]

Instead of being able to add a meter I need to make sure it's an array first. Problem with the schema or the editor?

andig, 2015-03-26:
Looks like a bug: https://github.com/jdorn/json-editor/issues/371

mbehr1, 2015-03-26:
strange because I tested using a page that should (tm) use the same editor. I'll take a look. 

Gruß
Matthias 

Sent from a mobile device.

mbehr1, 2015-03-26:
There seems to be a bug in the schema.
If I use this one it works (didn’t check the difference.)

I’ll post a PR where the schema file is contained.

{
    "$schema": "http://json-schema.org/draft-04/schema#",

    "definitions" : {

        "local": {
            "type": "object",
            "properties": {
                "enabled": {
                    "id": "/local/enabled",
                    "type": "boolean",
                    "description": "should we start the local HTTPd for serving live readings?"
                },
                "port": {
                    "id": "/local/port",
                    "type": "integer",
                    "default": 8080,
                    "description": "the TCP port for the local HTTPd"
                },
                "index": {
                    "id": "/local/index",
                    "type": "boolean",
                    "description": "should we provide a index listing of available channels if no UUID was requested?"
                },
                "timeout": {
                    "id": "/local/timeout",
                    "type": "integer"
                },
                "buffer": {
                    "id": "/local/buffer",
                    "type": "integer"
                }
            },
            "required": [ "enabled" ]
        },

        "channel": {
            "type": "object",
            "title": "channel",
            "properties": {
                "uuid": { "type": "string", "description": "uuid of this channel towards the middleware" },
                "identifier": { "type": "string", "description": "identifier of this channel from the meter. E.g. 1-0:1.8.0" },
                "api": {
                    "type": "string",
                    "enum": ["volkszaehler", "mysmartgrid", "null"],
                    "default": "volkszaehler",
                    "description": "middleware api to be used. Defaults to volkszaehler.org"
                },
                "aggmode": {
                    "type": "string",
                    "enum": ["avg", "max", "sum", "none"],
                    "description": "Mittelwert fuer Leistung, MAX fuer Zaehler, SUM fuer Counter",
                    "default": "none"
                },
                "duplicates": {
                    "type": "integer",
                    "minimum": 0,
                    "default": 0,
                    "description": "default 0 (send duplicate values), >0 = send duplicate values only each <duplicates> seconds. Activate only for abs. counter values (Zaehlerstaende) and not for impulses!"
                }
            },
            "required": ["uuid", "identifier"]
        },

        "meter": {
[... 104 lines cut ...]
      "description": "0 = log_error or log-warning, 5 = log_info, 10 = log-debug, 15 = log_finest"
    },
    "log": {
      "id": "/log",
      "type": "string",
      "description": "path to log file"
    },
    "local": { "$ref": "#/definitions/local" },
    "meters": { "$ref": "#/definitions/meters" }
  },
  "required": [
    "daemon",
    "log"
  ]
}

> Updated gh-pages. If you switch to that branch you can:
>
> edit the schema in the editor itself
> then update schema.json which is committed as well
> This seems buggy: [...]

Gruß

Matthias

andig, 2015-03-26:
I think the bug is in the editor with use of definitions...

dave-watts, 2015-04-08:
Nice idea to use the json-editor, If only the json-editor had the option to include help text in the schema for the html form elements that could help the user with configuration options

mbehr1, 2015-04-09:
I used the description field for help texts. But it would be really nice if some links or yaml code,… could be embedded in the description text. 

> 
> Nice idea to use the json-editor, If only the json-editor had the option to include help text in the schema for the html form elements that could help the user with configuration options

Gruß

Matthias

skre, 2015-04-11:
Looks nice. Could it be that setting "middleware" is missing?

mbehr1, 2015-04-11:
No, it’s there: add a Channel. The parameter is called „api“.

> 
> Looks nice. Could it be that setting "middleware" is missing?

Gruß

Matthias

skre, 2015-04-11:
Sorry, but I'm confused. In my old config I set
middleware: <path to my internet provider>

Now I can choose API between volkszaehler, mysmartgrid and none - without path..

mbehr1, 2015-04-11:
You’re right! I forgot the „middleware“. Will provide a patch.

> 
> Sorry, but I'm confused. In my old config I set
> middleware:
> 
> Now I can choose API between volkszaehler, mysmartgrid and none - without path..

Gruß

Matthias

mbehr1, 2015-04-11:
See PR #155.
Let me know if you miss any other parameters.

A „success“ message to have your config generated with the editor would be good as well ;-)

> 
> You’re right! I forgot the „middleware“. Will provide a patch.
> 
>> 
>> Sorry, but I'm confused. In my old config I set
>> middleware:
>> 
>> Now I can choose API between volkszaehler, mysmartgrid and none - without path..

Gruß

Matthias

skre, 2015-04-12:
I think "resolution"  (number of impulses per kWh, feature of the meter) is missing as well.

mbehr1, 2015-04-12:
Isn’t that „scaler“?
Can you send me an example config?

> 
> I think "resolution" (number of impulses per kWh, feature of the meter) is missing as well.

Gruß

Matthias

skre, 2015-04-13:
Hi Matthias,
I'm refering to the config file on this wiki page:
http://wiki.volkszaehler.org/hardware/controllers/ferrariszaehler_lesekopf

Best regards
Stefan

skre, 2015-04-14:
Hi Matthias,
is resolution the same as scale? I would like to update the wiki page - if so.

mbehr1, 2015-04-14:
Hi,

after greping the code I found „resolution“. It’s not an option for the api/middleware but for the MeterS0.

I’ll add it to the MeterS0 schema..
Thanks!

> 
> Hi Matthias,
> is resolution the same as scale? I would like to update the wiki page - if so.

Gruß

Matthias

mbehr1, 2015-04-14:
I updated PR #155.

> 
> Hi,
> 
> after greping the code I found „resolution“. It’s not an option for the api/middleware but for the MeterS0.
> 
> I’ll add it to the MeterS0 schema..
> Thanks!
> 
> 
>> 
>> Hi Matthias,
>> is resolution the same as scale? I would like to update the wiki page - if so.
>> 
>> 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

andig, 2015-04-15:
I'd suggest to commit the obvious schema fixes right on master or pull immediately.

mbehr1, 2015-06-11:
issue can be closed, or?

closed 2015-06-11 by andig