volkszaehler/vzlogger #144: vzlogger should print start message to logfile if loglevel > 0

andig, 2015-03-15:
Do we have a definition of the log levels anywhere?

mbehr1, 2015-03-19:
I'm not aware of any definition. The code is defined as:
```
typedef enum {
	log_error = -1,
	log_warning = 0,
	log_info = 5,
	log_debug = 10,
	log_finest = 15
} log_level_t;
```
I suggest some definition like e.g. here:
http://projects.genivi.org/diagnostic-log-trace/sites/projects.genivi.org.diagnostic-log-trace/files/dlt_cheatsheet.html 

"...DLT_LOG_ERROR and DLT_LOG_WARN should be used in your application, when something is going wrong. DLT_LOG_INFO should be used to send the most important information. DLT_LOG_DEBUG and DLT_LOG_VERBOSE should be only used for testing information."
(so VERBOSE would be like our "log_finest")

mbehr1, 2015-03-19:
Shall I send the output from e.g. vzlogger -V to log file on startup?

andig, 2015-03-20:
> Shall I send the output from e.g. vzlogger -V to log file on startup?

I guess only for log_info. Maybe for now we should just add the numbers to the vzlogger.conf as a comment?

mbehr1, 2015-03-20:
I suggest to always print the version together with the info which log level is set. This makes reading logs easier as you can understand which log level was set.
And adding the values/symbols to the conf file would be fine.

mbehr1, 2015-03-20:
see PR #147.

closed 2015-03-21 by andig