volkszaehler/vzlogger #145: documentation needs update to mention microhttpd install

dave-watts, 2015-03-19:
Documentation for install is v.good, just missing microhttpd dependancy,

 sudo apt-get install libmicrohttpd-dev

 I would also make note to copy files to /usr/local/lib as it seems to be a more usual location

 sudo cp -R sml/include/* /usr/local/include/

Cheers

Dave

andig, 2015-03-19:
@dave-watts thankd for your PR. Are we missing your changes in the install script or in the wiki? If the former, could you prepare a PR for us?

dave-watts, 2015-03-19:
Your welcome, I only followed the wikki install instructions so I'm not really sure about the script

andig, 2015-03-19:
You can update the wiki yoursel, please feel free to add the missing install instructions!

dave-watts, 2015-03-19:
Can't get past the Captcha to register!

andig, 2015-03-20:
The httpd lib is documented here as optional http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version

working for me

closed 2015-03-24 by andig