volkszaehler/vzlogger #146: timestamps should be integers when sending to middleware

andig, 2015-03-19:
From the ml:

> [ [ 1426414635584.1191, 1658272.1000000001 ] ]
> vzlogger sendet die timestamps als float? nicht soo sinnvoll...

mbehr1, 2015-03-19:
That's easy to fix. 
```
		json_object_array_add(json_tuple, json_object_new_double(value));
```
can be changed to use int64.
I'm just worried that somebody parsing the output expects a json type double and can't handle int64.

andig, 2015-03-19:
Shouldnt happen. An int64 is still a float imho. 

Viele Grüße,
Andreas

mbehr1, 2015-03-22:
Does anybody mind if I change the internal storage of timestamps to int64_t as well? Seems like nobody really needs the doubles and the int->double->int conversions.
Btw: does anybody use the mysmartgrid api? Seems like this uses int for timestamps as well (but rounded to seconds).
This would mean changing the following parts of class Reading:
```
	double tvtod() const;
	double tvtod(struct timeval const &tv) const;
	void time() { gettimeofday(&_time, NULL); }
	void time(struct timeval const &v) { _time = v; }
	struct timeval dtotv(double const &ts) const;
```
and all the users of it.

andig, 2015-03-22:
Dont think anybody cares about the actual implementation details.

mbehr1, 2015-03-22:
see PR #148.

mbehr1, 2015-04-11:
can be closed, right?

andig, 2015-04-11:
Of course, thank you!

closed 2015-04-11 by andig