volkszaehler/vzlogger #156: dlms with arduino

Totte1210, 2015-04-17:
Hello,

has anyone ported code for arduino to read dlms protocol, yet?

I would like to use IR-TTL to read my Landis Gyr ZMD120.

Your help will be appreciated!

best regards

Thorsten

mbehr1, 2015-04-17:
which kind of arduino? running linux or some 8bit/atmel version?

> 
> Hello,
> 
> has anyone ported code for arduino to read dlms protocol, yet?
> 
> I would like to use IR-TTL to read my Landis Gyr ZMD120.
> 
> Your help will be appreciated!
> 
> best regards
> 
> Thorsten

Gruß

Matthias

Totte1210, 2015-04-17:
Es ist ein Arduino Nano, also Atmel 8bit.

Gruß

Thorsten

mbehr1, 2015-04-18:
Hi,

der hat max. 2KB RAM? Das wird mehr als eng. Warum nimmst du nicht einen RPI Model A+ (~25€), das wäre deutlich einfacher.

> 
> Es ist ein Arduino Nano, also Atmel 8bit.
> 
> Gruß
> 
> Thorsten

Gruß

Matthias

Totte1210, 2015-04-18:
Hallo Matthias,

habe aktuell nur nen Arudino Nano rumliegen. Besorge mir jetzt wohl aber noch nen zweiten Pi.
Habe es dennoch geschafft den Landis erfolgreich auszulesen:

Mein Code ist folgender:

#include <SoftwareSerial.h>

#define irRxPin 7
#define irTxPin 6
SoftwareSerial ir = SoftwareSerial(irRxPin,irTxPin); // RX, TX
const unsigned int _requestIrData = 5000;
unsigned long _lastRequestIrData = 0;

void RequestMeterData(unsigned long currentMillis){
  if (currentMillis - _lastRequestIrData > _requestIrData){
    // save the last time you requested ir data
    _lastRequestIrData = currentMillis;
   
    Serial.println("Requesting IR-data...");
    
    ir.print("/?!\r\n");
  }
}
void ReceiveMeterData(){ 
  if (ir.available()){
    Serial.print("Response: ");
    char response = ir.read();
    // convert 8N1 to 7E1
    response &= 0x7F;
    Serial.println(response);
  }
}
void setup()
{
  ir.begin(300);
  ir.listen();
  Serial.begin(57600);
}
void loop()
{
  unsigned long timeMillis = millis();
  RequestMeterData(timeMillis);
  ReceiveMeterData();
}

mbehr1, 2015-04-18:
Ja das auslesen ist so kein Problem. Aber was machst du mit den Daten dann? Wenn du die in Datenbank auf Webserver,… packen willst, brauchst du http/mysql,… Lib und dann wird es mit 2kb eng.

> 
> Hallo Matthias,
> 
> habe aktuell nur nen Arudino Nano rumliegen. Besorge mir jetzt wohl aber noch nen zweiten Pi.
> Habe es dennoch geschafft den Landis erfolgreich auszulesen:
> 
> Mein Code ist folgender:
> 
> #include
> 
> #define irRxPin 7
> #define irTxPin 6
> SoftwareSerial ir = SoftwareSerial(irRxPin,irTxPin); // RX, TX
> const unsigned int _requestIrData = 5000;
> unsigned long _lastRequestIrData = 0;
> 
> void RequestMeterData(unsigned long currentMillis){
> if (currentMillis - _lastRequestIrData > _requestIrData){
> // save the last time you requested ir data
> _lastRequestIrData = currentMillis;
> 
> Serial.println("Requesting IR-data...");
> 
> ir.print("/?!\r\n");
> }
> }
> void ReceiveMeterData(){ 
> if (ir.available()){
> Serial.print("Response: ");
> char response = ir.read();
> // convert 8N1 to 7E1
> response &= 0x7F;
> Serial.println(response);
> }
> }
> void setup()
> {
> ir.begin(300);
> ir.listen();
> Serial.begin(57600);
> }
> void loop()
> {
> unsigned long timeMillis = millis();
> RequestMeterData(timeMillis);
> ReceiveMeterData();
> }

Gruß

Matthias

Totte1210, 2015-04-18:
Ich habe den Datenstrom jetzt erstmal auf einen Float Wert nur für den aktuellen Zählerstand reduziert.
Ich muss mal schauen, ob ich mir ein Ethernetshield besorge, um dann per Webserver den aktuellen Wert meiner FHEM Installation bereitzustellen, oder ob ich nicht direkt einen Pi auf Hutschiene in den Schaltschrank setzte. Ist wahrscheinlich die beste Lösung.
Vielen Dank auf jeden Fall.

beste Grüße

Thorsten

justinotherguy, 2015-04-19:
Hi,
wie Matthias schon schreibt: was ist Dein Ziel? Wenn Du da keine Schnittstelle am Arduino hast - wo sollen die Daten hingehen? Wenn Du eine hast: für die YPort-Funktion ("seriell to (W)LAN") reicht der Arduino mit seinen 2 KB allemal; dann kannst Du das Parsen auf ner Linux-Box machen (vzlogger).
Den Parser auf dem Arduino zu schreiben halte ich auch für vertane Zeit (wenngleich eine spannende Aufgabe und eine tolle Idee).
Ich habe damals nicht die Arduino-Umgebung, sondern Ethersex (http://www.ethersex.de/) auf einem ATmega-Board (http://www.pollin.de/shop/dt/NjI5OTgxOTk-/Bausaetze_Module/Entwicklerboards/AVR_NET_IO_Fertigmodul.html) verwendet; das kommt schon mit der LAN-Schnittstelle daher.
Halte uns auf dem Laufenden :-)
Gruß, J.

Totte1210, 2015-04-20:
Hallo Justin,
ich habe mich nun wie folgt festgelegt:
- ein Arduino Nano nimmt die Werte des Stromzählers mitttels IR-TTL Modul entgegen.
- ein zweiter Sensor geht mit dran (Read-Kontakt), der den Gaszähler erfasst
- als Übertragungsschnittstelle für den Nano habe ich ein mini Ethernet Modul (ENC28J60 => SPI-Kommunikation) für 5€ besorgt.
- der Nano macht dann einen Webserver auf, der die Messwerte bereitstellt
- auf einem Pi (im Büro) habe FHEM laufen. Dies kann per modul HTTPMOD die Daten holen, speichern, auswerten und grafisch darstellen

Somit habe ich eine Datenerfassung und Bereitstellung für wenig Geld (nano+Ethernet bei *Bay für <15€) und sehr kleinem Stromverbrauch.

justinotherguy, 2015-04-20:
Hi,
das klingt gut - ich würde einem HTTP-Client auf dem Arduino eher trauen als einem HTTP-Server (ja, ich weiß - das kann der auch - trotzdem). Ich würde eher den Arduino den Web-Server auf dem FHEM ansprechen lassen.
Abgesehen davon: das Parsen musst Du dann immer noch selbst übernehmen.
Ich fänd's toll, wenn Du Dein Ergebnis dokumentierst (in unserem Wiki?) und den Code als PR für unser Repo bereit stellst.
toi-toi-toi :-)
Gruß, J.
P.S.: können wir den issue hier schließen?

Totte1210, 2015-04-20:
Hi,

Danke für den Tipp.
Ich denke nach dem kommenden Wochenende bin ich weiter und schlauer.
Ich halte euch auf dem Laufenden und werde dokumentieren, wenn alles getestet ist.

Den Issue kannst du gerne schließen.

Beste Grüße

Thorsten

closed 2015-04-21 by justinotherguy