volkszaehler/vzlogger #161: Problem with calculating Values with meter type sml and inverval setting

griesi, 2015-05-01:
I have "sml" meters. I don't want to send a measurement every time the meter sends one so i configured a interval of 10 seconds. The vzlogger then sends a measurement every 10 seconds to the middleware... everything seems fine ...

The Problem now is that the shown value in the frontent is too low (i calculated a factor of 6) compared to the real time reading on the meter. 

I think this factor is somewhat in the interval the meter sends values to vzlogger (something around 2 seconds) and the configured interval of 10. With an interval of 20 seconds the factor is 12. 

With no interval set in the configuration of vzlogger the values are correct but the values are sent about every 2 seconds to the Middleware ...

griesi, 2015-05-01:
Logfile is here:
<pre>
[May 01 17:50:22][main] vzlogger v0.4.0 based on heads/master-0-g19eb2d0ebf from Sun, 19 Apr 2015 08:29:22 +0200 started.
[May 01 17:50:22][main] log level is 15
[May 01 17:50:22][mtr0] Creating new meter with protocol sml.
[May 01 17:50:22][mtr0] Meter configured, enabled.
[May 01 17:50:22]       New meter initialized (protocol=sml)
[May 01 17:50:22]       Configure channel.
[May 01 17:50:22][chn0] New channel initialized (uuid=...77fc12 api=volkszaehler id=1-0:1.8.0)
[May 01 17:50:22]       Configure channel.
[May 01 17:50:22][chn1] New channel initialized (uuid=...9861ad api=volkszaehler id=1-0:2.8.0)
[May 01 17:50:22][mtr1] Creating new meter with protocol sml.
[May 01 17:50:22][mtr1] Meter configured, enabled.
[May 01 17:50:22]       New meter initialized (protocol=sml)
[May 01 17:50:22]       Configure channel.
[May 01 17:50:22][chn2] New channel initialized (uuid=...439864 api=volkszaehler id=1-0:2.8.1)
[May 01 17:50:22]       Configure channel.
[May 01 17:50:22][chn3] New channel initialized (uuid=...84b729 api=volkszaehler id=1-0:15.7.0)
[May 01 17:50:22]       Have 2 meters.
[May 01 17:50:22][main] daemon=0, local=0
[May 01 17:50:22]       Process not  daemonized...
[May 01 17:50:22]       Opened logfile /var/log/vzlogger.log
[May 01 17:50:22][]     ===> Start meters
[May 01 17:50:22][mtr0] Meter connection established
[May 01 17:50:22][mtr0] Meter thread started
[May 01 17:50:22][mtr0] Meter is opened. Starting channels.
[May 01 17:50:22][chn0] Logging thread started
[May 01 17:50:22][mtr0] Number of readers: 32
[May 01 17:50:22][chn1] Logging thread started
[May 01 17:50:22][chn0] Start logging thread for volkszaehler-api. Running as daemon: no
[May 01 17:50:22][mtr0] Config.daemon: 0
[May 01 17:50:22][mtr0] Config.local: 0
[May 01 17:50:22][chn1] Start logging thread for volkszaehler-api. Running as daemon: no
[May 01 17:50:22][chn0] Using default volkszaehler api.
[May 01 17:50:22][chn1] Using default volkszaehler api.
[May 01 17:50:22][mtr1] Meter connection established
[May 01 17:50:22][mtr1] Meter thread started
[May 01 17:50:22][mtr1] Meter is opened. Starting channels.
[May 01 17:50:22][chn2] Logging thread started
[May 01 17:50:22][mtr1] Number of readers: 32
[May 01 17:50:22][chn3] Logging thread started
[May 01 17:50:22][chn3] Start logging thread for volkszaehler-api. Running as daemon: no
[May 01 17:50:22][chn2] Start logging thread for volkszaehler-api. Running as daemon: no
[May 01 17:50:22][mtr1] Config.daemon: 0
[May 01 17:50:22][mtr1] Config.local: 0
[May 01 17:50:22][chn3] Using default volkszaehler api.
[May 01 17:50:22][chn2] Using default volkszaehler api.
[May 01 17:50:22][]     Startup done.
[May 01 17:50:23][mtr1] Got 3 new readings from meter:
[May 01 17:50:23][mtr1] Reading: id=1-0:2.8.0*255/ObisItentifier:1-0:2.8.0*255 value=13908967.50 ts=1430495423493
[May 01 17:50:23][mtr1] Reading: id=1-0:2.8.1*255/ObisItentifier:1-0:2.8.1*255 value=13908967.50 ts=1430495423493
[May 01 17:50:23][mtr1] Reading: id=1-0:15.7.0*255/ObisItentifier:1-0:15.7.0*255 value=261.00 ts=1430495423493
[May 01 17:50:23][chn2] Adding reading to queue (value=13908967.50 ts=1430495423493)
[May 01 17:50:23][chn3] Adding reading to queue (value=261.00 ts=1430495423493)
[May 01 17:50:23][chn2] Buffer dump (size=1): {13908967.5000,}
[May 01 17:50:23][chn3] Buffer dump (size=1): {261.0000,}
[May 01 17:50:23][mtr1] Next reading in 10 seconds
[May 01 17:50:23][chn3] ==> number of tuples: 1
[May 01 17:50:23][chn2] ==> number of tuples: 1
[May 01 17:50:23][chn3] compare: 0 1430495423493
[May 01 17:50:23][chn2] compare: 0 1430495423493
[May 01 17:50:23][chn2] JSON request body: [ [ 1430495423493, 13908967.5 ] ]
[May 01 17:50:23][chn2] CURL: About to connect() to localhost port 80 (#0)
[May 01 17:50:23][chn2] CURL:   Trying 127.0.0.1...
[May 01 17:50:23][chn2] CURL: connected
[May 01 17:50:23][chn2] CURL: Connected to localhost (127.0.0.1) port 80 (#0)
[May 01 17:50:23][chn2] CURL: Sent 33 bytes.. 
[May 01 17:50:23][chn2] CURL: Sent '[ [ 1430495423493, 13908967.5 ] ]' bytes
[May 01 17:50:23][chn2] CURL: upload completely sent off: 33 out of 33 bytes
[May 01 17:50:23][chn2] CURL: additional stuff not fine transfer.c:1037: 0 0
[... 774 lines cut ...]
[May 01 17:51:21][chn0] CURL: Sent '[ [ 1430495481481, 8939346.5999999996 ] ]' bytes
[May 01 17:51:21][chn0] CURL: upload completely sent off: 41 out of 41 bytes
[May 01 17:51:21][chn0] CURL: additional stuff not fine transfer.c:1037: 0 0
[May 01 17:51:21][chn0] CURL: HTTP 1.0, assume close after body
[May 01 17:51:21][chn0] CURL: Received 26 bytes
[May 01 17:51:21][chn0] CURL: Received '{"version":"0.3","rows":1}' bytes
[May 01 17:51:21][chn0] CURL: Closing connection #0
[May 01 17:51:21][chn0] CURL Request succeeded with code: 200
[May 01 17:51:21][chn1] JSON request body: [ [ 1430495481481, 10591953.800000001 ] ]
[May 01 17:51:21][chn1] CURL: About to connect() to localhost port 80 (#0)
[May 01 17:51:21][chn1] CURL:   Trying 127.0.0.1...
[May 01 17:51:21][chn1] CURL: connected
[May 01 17:51:21][chn1] CURL: Connected to localhost (127.0.0.1) port 80 (#0)
[May 01 17:51:21][chn1] CURL: Sent 41 bytes.. 
[May 01 17:51:21][chn1] CURL: Sent '[ [ 1430495481481, 10591953.800000001 ] ]' bytes
[May 01 17:51:21][chn1] CURL: upload completely sent off: 41 out of 41 bytes
[May 01 17:51:21][chn1] CURL: additional stuff not fine transfer.c:1037: 0 0
[May 01 17:51:21][chn1] CURL: HTTP 1.0, assume close after body
[May 01 17:51:21][chn1] CURL: Received 26 bytes
[May 01 17:51:21][chn1] CURL: Received '{"version":"0.3","rows":1}' bytes
[May 01 17:51:21][chn1] CURL: Closing connection #0
[May 01 17:51:21][chn1] CURL Request succeeded with code: 200
[May 01 17:51:23]       terminating on signal 2.
[May 01 17:51:23]       Closing connections to terminate
</pre>

mbehr1, 2015-05-01:
pls send your conf file.

> I have "sml" meters. I don't want to send a measurement every time the meter sends one so i configured a interval of 10 seconds. The vzlogger then sends a measurement every 10 seconds to the [...]

Gruß

Matthias

griesi, 2015-05-01:
Here is the config-file
<pre>
/**
 * vzlogger configuration
 *
 * use proper encoded JSON with javascript comments
 *
 * take a look at the wiki for detailed information:
 * http://wiki.volkszaehler.org/software/controller/vzlogger#configuration
 */

{
"retry" : 30,                   /* how long to sleep between failed requests, in seconds */
"daemon": false,                /* run periodically */
"verbosity" : 15,               /* between 0 and 15 */
"log" : "/var/log/vzlogger.log",/* path to logfile, optional */

"local" : {
        "enabled" : false,      /* should we start the local HTTPd for serving live readings? */
        "port" : 80,            /* the TCP port for the local HTTPd */
        "index" : true,         /* should we provide a index listing of available channels? */
        "timeout" : 30,         /* timeout for long polling comet requests, 0 disables comet, in seconds */
        "buffer" : -1          /* how long to buffer readings for the local interface, in seconds */
},

"meters" : [{
        "enabled" : true,       /* disabled meters will be ignored */
        "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
        "device" : "/dev/eHZ1",
        /* "interval" : 10, */
                "channels": [{
                        "uuid" : "05b82ae0-f00d-11e4-a9ff-e79e5977fc12",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:1.8.0",
                }, {
                        "uuid" : "2606f940-f00d-11e4-8acb-e5fd329861ad",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:2.8.0",
                }]
        },{
        "enabled" : true,       /* disabled meters will be ignored */
        "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
        "device" : "/dev/eHZ2",
        "interval" : 10,
                "channels": [{
                        "uuid" : "4c51aff0-f00d-11e4-8449-db537e439864",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:2.8.1",
                }, {
                        "uuid" : "a58247a0-f013-11e4-833c-e1c23184b729",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:15.7.0",
                }]
        }
]}
</pre>

mbehr1, 2015-05-01:
Und wie sind die 4 Kanäle im Frontend konfiguriert?
Tritt das Verhalten bei allen 4 auf?

> 
> Here is the config-file
> 
> /**
>  * vzlogger configuration
>  *
>  * use proper encoded JSON with javascript comments
>  *
>  * take a look at the wiki for detailed information:
>  * http://wiki.volkszaehler.org/software/controller/vzlogger#configuration
>  */
> 
> {
> "retry" : 30,                   /* how long to sleep between failed requests, in seconds */
> "daemon": false,                /* run periodically */
> "verbosity" : 15,               /* between 0 and 15 */
> "log" : "/var/log/vzlogger.log",/* path to logfile, optional */
> 
> "local" : {
>         "enabled" : false,      /* should we start the local HTTPd for serving live readings? */
>         "port" : 80,            /* the TCP port for the local HTTPd */
>         "index" : true,         /* should we provide a index listing of available channels? */
>         "timeout" : 30,         /* timeout for long polling comet requests, 0 disables comet, in seconds */
>         "buffer" : -1          /* how long to buffer readings for the local interface, in seconds */
> },
> 
> "meters" : [{
>         "enabled" : true,       /* disabled meters will be ignored */
>         "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
>         "device" : "/dev/eHZ1",
>         /* "interval" : 10, */
>                 "channels": [{
>                         "uuid" : "05b82ae0-f00d-11e4-a9ff-e79e5977fc12",
>                         "middleware" : "http://localhost/volkszaehler.org/middleware.php",
>                         "identifier" : "1-0:1.8.0",
>                 }, {
>                         "uuid" : "2606f940-f00d-11e4-8acb-e5fd329861ad",
>                         "middleware" : "http://localhost/volkszaehler.org/middleware.php",
>                         "identifier" : "1-0:2.8.0",
>                 }]
>         },{
>         "enabled" : true,       /* disabled meters will be ignored */
>         "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
>         "device" : "/dev/eHZ2",
>         "interval" : 10,
>                 "channels": [{
>                         "uuid" : "4c51aff0-f00d-11e4-8449-db537e439864",
>                         "middleware" : "http://localhost/volkszaehler.org/middleware.php",
>                         "identifier" : "1-0:2.8.1",
>                 }, {
>                         "uuid" : "a58247a0-f013-11e4-833c-e1c23184b729",
>                         "middleware" : "http://localhost/volkszaehler.org/middleware.php",
>                         "identifier" : "1-0:15.7.0",
>                 }]
>         }
> ]}

Gruß

Matthias

griesi, 2015-05-01:
Die Kanäle "1-0:x.8.0" sind als "El. Energie (Zählerstände)" konfiguriert, der Kanal "15.7.0" ist als "El. Energier (Leistungswerte)" konfiguriert.

Der erste Zähler an "/dev/eHZ1" ist ohne Interval konfiguriert und hat das Problem auch nicht. Die Daten werden bei jedem "reading" an die Middleware geschickt und korrekt dargestellt.

Das Verhalten tritt nur beim zweiten Zähler bzw. beim Kanal "2.8.0" auf, wo das Interval auf "10" steht. Aus diesem Grund habe ich auch den Kanal "15.7.0" mit aufgenommen, um die wirkliche Leistung als Vergleich zu haben. Die Werte müssten ja im Prinzip gleich sein, sind aber nur ca. 1/6.

mbehr1, 2015-05-01:
Aber die Einheit vom 15.7.0 ist „W“ wohingegen die vom 2.8.0 doch Wh ist, oder? Die können doch nicht gleich sein.

> Die Kanäle "1-0:x.8.0" sind als "El. Energie (Zählerstände)" konfiguriert, der Kanal "15.7.0" ist als "El. Energier (Leistungswerte)" konfiguriert.
>
> Der erste Zähler an "/dev/eHZ1" ist ohne Interval [...]

Gruß

Matthias

griesi, 2015-05-01:
Du hast schon recht, das Frontend rechnet aber aus dem Zählerstand (Wh) und der Zeit die Leistung (W) aus, um eine sinnvolle Kurve darstellen zu können. 

Würde man den Zählerstand (Wh) darstellen bekäme man nur eine (relativ) konstant nach oben verlaufende Kurve, weil dieser ja stetig wächst.

Anbei ein Bild, auf dem man sieht, dass die Werte gleich sind (ohne Interval-Einstellung) im vzlogger.

[image]
 
Vielleicht liegt das Problem auch im Frontend. Dann ist aber nicht klar, warum die Einstellung "Interval" im vzlogger dieses Verhalten auslöst. Ich vermute die Ursache doch eher im vzlogger.

andig, 2015-06-06:
Ich grad durch Zufall diese ältere Issue gefunden. Tritt das problem noch auf?

> Die Kanäle "1-0:x.8.0" sind als "El. Energie (Zählerstände)" konfiguriert, der Kanal "15.7.0" ist als "El. Energier (Leistungswerte)" konfiguriert.

Das ist erstmal korrekt. Betrachten wir mal nur 2.8.0 und 15.7.0. Die beiden Kanäle sollten ja- in Leistung umgerechnet- aus das Gleiche anzeigen.

> Das Verhalten tritt nur beim zweiten Zähler bzw. beim Kanal "2.8.0" auf, wo das Interval auf "10" steht. Aus diesem Grund habe ich auch den Kanal "15.7.0" mit aufgenommen, um die wirkliche Leistung als Vergleich zu haben. Die Werte müssten ja im Prinzip gleich sein, sind aber nur ca. 1/6.

Genau. In Deinem Screenshot zeigst Du auch zwei Kanäle 2.8.0 und vmtl 15.7.0 (?) die genau das Verhalten zeigen. Wo siehst Du also den Fehler?

>> Aber die Einheit vom 15.7.0 ist „W“ wohingegen die vom 2.8.0 doch Wh ist, oder? Die können doch nicht gleich sein.

> Du hast schon recht, das Frontend rechnet aber aus dem Zählerstand (Wh) und der Zeit die Leistung (W) aus, um eine sinnvolle Kurve darstellen zu können. 

Stimmt Beides!

Was mich an der Konfig wundert: wenn vzlogger Zählerstände auslässt (=interval) dann muss er mit den anderen ja etwas tun. Ich dachte immer (Achtung Laie!) dafür wäre `aggmode`- in Deinem Fall also `MAX`- zusammen mit `aggtime` zuständig. Was tut `interval` an dieser Stelle?

griesi, 2015-06-08:
Danke für den Kommentar. 
> Was mich an der Konfig wundert: wenn vzlogger Zählerstände auslässt (=interval) dann muss er mit den anderen ja etwas tun. Ich dachte immer (Achtung Laie!) dafür wäre  aggmode - in Deinem Fall also  MAX - zusammen mit  aggtime  zuständig. Was tut  interval  an dieser Stelle?

Bei mir handelt es sich aber um eHZ mit SML, die ca. alle 2 Sekunden für die Zähler x.8.0 ABSOLUTE Werte liefern. Somit sind die Zählerstände zwischen dem Intervall nicht relevant. 

Der Grund warum die Intervall-Einstellung für ich interessant ist, ist der dass ich nicht alle zwei Sekunden einen Messwert zum Backend schicken möchte. Ich benötige die Daten nicht in dieser Auflösung und die Datenbank wird dadurch auch relativ schnell groß.

Ich habe mich mit Aggmode und Aggtime noch nicht ausseinandergesetzt, da ich dachte das wäre für S0-Zähler gemacht (ist in der Referenz-Konfig zumindest dort aufgeführt). Vermutlich liegt hier der Fehler ...

Ich teste das jetzt mal mit neuer Config:
<pre>
        "enabled" : true,       /* disabled meters will be ignored */
        "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
        "device" : "/dev/eHZ2",
        "aggtime" : 10,
                "channels": [{
                        "uuid" : "4c51aff0-f00d-11e4-8449-db537e439864",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:2.8.1",
                        "aggmode"    : "MAX",
                }, {
                        "uuid" : "a58247a0-f013-11e4-833c-e1c23184b729",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:15.7.0",
                        "aggmode"    : "AVG",
                }]
</pre>

andig, 2015-06-10:
>>    Was mich an der Konfig wundert: wenn vzlogger Zählerstände auslässt (=interval) dann muss er mit den anderen ja etwas tun. Ich dachte immer (Achtung Laie!) dafür wäre aggmode - in Deinem Fall also MAX - zusammen mit aggtime zuständig. Was tut interval an dieser Stelle?

> Bei mir handelt es sich aber um eHZ mit SML, die ca. alle 2 Sekunden für die Zähler x.8.0 ABSOLUTE Werte liefern. Somit sind die Zählerstände zwischen dem Intervall nicht relevant. 

Physikalisch gebe ich Dir recht, ich habe aber keine Ahnung was vzlogger bei `interval` tut (ping @mbehr1)? Stimmen denn die Werte die Du für 1.8.0 im Logfile als diejenigen siehst die an curl übergeben werden (btw: endlose logs bitte auf pastebin).

> Der Grund warum die Intervall-Einstellung für ich interessant ist, ist der dass ich nicht alle zwei Sekunden einen Messwert zum Backend schicken möchte. Ich benötige die Daten nicht in dieser [...]

Ja, das geht mit `aggmode/interval` auch. Deshalb bin ich gerade lost was `interval` eigentlich tut und wofür wir das brauchen...

r00t-, 2015-06-10:
interval tut einfach das hier:
 ./src/threads.cpp:	sleep(mtr->interval());
interval = hartes delay zwischen meter_read aufrufen.

das wird fuer meter gebraucht, die nicht von sich aus auf einen wert warten muessen,
sondern diesen beliebig oft liefern koennen. (zB das random-meter, auslesen einer datei,
vlt. ein smartmeter dem fuer jede auslesung eine pull-sequenz geschicht wird).
bei allen anderen macht interval keinen sinn, und sollte nie gesetzt sein.
(es gibt in den meter-definitionen den eintrag "periodic", der angibt ob ein interval
 erforderlich ist.)

insbesondere waehre es bei einem s0-meter dumm, zwischen den impulsen zu warten.
(oder, vlt will das jemand zum "entprellen" misbrauchen ;) )
bei einem smartmeter das von sich aus daten schickt, ist es schlecht zu warten,
weil sich empfangene daten im puffer des betriebssystems sammeln,
bis dieser ueberlaeuft.
das koennte in einzelnfaellen funktionieren, gut ist es aber nicht.
(man sollte zumindest nach dem delay den empfangspuffer leeren,
 um ein aktuelles datentelegramm abzuwarten, anstatt ein am anfang
 der verzoegerungszeit gepuffertes.)

andig, 2015-06-11:
> interval tut ... hartes delay zwischen meter_read aufrufen.

> das wird fuer meter gebraucht, die nicht von sich aus auf einen wert warten muessen,

Fazit: Konfigurationsfehler bei @griesi.

> insbesondere waehre es bei einem s0-meter dumm,  zwischen den impulsen zu warten.

Lustigerweise ist mir beim Split der mySmartGrid Konfiguration gerade aufgefallen dass genau das so in den Beispielen drin ist: https://github.com/volkszaehler/vzlogger/blob/master/etc/vzlogger.conf#L164
Ist da kaum verwendet sicher zu verschmerzen, aber schon seltsam...

griesi, 2015-06-22:
so funktioniert das jetzt:

<pre>
        "enabled" : true,       /* disabled meters will be ignored */
        "protocol" : "sml",     /* use 'vzlogger -h' for list of available protocols */
        "device" : "/dev/eHZ2",
        "aggtime": 10,
                "channels": [{
                        "uuid" : "4c51aff0-f00d-11e4-8449-db537e439864",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:2.8.1",
                        "aggmode": "MAX"
                }, {
                        "uuid" : "a58247a0-f013-11e4-833c-e1c23184b729",
                        "middleware" : "http://localhost/volkszaehler.org/middleware.php",
                        "identifier" : "1-0:15.7.0",
                        "aggmode": "AVG"
                }]

</pre>

@andig : Bitte bei Gelegenheit in der "Referenz-Dokumentation" der vzlogger.conf noch 
"aggtime" und "aggmode" mit bei einem "SML"-Zähler aufnehmen. 

Ich denke das könnte für den ein oder anderen trotzdem interessant sein !!!

closed 2015-06-22 by griesi

andig, 2015-06-22:
> Bitte bei Gelegenheit in der "Referenz-Dokumentation" der vzlogger.conf noch
"aggtime" und "aggmode" mit bei einem "SML"-Zähler aufnehmen. 

@griesi Was meinst Du- das Wiki? Freiwillige vor...