volkszaehler/vzlogger #176: Neuer Typ Zähler

UdoSchake, 2015-07-05:
In Österreich ( und wahrscheinlich auch in naher Zukunft in Deutschland) gibt es einen neuen Typ Zähler der über die IR-Schnittstelle die Daten im M-Bus-Format mit Verschlüsselung ausgibt (AMIS-Zähler).
http://strom.netzgmbh.at/eag_at/page/284094835607631929_0_1060376991645160446,de.html
Es wäre vorteilhaft, wenn vzlogger auch solche Zähler auslesen könnte.

Gruß
Udo

andig, 2015-07-05:
Hi Udo,

Ist das "dieser" Mbus: https://github.com/rscada/libmbus ? BSD Lizenz...

Viele Grüße,
Andreas

UdoSchake, 2015-07-05:
Hallo Andreas,

nicht ganz. Die verwenden bei dem Zähler ein 'vereinfachtes' Protokoll des M-Bus-Protokolls und die IR-Schnittstelle des Zählers, entweder über USB oder UART.

Gruß
Udo

schnello, 2015-07-05:
Hallo.

Ich hab so einen Zähler im Einsatz.
Der IR Kopf wird mit 9600 8N1 angesteuert und der Zähler sendet jede Minute ein: 1040F03016h

Wird diese Nachricht mit einem E5h bestätigt (0,5 Sekunden) fängt er an die Daten zu senden. Der Zähler erwartet das jede Nachricht mit einem E5h bestätigt wird. Macht man dies nicht geht er nach 6 Nachrichten wieder in den Suchmodus.

Eine Nachricht:
68 5F 5F 68 53 F0 5B 00 00 00 00 2D 4C 01 0E 0D 00 50 05 3F D0 FE B7 26 76 0C C7 AA F0 B5 2B 41 F0 C5 41 BD 63 06 DC D8 B9 1B 3D A2 31 1E F1 3D 25 14 D0 96 00 82 16 1E FE C4 B6 CB 1E 0B 33 28 BE 61 77 DC A5 94 C1 28 00 24 A8 35 F1 D6 55 BA 71 82 B2 56 E9 4B D3 3A C0 A6 B0 8D A4 67 81 EB 4E 91 E0 12 16

Wobei dieser Teil der Berreich ist mit den Verschlüsselten Daten (AES128-cbc):
3F D0 FE B7 26 76 0C C7 AA F0 B5 2B 41 F0 C5 41 BD 63 06 DC D8 B9 1B 3D A2 31 1E F1 3D 25 14 D0 96 00 82 16 1E FE C4 B6 CB 1E 0B 33 28 BE 61 77 DC A5 94 C1 28 00 24 A8 35 F1 D6 55 BA 71 82 B2 56 E9 4B D3 3A C0 A6 B0 8D A4 67 81 EB 4E 91 E0

Entschlüsselt wird die Nachricht mit dem AES Key (bekommt man von der NetzAG) und dem IV Key der aus der Nachricht selbst gebildet wird.

In bash codiere ich die Nachricht mit folgender Zeile:
openssl enc -d -nosalt -nopad -aes-128-cbc -K "$AES_KEY" -iv "$KEY_IV"

Der IV Key selbst wird aus der Message gebildet:
Teil 1: 4C2D00000000010E
Teil 2: Byte Nr 16 (also in dem Beispiel oben: 0D
V: 4C2D00000000010E0D0D0D0D0D0D0D0D
enschlüsselte Nachricht:
2F 2F 06 6D 5F 0C 48 C1 17 1B 04 03 00 72 0A 00 04 83 3C 84 F9 01 00 84 10 FB 82 73 1E 76 05 00 84 10 FB 82 F3 3C 26 3E 01 00 04 2B 00 00 00 00 04 AB 3C 75 00 00 00 04 FB 14 00 00 00 00 04 FB 94 3C 00 00 00 00 04 83 FF 04 14 00 00 00 2F 2F

Anmerkung: Bei mir sind die ersten Füllbytes 4E4E

[image]

[image]

Grüße

andig, 2015-07-05:
Hier ist das Telegrammformat nachzulesen, passt zum Beispiel oben: http://www.m-bus.com/mbusdoc/md5.php

andig, 2015-07-06:
> Der IR Kopf wird mit 9600 8N1 angesteuert und der Zähler sendet jede Minute ein: 1040F03016h

Laut spec ist das ein Short Frame:

  - 10h Start 
  - 40h C Field: SND_NKE (Initialization of Slave)
  - F0h A Field: Address of M-Bus Slave = 240
  - 30h Check Sum: arithmetical sum without carry (40 + F0 = 130)
  - 16h Stop 
  
> Eine Nachricht:
> 68 5F 5F 68 53 F0 5B 00 00 00 00 2D 4C 01 0E ... 12 16

Laut spec ist das ein Long Frame (bzw. Control Frame):

  - 68h Start 
  - 5Fh L Field  Length field (2x)
  - 68h Start 
  - 53h C Field: SND_UD (Send User Data to Slave)
  - F0h A Field: Address of M-Bus Slave = 240
  - 5Bh: CI Field: **undokumentiert** Laut www.m-bus.com/files/w4b21021.pdf "reserved", also anscheinend zählerspezifisch
  - User Data (0-252 Byte) entsprechend Length field
    - bestehend aus undokumentierten Block bevor die o.a. Nutzdaten beginnen:

            00 00 00 00 2D 4C 01 0E 0D 00 50 05

    - und Nutzdaten 

            3F D0 FE B7...

  - 12h Check Sum
  - 16h Stop

schnello, 2015-07-06:
Hallo

Die details Spec gibts direkt auf der OMS Seite:
http://oms-group.org/download4all/

Der Zähler selbst setzt auf die Gen 3 auf:
http://oms-group.org/fileadmin/pdf/OMS-Spec_Vol2_Primary_v301.pdf

>bestehend aus undokumentierten Block bevor die o.a. Nutzdaten beginnen:
>
>>00 00 00 00 2D 4C 01 0E 0D 00 50 05

[image]

>5Bh: CI Field: undokumentiert Laut www.m-bus.com/files/w4b21021.pdf "reserved", also anscheinend zählerspezifisch

[image]

Grüße

schnello, 2015-07-06:
Hallo.

Hier noch eine Nachricht aus dem Siemens voll aufgeschlüsselt:

68 Start byte
5F L Field
5F L Field
68 Start byte

53 C Field Respond user data
F0 A Field Secondary addressing mode
5B Ci Field

00 Ident NR 
00 Ident NR
00 Ident NR
00 Ident NR
2D Manufac Code
4C Manufac Code
01 Version
0E Device Type
01 Acess Nr 

00 M Bus State (alert or errors)
50 Config word ??
05 Config word ??
decrypted Payload:
FED3BA7A77FD28423CD69AE14CEF331A71B18F3DB1D372F106EEF1995E1BC82E4DAF962F331C812D
BAAAFB43ED3C32D14D7FA0A8647FCC37B82B5FCE3939F998682C378267053933226B4A421A742E13

DC Checksum
16 Stop Byte

Payload encrypted:

2F2F066D4324CEFB161A04037EA70E0004833C000000008410FB8273AD2F00008410FB82F33C200C
0200042B3901000004AB3C0000000004FB140000000004FB943C900000000483FF04000000002F2F

Daten:
2F2F fill bytes

066D (date and time Obis 0.9.1 und 0.9.2)
4324CEFB161A ==> 1A 16 FB CE 24 43 ==> 110100001011011111011110011100010010001000011
000011 =  3 sec
100100 = 36 min
[image]

0403 (Energie A+ Obis 1.8.0)
7EA70E00 ==> 00 0E A7 7E ==> 960382 Wh

04833C (Energie A- Obis 2.8.0)
00000000 ==> 00 00 00 00  ==> 0

8410FB8273 (Energie R+ Obis 3.8.1)
AD2F0000 ==> 00 00 2F AD ==> 12205 varh

8410FB82F33C (Energie R- Obis 4.8.1)
200C0200 ==> 00 02 0C 20 ==>  134176varh

042B (Wirkleistung P+ Obis 1.7.0)
39010000 ==> 00 00 01 39 ==> 313 W

04AB3C (Wirkleistung P- Obis 2.7.0)
00000000 

04FB14 (Blindleistung Q+ Obis 3.7.0)
00000000 = 00 00 00 00  ==> 0

04FB943C (Blindleistung Q- Obis 4.7.0)
90000000 = 00 00 00 90  ==> 144 Var

0483FF04 (Inkasso Obis 1.128.0 )
00000000  = 00 00 00 00  ==> 0Wh

2F2F fill bytes

andig, 2015-07-06:
@schnello woher kommt die Information über die Feinspezifikation der Daten nach dem CI Byte 5Bh? In In der OMS Spec finde ich:

    5Bh CMD to device 12 Bytes M-Bus(a)

aber keine Aufschlüsselung der Daten nach dem 5B. Übersehe ich einen Anhang (woher finden sich die beiden Screenshots die Du oben eingefügt hast)?

schnello, 2015-07-06:
Hallo.

Die Daten habe ich aus der http://oms-group.org/fileadmin/pdf/OMS-Spec_Vol2_Primary_v301.pdf
Seite 99

mbehr1, 2015-07-07:
Shall I call it
```
MeterMBus
```
or
```
MeterOMS
```
?

schnello, 2015-07-07:
I vote for "MeterOMS"

mbehr1, 2015-07-07:
Any issues if I use libmbus? (will need to be installed on the system)

schnello, 2015-07-07:
Not from my side. I have already compiled it on the pi.

schnello, 2015-07-07:
But how can this work with libmbus? Libmbus works as master only or?

mbehr1, 2015-07-07:
Doesn't seem so (from readme): 
```
...The role of libmbus is to decode/encode M-bus data, and to handle the communication with M-Bus devices....
```

andig, 2015-07-07:
Seems to be MIT License, sounds good.

schnello, 2015-07-07:
>Doesn't seem so (from readme):
>>...The role of libmbus is to decode/encode M-bus data, and to handle the communication with M-Bus devices....

The webpage says:
http://www.rscada.se/libmbus/
Open source C-library for communicating with M-bus slaves and for encoding/decoding the M-bus protocol.

I tried also all tools (mbus-serial-scan, mbus-serial-request, etc). One time a get a error message which has stated about: conflict with second master

UdoSchake, 2015-07-07:
the description of the AMIS-Meter says:
2.1. Kommunikation des AMIS-Zählers
In einem M-Bus-Netzwerk wird die Kommunikation durch genau einen M-Bus-Master gesteuert. Dabei wird die Datenübertragung immer durch diesen initiiert, so dass diese nach dem Master-Slave-Prinzip abläuft. Diese M-Bus-Master Funktionalität ist im AMIS-Zähler umgesetzt. Daher muss das Endgerät (z.B. Kundendisplay) einem M-Bus-Slave entsprechen.

Gruß
Udo

andig, 2015-07-08:
Es würde ja reichen die Lib für Validierung und parsen der Mbus Telegramme einzusetzen.

mbehr1, 2015-07-08:
Ich fange mal mit der Entwicklung an.

Kann mir jemand ggf. Remote-Zugriff (IP port forwarding) auf so einen (per TCP angebunden) Zähler oder Zugriff auf z.B. RPI bei per UART angebundenen Zähler geben? Gerne per PM an mbehr (a) mcbehr . de

> 
> Es würde ja reichen die Lib für Validierung und parsen der Mbus Telegramme einzusetzen.
> 
> 
> 
> > 
> > the description of the AMIS-Meter says:
> > 2.1. Kommunikation des AMIS-Zählers
> > In einem M-Bus-Netzwerk wird die Kommunikation durch genau einen M-Bus-Master gesteuert. Dabei wird die Datenübertragung immer durch diesen initiiert, so dass diese nach dem Master-Slave-Prinzip abläuft. Diese M-Bus-Master Funktionalität ist im AMIS-Zähler umgesetzt. Daher muss das Endgerät (z.B. Kundendisplay) einem M-Bus-Slave entsprechen.
> > 
> > Gruß
> > Udo

Gruß

Matthias

schnello, 2015-07-08:
Hallo.

>Kann mir jemand ggf. Remote-Zugriff (IP port forwarding) auf so einen (per TCP angebunden) Zähler >oder Zugriff auf z.B. RPI bei per UART angebundenen Zähler geben? 

Du hast Post.

Grüße

mbehr1, 2015-07-10:
Hallo,

kann mir jemand hierzu den korrekten IV angeben? Ich scheine den falsch zu berechnen, da aes_128_cbc Fehler meldet:

mbus_frame_print: Dumping M-Bus frame [type 4, 101 bytes]: 68 5F 5F 68 73 F0 5B 00 00 00 00 2D 4C 01 0E 00 00 50 05 81 A0 00 A0 D2 41 D0 E1 A8 B6 F4 F8 D0 3C 21 2E 60 99 FA 3B 4A 36 D8 1B 8D 01 86 3F 58 38 81 09 33 54 F7 AD D2 EC 10 12 3C BB FE 86 8C 14 FF F0 87 59 46 91 B4 D7 95 C1 2E 85 34 01 B2 DC 08 5C FB 1A EE D0 00 A1 9E 9D CE A6 50 15 39 2D 15 3B 98 16 

> 
> Hallo.
> 
> Kann mir jemand ggf. Remote-Zugriff (IP port forwarding) auf so einen (per TCP angebunden) Zähler >oder Zugriff auf z.B. RPI bei per UART angebundenen Zähler geben?
> 
> Du hast Post.
> 
> Grüße

Gruß

Matthias

schnello, 2015-07-10:
Hallo.
Dir fehlt ja noch dazu der AES Key.

Ich schick dir per Mail eine neue Message inkl Key und IV key.

Grüße

mbehr1, 2015-07-11:
see #181 for first version

mbehr1, 2015-07-12:
@UdoSchake kann Issue geschlossen werden? Es fehlen vermutlich noch viele Tests/Themen, die mit neuen/anderen Geräten auftreten. Aber die sollten wir als normale/neue Issues behandeln, oder?

UdoSchake, 2015-07-12:
Ja, sicher.

Gruß
Udo

closed 2015-07-12 by UdoSchake