volkszaehler/vzlogger #178: Neuer Zähler vom Typ Frequenz

bgewehr, 2015-07-07:
Hallo,

seht Ihr eine Chance, VZLogger das Frequenzzählen am GPIO als Zählertyp beizubringen? Das soll bis 25MHz gehen, wenn man auf die Interrupts verzichtet...

http://codeandlife.com/2012/07/03/benchmarking-raspberry-pi-gpio-speed/

http://abyz.co.uk/rpi/pigpio/examples.html#C_code

andig, 2015-07-07:
@bgewehr Was ist Dein Einsatzszenario?

OT: Middlewareseitig wäre der Frequenzmesser einfach ein Impulszähler analos S0.

bgewehr, 2015-07-07:
Es gibt Sensoren, die Messwerte in lineare Frequenz umwandeln wie diese hier:
http://www.adafruit.com/products/828ö
https://www.sparkfun.com/products/9768

Wenn die Frequenz zu hoch wird, taugen aber - je nach Plattform - die Interrupts nicht mehr für die zuverlässige Impulsverarbeitung, daher meine Frage...

mbehr1, 2015-07-07:
Das sind eigentlich normale S0/Impuls Zähler (nur mit hoher Frequenz).
Wie in dem anderen Thread beschrieben, müsste die Implementierung unseres S0 Zählers ein wenig optimiert werden. Dann geht bestimmt bis paar kHz (aber nicht bis 25 MHz).

> Es gibt Sensoren, die Messwerte in lineare Frequenz umwandeln wie diese hier:
> http://www.adafruit.com/products/828ö <http://www.adafruit.com/products/828%C3%B6> [...]

Gruß

Matthias

bgewehr, 2015-07-07:
Naja, paar KiloHertz wäre ja top!
Ich kann leider überhaupt kein C, also wie kann ich helfen?

UdoSchake, 2015-07-08:
http://codeandlife.com/2012/07/03/benchmarking-raspberry-pi-gpio-speed/
Dieser Artikel beschreibt wie man die GPIOs des Raspi per Programm als Ausgang schalten kann. Nix mit Frequenz zählen.

http://abyz.co.uk/rpi/pigpio/examples.html#C_code
Das kommt schon eher hin.

Gruß
Udo

andig, 2015-07-08:
> http://abyz.co.uk/rpi/pigpio/examples.html#C_code
> Das kommt schon eher hin.

Insbesondere http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetGetSamplesFunc

bgewehr, 2015-07-12:
Ich habe mir mal mit python geholfen, wäre aber weiterhin an einer "sauberen" Lösung mit vzlogger interessiert, weil mir immer alle Pulse verloren gehen, die während meiner http calls auftreten.
https://github.com/bgewehr/flow-meter-hall-sensor-frequency-counter-for-volkszaehler.org

mbehr1, 2015-07-13:
Folgende Änderungen schlage ich am S0 GPIO Meter vor:
a) Einführen eines sep. Counterthreads, der immer läuft, sobald der S0 Meter gestartet wurde und für Echtzeitfähigkeit mit anderer Scheduling-Klasse/Prio versehen werden kann.
b) Umstellen des "read" Calls, so dass nicht mehr auf einen Impuls gewartet wird, sondern die Impulse pro Sekunde (min. aber 1 Impuls für die von @andig geforderte Rückwärtskompatibilität) geliefert wird (das ist defacto dann eine Frequenzmessung)
c) Mit dem akt. sysfs-Linux generischem Ansatz werden damit vermutlich paar kHz gehen. Für schnellere Lösungen könnte eine zusätzliche  Raspberry-Pi spezifische Lösung (pigpio oder einfach direkt memory-mapped gpio auslesen) implementiert werden. Der Counterthread ist dann lediglich CPU begrenzt (bei 100% min. >40 MHz Sampling Rate. Ich würde hier ein künstliches Limit von z.B. 100kHz einbauen, damit die CPU Last verträglich bleibt (bei rpi2 könnte man vermutlich problemlos auf 250kHz gehen).  Vorteil wäre dann auch, dass die CPU Last immer konstant bleibt und nicht mit steigender Frequenz steigt, wie die akt. Implementierung.
Das Debouncing muss dann auf noch feingranularer einstellbar geändert werden (1ms Granularität ergibt ja max 1kHz).

Was meint ihr?

mbehr1, 2015-07-13:
Grad noch mit @andig eine weitere Änderung besprochen:
d) Parameter "sendZero" (senden von 0/keinem Impuls): Falls aktiv, würde der S0 Meter zum wirklichen Frequenzzähler, der auch 0 Hz zurückliefert. Die Datenrate kann dann per "aggregation" z.B. auf einen Summenwert alle 5 Min reduziert werden. Falls nicht aktiv, ist das Verhalten wie heute und ohne Impuls wird auch kein Wert an die Middleware/Datenbank gemeldet.

bgewehr, 2015-07-13:
Klingt wie ein perfekter Plan!

mbehr1, 2015-07-18:
see #185 for a first implementation of a, b and d). 
Untested! Tests welcome!

UdoSchake, 2015-07-19:
#185 getestet.

Testaufbau:
Raspberry Pi 2
Frequenzgenerator direkt an GPIO4.
Rechteckimpuls, Tastverhältnis 50%, Amplitude 0v-3,3V.
Frequenzbereich 1 Hz bis 100kHz.

Ergebnisse:
Bei 10kHz  CPU-Last: vzlogger bei 21%, kworker bei 22%, Apache bei 3%
Bei 50kHz  CPU-Last: vzlogger bei 59%, kworker bei 68%, Apache bei 6%
Bei 100kHz CPU-Last vzlogger bei 68%, kworker bei 89%, Apache bei 7%

Bis 50kHz halten sich die Ausreißer im Frontend im Rahmen.

Gruß
Udo

bgewehr, 2015-07-20:
Super! Das ist ja besser als erwartet!

andig, 2015-07-21:
Moin Udo,

Das ist ja überraschend. Was hast Du für aggtime eingestellt? Middleware auf dem gleichen Pi? Mit oder ohne Push?

UdoSchake, 2015-07-21:
Moin Andi,

keine agg.. Alles auf dem Pi, ohne Push.

andig, 2015-07-21:
> keine agg.. Alles auf dem Pi, ohne Push.

Aber mit Speicherung? Das klingt doch ziemlich gut!

UdoSchake, 2015-07-21:
ja, mit Speicherung.

Gruß
Udo

mbehr1, 2015-07-22:
Hi. Please see update in #185. @UdoSchake can you test it again?

mbehr1, 2015-07-30:
Implementiert mit Merge von #185. Kann Issue geschlossen werden?

bgewehr, 2015-08-01:
Wo finde ich die Anleitungen für die vzlogger.conf, um diese Funktionen zu testen?

mbehr1, 2015-08-01:
Ein paar Erklärungen hat der Conf Editor: http://volkszaehler.github.io/vzlogger/ <http://volkszaehler.github.io/vzlogger/> 

> 
> Wo finde ich die Anleitungen für die vzlogger.conf, um diese Funktionen zu testen?

Gruß

Matthias

bgewehr, 2015-08-14:
Also, ich habe grade den aktuellen vzlogger mit den vier water-flow-metern in Betrieb genommen und muss sagen,  dass alles super funktioniert.
Also vielen Dank!

closed 2015-08-14 by bgewehr

mbehr1, 2015-08-14:
Super. Danke fürs Feedback. 

Gruß
Matthias 

Sent from a mobile device.