volkszaehler/vzlogger #180: S0 via Gpio: Conditional logging

andig, 2015-07-09:
Ich habe einen Zweirichtungszähler mit Impulsschnittstelle den ich gerne über GPIO auslesen würde. Mit den letzten großartigen Änderungen von @mbehr1 geht es .... fast.

Problem: während eine Led die Impulse ausgibt, zeigt eine zweite die Richtung (Bezug/Einspeisung) an. Um dies mit vzlogger zu realisieren würde ich mir eine condition wünschen unter der ein Impuls für einen Kanal tatsächlich geloggt wird, also etwa so:

    meters: [
      {
        gpio: 4,  // impuls
        channels: [
          { // Bezug
            uuid: 123,
            condition: {
              gpio: 5, // richtung
              value: 0 // lo
            }
          },
          { // Lieferung
            uuid: 456,
            condition: {
              gpio: 5, // richtung
              value: 1 // hi
          }
        ]
      }, ....
    ]

Wäre das mit überschaubarem Aufwand umsetzbar? Eine "richtige" Kundenschnittstelle wird vom EV mit 500€ bepreist...

mbehr1, 2015-07-09:
Leuchtet die Richtungs-LED immer oder ist die auch gepulsed? (also muss Impuls und Richtung synchronisiert werden?)

> Ich habe einen Zweirichtungszähler mit Impulsschnittstelle den ich gerne über GPIO auslesen würde. Mit den letzten großartigen Änderungen von @mbehr1 <https://github.com/mbehr1> geht es .... fast. [...]

Gruß

Matthias

andig, 2015-07-10:
Die ist permanent an oder aus je nach Betriebszustand. In der Dämmerung wenn die PV langsam aufgibt gehts auch hin und her, aber mit entsprechend niedriger Frequenz.

mbehr1, 2015-07-10:
Wie wäre es, wenn ich die 2. LED als „Vorzeichen“ unterstützen würde, d.h. es würde negative Impulse geben. Reicht das oder muss es auf sep. UUID laufen?

> 
> Die ist permanent an oder aus je nach Betriebszustand. In der Dämmerung wenn die PV langsam aufgibt gehts auch hin und her, aber mit entsprechend niedriger Frequenz.
> 
> > 
> > Leuchtet die Richtungs-LED immer oder ist die auch gepulsed? (also muss Impuls und Richtung synchronisiert werden?)
> > 
> > 
> > > 
> > > Ich habe einen Zweirichtungszähler mit Impulsschnittstelle den ich gerne über GPIO auslesen würde. Mit den letzten großartigen Änderungen von @mbehr1 <https://github.com/mbehr1> geht es .... fast.
> > > 
> > > Problem: während eine Led die Impulse ausgibt, zeigt eine zweite die Richtung (Bezug/Einspeisung) an. Um dies mit vzlogger zu realisieren würde ich mir eine condition wünschen unter der ein Impuls für einen Kanal tatsächlich geloggt wird, also etwa so:
> > > 
> > > meters: [
> > > {
> > > gpio: 4, // impuls
> > > channels: [
> > > { // Bezug
> > > uuid: 123,
> > > condition: {
> > > gpio: 5, // richtung
> > > value: 0 // lo
> > > }
> > > },
> > > { // Lieferung
> > > uuid: 456,
> > > condition: {
> > > gpio: 5, // richtung
> > > value: 1 // hi
> > > }
> > > ]
> > > }, ....
> > > ]
> > > Wäre das mit überschaubarem Aufwand umsetzbar? Eine "richtige" Kundenschnittstelle wird vom EV mit 500€ bepreist...
> > > 
> > > 
> > 
> > Gruß
> > 
> > Matthias

Gruß

Matthias

andig, 2015-07-12:
> Reicht das oder muss es auf sep. UUID laufen?

Das würde bzgl. Kurvenverlauf für diesen Zähler genügen, allerdings können wir dann keinen Verbrauch/Zählerstand mehr für die beiden Zählwerke berechnen.

Es müssten also tatsächlich separate Kanäle sein falls das ohne den Code total zu verbiegen.

mbehr1, 2015-07-12:
Was leicht ginge, wäre dass ich bei neg. Impulsen, diese nicht per Identifier "Impulse" sondern z.B. "Impulse_neg" (und statt Power Power_neg) schicke. Dann könntest du in der Config einfach auf 2 unterschiedliche UUIDs mappen. (Allerdings muss man dann auch auf 2. unterschiedliche UUIDs mappen und kann ein echtes Vorzeichen nicht konfigurieren.)

andig, 2015-07-12:
Bin mir nicht sicher was Du meinst. Vorzeichen ist mir egal, sollten positive Werte sein wie alle anderen auch. Wie müsste ich das dann mappen?

mbehr1, 2015-07-12:
So:
```
channels: [
{ // Bezug
"uuid": "123",
"identifier: "Impulse"
},
{ // Lieferung
"uuid": "456",
"identifier": "Impulse_neg"
}
]
```
Impulse_neg wären auch positive Werte.
Man könnte dann leider nur nicht bei Nutzung nur eines Identifiers diesen rückwärts laufen lassen, aber das willst du ja eh nicht.

mbehr1, 2015-07-12:
Ich könnte noch einen Identifier "Impulse_sum" hinzufügen, der zusätzlich auch rückwärtszählen würde (also Impulse - Impulse_neg darstellt). Dann könnte sich das jeder nach Bedarf konfigurieren.

mbehr1, 2015-07-12:
Is it ok/sufficient if I add it only for real gpios and not for the UART based ones?

mbehr1, 2015-07-12:
For configuration I'd add:

"gpio_dir": { "type": "integer", "default": -1, "description": "Optional: number of GPIO port that indicated the direction of the impulses. If used you get identifiers Impulse_neg, Power_neg, Impulse_signed and Power_signed as well." },

mbehr1, 2015-07-12:
on change from positive Impulses to negative Impulses do you need a:
```
Impulse 1 (with timestamp)
...
Impulse 1 (with timestamp)
Impulse 0 (with timestamp)
Impulse_neg 1 (with timestamp)
...
```
i.e. a 0 value on the "positive" Impulses?

mbehr1, 2015-07-12:
see here a first version: https://github.com/mbehr1/vzlogger/tree/enh180_direction_GPIO
completely untested.
@andig would be good if you could test.

mbehr1, 2015-07-27:
See #185.

mbehr1, 2015-07-30:
Implementiert mit Merge von #185. Ok to close?

andig, 2015-07-31:
Thank you so much!

closed 2015-07-31 by andig

mbehr1, 2015-07-31:
Welcome!

> 
> Thank you so much!

Gruß

Matthias