volkszaehler/vzlogger #186: Unable to build PushData support

andig, 2015-07-25:
This is on an older Pi that I've updated to current git and ran `install.sh`:

	[100%] Building CXX object src/CMakeFiles/vzlogger.dir/PushData.cpp.o
	cd /home/pi/vzlogger/src && /usr/bin/c++   -DHAVE_CONFIG_HPP -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++0x -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers -I/usr/local/include -I/home/pi/vzlogger -I/home/pi/vzlogger/include    -g3 -o CMakeFiles/vzlogger.dir/PushData.cpp.o -c /home/pi/vzlogger/src/PushData.cpp
	Linking CXX executable vzlogger
	cd /home/pi/vzlogger/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
	/usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++0x -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers    CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger -rdynamic protocols/libproto.a libvz.a api/libvz-api.a -Wl,-Bstatic -ljson -Wl,-Bdynamic -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -Wl,-Bstatic -lcurl -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -Wl,-Bdynamic -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -Wl,-Bstatic -lcurl -lgnutls -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -Wl,-Bstatic -lgcrypt -lgpg-error -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -lsasl2 -lssl -lcrypto -Wl,-rpath,/usr/local/lib
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `Curl_disconnect':
	(.text+0xcc0): undefined reference to `idn_free'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `Curl_disconnect':
	(.text+0xcd0): undefined reference to `idn_free'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0x94): undefined reference to `stringprep_check_version'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0xb4): undefined reference to `idna_to_ascii_lz'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0xbc): undefined reference to `stringprep_locale_charset'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0x110): undefined reference to `idna_to_unicode_lzlz'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0x124): undefined reference to `tld_check_lz'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0x130): undefined reference to `tld_strerror'
	/usr/lib/gcc/arm-linux-gnueabihf/4.6/../../../arm-linux-gnueabihf/libcurl.a(url.o): In function `fix_hostname':
	(.text.unlikely+0x188): undefined reference to `idn_free'

Do we need a specific `curl` version (and check for it)?

ping @mbehr1

andig, 2015-07-25:
Mhm, looks like https://www.mail-archive.com/volkszaehler-dev@demo.volkszaehler.org/msg00681.html

andig, 2015-07-25:
Resolved. It was as simple as removing `CMakeCache.txt`. Sorry for bothering.

mbehr1, 2015-07-25:
Hi,

do you want to add
rm CMakeCache.txt
to install.sh?

Gruß
Matthias 

Sent from a mobile device.

andig, 2015-07-25:
Mhm. It would be nice to have a clean make target that does that but not by
default. Not sure how to add targets though with cmake.

Viele Grüße,
Andreas

mbehr1, 2015-07-25:
we have make clean already. I can check how to add CMakeCache.txt

Gruß
Matthias 
Sent from a mobile device.

andig, 2015-07-25:
Excellent, reopening.

mbehr1, 2015-07-26:
Hmm. Doesn't seem so easy. How about a solution similar to the one from http://solvedstack.com/questions/looking-for-a-cmake-clean-command-to-clear-up-cmake-output ?
(Adding a new target clean-all that removed the special make files and calls clean as well?)

mbehr1, 2015-08-20:
see PR #200

closed 2015-09-03 by andig