volkszaehler/vzlogger #187: RAM Speicher vzlogger

schnello, 2015-07-27:
Hallo.

Heute ist mir aufgefallen das die Speicherauslastung des vzlogger scheinbar is unendliche ansteigt. Zb war heute der Ramspeicherverbrauch nach etwa 24 Stunden laufzeit 630MB.

System: Rbi 2
Version: aktuell vom Git
vzlogger.conf:
root@raspberrypi:~# cat /etc/vzlogger.conf
{
"retry": 30,
"daemon": false,
"verbosity": 0,
"log": "/media/usb_stick/vzlogger.log",

"meters": [
        {
        "enabled": true,
        "mbus_debug": false,
        "aggtime": 60,
        "device" : "/dev/reader1",
        "key": "00785xxxxxxxxxxxxxxxxx777FA",
        "channels": [
        {
          "uuid": "a7b84340-249d-11e5-b65e-f13edf25e0a4",
          "aggmode": "none",
          "identifier": "1.7.0",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php"
        },
        {
          "uuid": "27632060-24c0-11e5-9890-fbeed22d3040",
          "aggmode": "max",
          "identifier": "1.8.0",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php"
        }
      ],
      "protocol": "oms"
    },
    {
      "enabled": true,
        "mbus_debug": false,
        "aggtime": 60,
          "key": "21DCXXXXXXXXXXXXXXXXXXXF75",
      "channels": [
        {
          "uuid": "74204420-249e-11e5-b277-99e79fe86b8c",
          "identifier": "1.7.0",
          "aggmode": "none",
  "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php"
        },
                {
        // "uuid": "ca869510-293f-11e5-88c8-9921d8a2d098",
        "uuid": "67fae3a0-2977-11e5-8608-a50d88545929",
        "identifier": "1.8.0",
        "aggmode": "max",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php"
        }
      ],
      "protocol": "oms",
      "device": "/dev/reader2"
    }
  ]
}

mbehr1, 2015-07-27:
I think this is due to MeterOMS usage of mbus_data_variable_parse.
As I use a stack variable the dynamically generated data doesn't get freed. So a few bytes add up a few times per second.
I'll prepare a fix for that.

mbehr1, 2015-07-28:
see fix at #188 . Confirmed by valgrind tests on mocks/MeterOMS.

mbehr1, 2015-07-29:
Fix confirmed and merged in master.

closed 2015-07-29 by mbehr1