volkszaehler/vzlogger #189: cmake error

schnello, 2015-07-29:
Hallo.

cmake findet aktuell nicht die libs bzgl OpenSLL auf einem Raspbery Pi 2. OpenSLL ist jedoch installiert.

>root@raspberrypi:~/vzlogger# apt-get install openssl
>Reading package lists... Done
>Building dependency tree
>Reading state information... Done
>openssl is already the newest version.
>0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Nach dem clonen vom git und dem starten des install.sh scripts erscheint am Ende folgende Fehlermeldung:

>CMake Warning at CMakeLists.txt:172 (message):
>  Disabled googlemock/-test tests due to GCC version < 4.8!

    >    ***** Configuration parameters *****
        >     prefix: /usr/local
      >       json: -L/usr/local/lib/libjson-c.a;-lrt -I/usr/local/include
    >         sml:  -L/usr/lib/libsml.a;-lrt -I/usr/include
  >           microhttpd: -L/usr/lib/arm-linux-gnueabihf/libmicrohttpd.a;-lrt -I/usr/include
  >           libmbus: -L/usr/local/lib/libmbus.so;-lm -I/usr/local/include
>CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
>Please set them or make sure they are set and tested correctly in the CMake files:
>OPENSSL_CRYPTO_LIBRARY (ADVANCED)
 >   linked by target "vzlogger" in directory /root/vzlogger/src
>OPENSSL_SSL_LIBRARY (ADVANCED)
 >   linked by target "vzlogger" in directory /root/vzlogger/src
>
>-- Configuring incomplete, errors occurred!

Grüße

schnello, 2015-07-29:
Nachtrag:
Installiert man "libssl-dev" läuft die installation durch.

Grüße

mbehr1, 2015-07-29:
Ja, analog .travis.yml (was immer einen guten Anhaltspunkt an notwendigen Libs liefert): 

 - sudo apt-get install -y libcurl4-openssl-dev openssl libmicrohttpd-dev uuid-dev uuid-runtime

> 
> Nachtrag:
> Installiert man "libssl-dev" läuft die installation durch.
> 
> Grüße

Gruß

Matthias

andig, 2015-11-10:
closing as dead

closed 2015-11-10 by andig