volkszaehler/vzlogger #190: Segmentation fault when using 1Wire

bgewehr, 2015-08-02:
I made the following config:
   
       {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "fc73fdb0-831f-11e2-ab63-050a66acca61",
          "identifier": "*28-000004858252",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "3a513760-89c9-11e2-8948-81bd1f67b0d0",
          "identifier": "*28-000004854a62",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "27fd1e00-89c9-11e2-bc92-c922d3bff0dc",
          "identifier": "*28-0000048553f3",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "3c391a00-8a11-11e2-92ec-0d9f1daff99d",
          "identifier": "*28-00000484d527",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "2dd2cb00-8a11-11e2-b8a9-214e007465a4",
          "identifier": "*28-000004851d7e",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "16a5a410-8a11-11e2-bfb5-55e9d027a1f8",
          "identifier": "*28-00000484c3e1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "1e6e25a0-8a11-11e2-96be-cd4b755a40c6",
          "identifier": "*28-000004857e63",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        },
        {
          "uuid": "2414a2e0-8a11-11e2-8319-eb6370d8f760",
          "identifier": "*28-000004856747",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "w1therm"
    }

The result was 

    sudo service vzlogger restart
    [....] Restarting smartmetering server: vzloggerSegmentation fault
    failed!

What can i do?

mbehr1, 2015-08-02:
which version? (vzlogger -V)

Gruß
Matthias 

Sent from a mobile device.

bgewehr, 2015-08-02:
0.3.3

bgewehr, 2015-08-02:
OK, now this seems to work:

````
pi@volkszaehler ~ $ sudo service vzlogger restart
[....] Restarting smartmetering server: vzlogger[Aug 02 14:19:44][main] vzlogger v0.4.4 based on heads/master-0-gccbc8f6625 from Thu, 30 Jul 2015 20:56:52 +0200 started.
[Aug 02 14:19:44]       New meter initialized (protocol=sml)
[Aug 02 14:19:44][chn0] New channel initialized (uuid=...28f8bf api=volkszaehler id=1-0:1.7.1*)
[Aug 02 14:19:44][chn1] New channel initialized (uuid=...7a3a7f api=volkszaehler id=1-0:1.8.1)
[Aug 02 14:19:44][s0]   unknown option for mmap (). Falling back to normal gpio.
[Aug 02 14:19:44]       New meter initialized (protocol=s0)
[Aug 02 14:19:44][chn2] New channel initialized (uuid=...a80770 api=volkszaehler id=)
[Aug 02 14:19:44][s0]   unknown option for mmap (). Falling back to normal gpio.
[Aug 02 14:19:44]       New meter initialized (protocol=s0)
[Aug 02 14:19:44][chn3] New channel initialized (uuid=...888ef4 api=volkszaehler id=)
[Aug 02 14:19:44]       New meter initialized (protocol=w1therm)
[Aug 02 14:19:44][chn4] New channel initialized (uuid=...acca61 api=volkszaehler id=*28-000004858252)
[Aug 02 14:19:44][chn5] New channel initialized (uuid=...67b0d0 api=volkszaehler id=*28-000004854a62)
[Aug 02 14:19:44][chn6] New channel initialized (uuid=...bff0dc api=volkszaehler id=*28-0000048553f3)
[Aug 02 14:19:44][chn7] New channel initialized (uuid=...aff99d api=volkszaehler id=*28-00000484d527)
[Aug 02 14:19:44][chn8] New channel initialized (uuid=...7465a4 api=volkszaehler id=*28-000004851d7e)
[Aug 02 14:19:44][chn9] New channel initialized (uuid=...27a1f8 api=volkszaehler id=*28-00000484c3e1)
[Aug 02 14:19:44][chn10]New channel initialized (uuid=...5a40c6 api=volkszaehler id=*28-000004857e63)
[Aug 02 14:19:44][chn11]New channel initialized (uuid=...d8f760 api=volkszaehler id=*28-000004856747)
[Aug 02 14:19:44][main] log level is 5
[Aug 02 14:19:44]       Daemonize process...
. ok
pi@volkszaehler ~ $ tailf /var/log/vzlogger.log
[Aug 02 14:15:32][mtr0] Meter thread started
[Aug 02 14:15:32][mtr0] Meter is opened. Starting channels.
[Aug 02 14:15:32][chn0] Logging thread started
[Aug 02 14:15:32][chn1] Logging thread started
[Aug 02 14:15:32][]     Startup failed: open direction failed
[Aug 02 14:15:32][mtr0] Number of readers: 32
[Aug 02 14:15:32][mtr0] Config.daemon: 1
[Aug 02 14:15:32][mtr0] Config.local: 0
[Aug 02 14:19:44][mtr0] Meter connection established
[Aug 02 14:19:44][]     Startup failed: open direction failed

````

bgewehr, 2015-08-02:
no data at all!

bgewehr, 2015-08-02:
Now the following log with verbosity 15:

````
[Aug 02 14:40:11][w1t]  read 22.875000 from /sys/bus/w1/devices/28-0000048553f3/w1_slave (t=22875
)
[Aug 02 14:40:11][w1t]  reading w1 device 28-0000048553f3 returned 22.875000
[Aug 02 14:40:12][push] waitAndSendOnceToAll empty dataMap (timeout?)
[Aug 02 14:40:12][w1t]  read 22.625000 from /sys/bus/w1/devices/28-000004851d7e/w1_slave (t=22625
)
[Aug 02 14:40:12][w1t]  reading w1 device 28-000004851d7e returned 22.625000
[Aug 02 14:40:13][w1t]  read 22.625000 from /sys/bus/w1/devices/28-000004856747/w1_slave (t=22625
)
[Aug 02 14:40:13][w1t]  reading w1 device 28-000004856747 returned 22.625000
[Aug 02 14:40:14][w1t]  read 24.937000 from /sys/bus/w1/devices/28-000004858252/w1_slave (t=24937
)
[Aug 02 14:40:14][w1t]  reading w1 device 28-000004858252 returned 24.937000
[Aug 02 14:40:14][w1t]  read 23.000000 from /sys/bus/w1/devices/28-000004854a62/w1_slave (t=23000
)
[Aug 02 14:40:14][w1t]  reading w1 device 28-000004854a62 returned 23.000000
[Aug 02 14:40:15][w1t]  read 22.562000 from /sys/bus/w1/devices/28-00000484c3e1/w1_slave (t=22562
)
[Aug 02 14:40:15][w1t]  reading w1 device 28-00000484c3e1 returned 22.562000
[Aug 02 14:40:16][w1t]  read 22.625000 from /sys/bus/w1/devices/28-00000484d527/w1_slave (t=22625
)
[Aug 02 14:40:16][w1t]  reading w1 device 28-00000484d527 returned 22.625000
[Aug 02 14:40:16][mtr1] Got 7 new readings from meter:
[Aug 02 14:40:16][mtr1] Reading: id=28-0000048553f3/StringItentifier: value=22.88 ts=1438519211702
[Aug 02 14:40:16][mtr1] Reading: id=28-000004851d7e/StringItentifier: value=22.62 ts=1438519212513
[Aug 02 14:40:16][mtr1] Reading: id=28-000004856747/StringItentifier: value=22.62 ts=1438519213323
[Aug 02 14:40:16][mtr1] Reading: id=28-000004858252/StringItentifier: value=24.94 ts=1438519214114
[Aug 02 14:40:16][mtr1] Reading: id=28-000004854a62/StringItentifier: value=23.00 ts=1438519214963
[Aug 02 14:40:16][mtr1] Reading: id=28-00000484c3e1/StringItentifier: value=22.56 ts=1438519215832
[Aug 02 14:40:16][mtr1] Reading: id=28-00000484d527/StringItentifier: value=22.62 ts=1438519216673
[Aug 02 14:40:16][chn2] ==> number of tuples: 0
[Aug 02 14:40:16][chn2] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn2] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn3] ==> number of tuples: 0
[Aug 02 14:40:16][chn3] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn3] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn4] ==> number of tuples: 0
[Aug 02 14:40:16][chn4] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn4] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn5] ==> number of tuples: 0
[Aug 02 14:40:16][chn5] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn5] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn6] ==> number of tuples: 0
[Aug 02 14:40:16][chn6] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn6] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn7] ==> number of tuples: 0
[Aug 02 14:40:16][chn7] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn7] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn8] ==> number of tuples: 0
[Aug 02 14:40:16][chn8] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn8] Buffer dump (size=0): {}
[Aug 02 14:40:16][chn9] ==> number of tuples: 0
[Aug 02 14:40:16][chn9] JSON request body is null. Nothing to send now.
[Aug 02 14:40:16][chn9] Buffer dump (size=0): {}
[Aug 02 14:40:17][push] waitAndSendOnceToAll empty dataMap (timeout?)

````

What does that mean and why are no values sent to the middleware?

mbehr1, 2015-08-02:
Die Direction kann bei dem Pin nicht gesetzt werden. 
Startup failed: open direction failed
Welchen Pin nutzt du? Nutzt den noch ein anderer Prozess?

Versuch mal Config von Hand und setz "configure_gpio" auf false. 

Gruß
Matthias 

Sent from a mobile device. 

> Am 02.08.2015 um 14:20 schrieb Bernd Gewehr <notifications@github.com>:
> 
> Startup failed: open direction failed

bgewehr, 2015-08-02:
"gpio_dir": -1,

sollte doch gehen, oder?

Einen anderen direction Pin nutze ich nicht.

andig, 2015-08-02:
Bernd, woran machst du das fest? Lustiges rätselraten?

bgewehr, 2015-08-02:
Hallo, 

nein, an der Beschreibung im Config-Editor:

gpio_dir 
 Number of GPIO port to be used as directional pin. If this pin is used and active Impulses_neg instead of Impulses is returned. Keep to -1 if unsure about the purpose.

Also scheint –1 doch richtig, oder?

Hat nur mit 1Wire nichts zu tun, die Meldung kommt ja vom s0-gpio, oder?

src/protocols/MeterS0.cpp 
Showing the top two matches. Last indexed 3 days ago.

      528 name.append("/sys/class/gpio/gpio");  
      529 name.append(std::to_string(_gpiopin));  
      530 name.append("/direction"); 
      531 fd = ::open(name.c_str(), O_WRONLY);  
      532 if (fd<0) throw vz::VZException("open direction failed"); 
      533 res=::write(fd,"in\n",3); 

Mit den besten Grüßen

Bernd 

From: andig 
Sent: Sunday, August 02, 2015 5:47 PM
To: volkszaehler/vzlogger 
Cc: Bernd Gewehr 
Subject: Re: [vzlogger] No data to middleware when using 1Wire (#190)

Bernd, woran machst du das fest? Lustiges rätselraten?

bgewehr, 2015-08-02:
After clearing all vzlogger.conf issues and re-enabling the i2c device in raspi-config, all works fine now.
Thank you for your support!

closed 2015-08-02 by bgewehr