volkszaehler/vzlogger #191: No data from gpio for S0 meter

bgewehr, 2015-08-02:
I added the follwing lines to rc.local:

    echo 17 > /sys/class/gpio/export && echo "rising" > /sys/class/gpio/gpio17/edge
    echo 18 > /sys/class/gpio/export && echo "rising" > /sys/class/gpio/gpio18/edge

And made the following vzlogger.conf

````    
    {
  "retry": 30,
  "daemon": true,
  "verbosity": 0,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "635481a0-6fcd-11e2-8587-eda25228f8bf",
          "identifier": "1-0:1.7.1*",
          "middleware": "http://localhost/middleware.php",
        },
        {
          "uuid": "ebb9c9b0-7058-11e2-b5ed-d12ec87a3a7f",
          "identifier": "1-0:1.8.1",
          "middleware": "http://localhost/middleware.php",
        }
      ],
      "protocol": "sml",
      "device": "/dev/ttyAMA0",
      "baudrate": 9600,
      "parity": "8n1"
    },
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "8102dbc0-6fcd-11e2-a8b7-a1307aa80770",
          "identifier": "",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 17,
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "resolution": 1000,
      "debounce_delay": 10000,
      "nonblocking_delay": 100000
    },
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "89a19f80-6fcd-11e2-be46-d7713e888ef4",
          "identifier": "",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 18,
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "resolution": 100,
      "debounce_delay": 30,
      "nonblocking_delay": 100000
    }
  ]
}
````

but get no data from S0 meters.

In vzlogger.log the following lines appear on each start:

````
[Aug 02 08:44:40][]     open(): No such file or directory
[Aug 02 08:44:40][mtr1] Cannot open meter
[Aug 02 08:44:40][]     Startup failed for Meteropen failed.
````

bgewehr, 2015-08-02:
With verbosity 10 bekomme ich folgenden output beim Start:

````
[....] Restarting smartmetering server: vzlogger[Aug 02 08:56:34][mtr0] Creating new meter with protocol sml.
[Aug 02 08:56:34][mtr0] Meter configured.
[Aug 02 08:56:34]       New meter initialized (protocol=sml)
[Aug 02 08:56:34]       Configure channel.
[Aug 02 08:56:34][chn0] New channel initialized (uuid=...28f8bf protocol=volkszaehler id=1-0:1.7.1*)
[Aug 02 08:56:34]       Configure channel.
[Aug 02 08:56:34][chn1] New channel initialized (uuid=...7a3a7f protocol=volkszaehler id=1-0:1.8.1)
[Aug 02 08:56:34][mtr1] Creating new meter with protocol s0.
[Aug 02 08:56:34][mtr1] Meter configured.
[Aug 02 08:56:34]       New meter initialized (protocol=s0)
[Aug 02 08:56:34]       Configure channel.
[Aug 02 08:56:34][chn2] New channel initialized (uuid=...a80770 protocol=volkszaehler id=)
[Aug 02 08:56:34][mtr2] Creating new meter with protocol s0.
[Aug 02 08:56:34][mtr2] Meter configured.
[Aug 02 08:56:34]       New meter initialized (protocol=s0)
[Aug 02 08:56:34]       Configure channel.
[Aug 02 08:56:34][chn3] New channel initialized (uuid=...888ef4 protocol=volkszaehler id=)
[Aug 02 08:56:34]       Have 3 meters.
[Aug 02 08:56:34][main] foreground=0, daemon=1, local=0
[Aug 02 08:56:34]       Daemonize process...
. ok

````

bgewehr, 2015-08-02:
With s0vz all went fine before when having the following rc.local

````
# Part of S0/Impulse to Volkszaehler 'RaspberryPI deamon'
# echo 17 > /sys/class/gpio/export && echo "in" > /sys/class/gpio/gpio17/direction && echo "falling" > /sys/class/gpio/gpio17/edge
# echo 18 > /sys/class/gpio/export && echo "in" > /sys/class/gpio/gpio18/direction && echo "falling" > /sys/class/gpio/gpio18/edge
````

and the s0vz.cfg like this

````
/* Hostname, FQDN or IP of your VZ - normaly this should be 'localhost' */
vzserver = "localhost";

/* Path to the VZ middleware.php script, WITHOUT preposed and trailing slash */
vzpath = "middleware.php";

/* HTTP Port of your VZ */
vzport = 80;

/* S0 Inputs; VZ chanel type: Strom_messer_ / Gas (Steps/Stufen) */
GPIO0 = "8102dbc0-6fcd-11e2-a8b7-a1307aa80770"; /* Gas */
GPIO1 = "89a19f80-6fcd-11e2-be46-d7713e888ef4"; /* Wasser */
GPIO2 = "zwei";
GPIO3 = "drei";
GPIO4 = "vier";
GPIO5 = "funf";
````

bgewehr, 2015-08-02:
I am using a raspberry Pi 1B

mbehr1, 2015-08-02:
same here: version?

Gruß
Matthias 

Sent from a mobile device.

bgewehr, 2015-08-02:
0.3.3

bgewehr, 2015-08-02:
Just made a git clone and executed install.sh two days ago...

mbehr1, 2015-08-02:
du brauchst 0.4.4 (oder 0.4.3)

> 
> 0.3.3

Gruß

Matthias

schnello, 2015-08-02:
Hallo.

>Just made a git clone and executed install.sh two days ago...

Was gibt denn die "install.sh" am Ende aus?

Grüße

bgewehr, 2015-08-02:
Vzlogger already Running. Make Site to ...

Oder so ähnlich.

Klang für mich sehr nach Erfolg.

Wie kann ich im src erkennen, welche Version ich im git clone habe?

bgewehr, 2015-08-02:
Oh Mann, sorry, mein init Script lief auf /usr/bin/vzlogger, nicht auf /usr/local/bin/vzlogger.

Jetzt V 0.4.4

Daten prüfe ich jetzt!

bgewehr, 2015-08-02:
So, nun was neues:

vzlogger.log
````
[Aug 02 14:07:28]       Opened logfile /var/log/vzlogger.log
[Aug 02 14:07:28][]     ===> Start meters
[Aug 02 14:07:28][mtr0] Meter connection established
[Aug 02 14:07:28][mtr0] Meter thread started
[Aug 02 14:07:28][mtr0] Meter is opened. Starting channels.
[Aug 02 14:07:28][chn0] Logging thread started
[Aug 02 14:07:28][chn1] Logging thread started
[Aug 02 14:07:28][]     Startup failed: open direction failed

````

bgewehr, 2015-08-02:
No data at all!

mbehr1, 2015-08-02:
mit welchen Rechten bzw als welchen User startest du vzlogger?

Gruß
Matthias 

Sent from a mobile device.

bgewehr, 2015-08-02:
init-Script:

````
root@volkszaehler:/var/www# cat /etc/init.d/vzlogger
#!/bin/sh
### BEGIN INIT INFO
# Provides:          vzlogger
# Required-Start:    $remote_fs
# Required-Stop:     $remote_fs
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: logging measurements to an volkszaehler.org middelware
### END INIT INFO

# Author: Steffen Vogel <info@steffenvogel.de>

# PATH should only include /usr/* if it runs after the mountnfs.sh script
PATH=/sbin:/usr/sbin:/bin:/usr/bin
DESC="smartmetering server"
NAME=vzlogger
DAEMON=/usr/local/bin/vzlogger
DAEMON_ARGS="-d"          # Arguments to run the daemon with
PIDFILE=/var/run/$NAME.pid
SCRIPTNAME=/etc/init.d/$NAME
USER=root
GROUP=users

# Exit if the package is not installed
[ -x $DAEMON ] || exit 0

# Read configuration variable file if it is present
[ -r /etc/default/$NAME ] && . /etc/default/$NAME

# Load the VERBOSE setting and other rcS variables
. /lib/init/vars.sh

# Define LSB log_* functions.
# Depend on lsb-base (>= 3.0-6) to ensure that this file is present.
. /lib/lsb/init-functions

#
# Function that starts the daemon/service
#
do_start()
...
````

bgewehr, 2015-08-02:
Also root. Habe auch group mal auf root gesetzt. Stand früher immer auf vzlogger:vzlogger.

bgewehr, 2015-08-02:
Im Moment scheint es so, als könnte der neue vzlogger bei mir gar nicht in die middleware schreiben. curl update nötig oder so was?

bgewehr, 2015-08-02:
Also, das hier funktionierte jetzt, nachdem ich diese Parameterzeilen für D0 gelöscht habe:

      "host": "",
      "pullseq": "",

Da muss man erst mal drauf kommen!

Danke für die Hilfe hierzu!

1Wire geht aber noch nicht - aber anderer Issue.

closed 2015-08-02 by bgewehr