volkszaehler/vzlogger #192: Log Level for oms based meter

schnello, 2015-08-02:
Hallo.

Könnte man bei der nächsten Anpassung die Message "encryption sanity check failed"  vom Loglevel 10 auch im Loglevel 0 ausgeben?

Scheinbar kann es vorkommen das die AES Keys einfach ausgetauscht werden.

Grüße

schnello, 2015-08-03:
Hallo.

Ich habs mal selbst versucht. Meine Lösung wäre:
[image]

Grüße

andig, 2015-08-03:
Kannst Du da einfach einen PR draus machen? Diffs helfen bei github nicht
richtig weiter...

Viele Grüße,
Andreas

schnello, 2015-08-03:
Hallo.

Ich werde es versuchen und die Git Doku studieren.

Grüße

schnello, 2015-08-03:
PR erstellt (https://github.com/volkszaehler/vzlogger/pull/194)

Ich hoffe das passt so.

Grüße

mbehr1, 2015-08-03:
bitte die erste print(log_debug,…) entfernen. Sonst haben wir 2.

> 
> PR erstellt (#194 <https://github.com/volkszaehler/vzlogger/pull/194>)
> 
> Ich hoffe das passt so.
> 
> Grüße

Gruß

Matthias

closed 2015-08-03 by schnello