volkszaehler/vzlogger #193: SendZero not contained in configEditor

bgewehr, 2015-08-02:
The config editor does not support the sendzero attribute for s0 meters.

mbehr1, 2015-08-03:
see #195

mbehr1, 2015-08-03:
can this issue be closed now? (with #195 merged)

bgewehr, 2015-08-03:
send_zero true / false seems ok! But aggmode must be AVG SUM or MAX, not avg sum or max, right? Or is it case insensitive?

closed 2015-08-03 by bgewehr

mbehr1, 2015-08-04:
It’s case insensitive:

const char *aggmode_str = optlist.lookup_string(pOptions, "aggmode");
src/Channel.cpp:                if (strcasecmp(aggmode_str, "max") == 0 ) {
src/Channel.cpp:                        _buffer->set_aggmode(Buffer::MAX);
src/Channel.cpp:                } else if (strcasecmp(aggmode_str, "avg") == 0 ) {
src/Channel.cpp:                        _buffer->set_aggmode(Buffer::AVG);
src/Channel.cpp:                } else if (strcasecmp(aggmode_str, "sum") == 0 ) {
src/Channel.cpp:                        _buffer->set_aggmode(Buffer::SUM);
src/Channel.cpp:                } else if (strcasecmp(aggmode_str, "none") == 0 ) {
src/Channel.cpp:                        _buffer->set_aggmode(Buffer::NONE);
src/Channel.cpp:                _buffer->set_aggmode(Buffer::NONE);

> 
> send_zero true / false seems ok! But aggmode must be AVG SUM or MAX, not avg sum or max, right? Or is it case insensitive?

Gruß

Matthias