volkszaehler/vzlogger #196: Errors in config-editor

bgewehr, 2015-08-04:
1. aggtime is not updated from textbox when using update button
2. aggmode select is lower case (code: enum aggmode { NONE, MAX, AVG, SUM }; dows that mean upper case?)
3. ...

Oh wait.
There is no stable function of the update button at all.

When i paste this (my currently working config):
````
{
  "retry": 30,
  "daemon": true,
  "verbosity": 0,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 10,
    "buffer": 5
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": 5,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "635481a0-6fcd-11e2-8587-eda25228f8bf",
          "identifier": "1-0:1.7.1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "ebb9c9b0-7058-11e2-b5ed-d12ec87a3a7f",
          "identifier": "1-0:1.8.1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "MAX",
          "duplicates": 0
        }
      ],
      "protocol": "sml",
      "device": "/dev/ttyAMA0",
      "baudrate": 9600,
      "parity": "8n1"
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 30,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "8102dbc0-6fcd-11e2-a8b7-a1307aa80770",
          "identifier": "Impulse",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "SUM",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 17,
      "configureGPIO": true,
      "send_zero": true,
      "resolution": 1000,
      "debounce_delay": 10000,
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 30,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "89a19f80-6fcd-11e2-be46-d7713e888ef4",
          "identifier": "Impulse",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "SUM",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 18,
      "configureGPIO": true,
      "send_zero": true,
      "resolution": 100,
      "debounce_delay": 30,
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 60,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "fc73fdb0-831f-11e2-ab63-050a66acca61",
          "identifier": "28-000004858252",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "3a513760-89c9-11e2-8948-81bd1f67b0d0",
          "identifier": "28-000004854a62",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "27fd1e00-89c9-11e2-bc92-c922d3bff0dc",
          "identifier": "28-0000048553f3",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "3c391a00-8a11-11e2-92ec-0d9f1daff99d",
          "identifier": "28-00000484d527",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "2dd2cb00-8a11-11e2-b8a9-214e007465a4",
          "identifier": "28-000004851d7e",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "16a5a410-8a11-11e2-bfb5-55e9d027a1f8",
          "identifier": "28-00000484c3e1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "1e6e25a0-8a11-11e2-96be-cd4b755a40c6",
          "identifier": "28-000004857e63",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "2414a2e0-8a11-11e2-8319-eb6370d8f760",
          "identifier": "28-000004856747",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        }
      ],
      "protocol": "w1therm"
    }
  ]
}
````
...switch to german...
macht der Editor irgendeine Konfig, die aber so nicht im Ansatz das gleiche ist. Ich habe mal einen Parameter geändert und wieder zurück geändert, um das Ergebnis zu erhalten:

````
{
  "retry": 0,
  "daemon": true,
  "verbosity": 0,
  "log": "",
  "push": [],
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": false,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [],
      "protocol": "s0",
      "device": "",
      "gpio": -1,
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "resolution": 1000,
      "send_zero": false,
      "debounce_delay": 30,
      "nonblocking_delay": 100000
    }
  ]
}
````

Ich glaube, ein detaillierter Test ist hier im Moment nicht erforderlich, eher ein grundsätzliche Überprüfung...

Ich mache gern mit, wenn Ihr mir sagt, wie.
Gruß, Bernd

bgewehr, 2015-08-04:
eine Zeile "host": "", verhindert das D0-Logging.

bgewehr, 2015-08-04:
die folgende Angabe ist obligatorisch für S0-Meter:
````
          "identifier": "Impulse",
````

Der Config-Editor sieht das aber nicht vor.

mbehr1, 2015-08-04:
Hast du für die "host":"" Zeile eine Beispiel-Config?

mbehr1, 2015-08-04:
das "Identifier": "Impulse" Problem ist "by design" / lässt sich mit dem Konzept nicht umgehen.. Aber das wäre noch akzeptabel. Das Laden (Update Form) von existierenden Konfigs funktioniert außer bei trivialen Beispielen tatsächlich nicht. Auch im original Editor unter http://jeremydorn.com/json-editor/ .

bgewehr, 2015-08-04:
Ja, wenn ich sie in meine Konfig ins SML Meter einfüge, dann wird in diesem Meter nichts mehr geloggt. (D0 war falsch, oder?)

bgewehr, 2015-08-04:
Warum wird denn die Angabe "Impulse" überhaupt benötigt? Klingt irgendwie überflüssig, oder?

mbehr1, 2015-08-05:
Der Anwender kann wählen zwischen: Impulse oder Power und neuerdings noch Impulse_neg und Power_neg. 

Gruß
Matthias 

Sent from a mobile device.

andig, 2015-08-05:
> Der Anwender kann wählen zwischen: Impulse oder Power und neuerdings noch Impulse_neg und Power_neg.

Könnten wir bei S0 `Impulse` forcieren falls nichts anderes konfiguriert ist?

bgewehr, 2015-08-07:
Wer kann mir kurz helfen, die Bedeutung von Power für S0 zu verstehen?

andig, 2015-08-07:
Die Mailingliste! Mit Power rechnet vzlogger selbst die Leistung aus und Du kannst in anderen Kanaltyp in der MW loggen.

bgewehr, 2015-08-12:
braucht man das noch?
Man könnte ja auch mal fragen, ob solche Specials überhaupt noch nötig sind... ich meine, identifier = Impulse klingt schon irgendwie schräg oder? Ist ja ein bisschen Parameter-Missbrauch, weil der Parameter identifier ja für was anderes gedacht war. Ich dachte ich frag mal...

bgewehr, 2015-08-13:
Hier ein Kommentar aus der Mailingliste, der Glaube, dass vzlogger case sensitive ist, ist weit verbreitet:

Zitat
>Am 09.08.2015 um 10:41 schrieb Alexander Held:
>"aggmode": "sum",
vzlogger ist Case-sensitiv. Es muss heißen 'SUM'

Gruß
Udo

Zitat Ende

UdoSchake, 2015-09-03:
Dann sollte man dem Glauben doch hier mal ein Ende setzen und vielleicht kann ja jemand dazu eine eindeutige Aussage machen.

Gruß
Udo

andig, 2015-09-03:
> Dann sollte man dem Glauben doch hier mal ein Ende setzen und vielleicht
> kann ja jemand dazu eine eindeutige Aussage machen.

Eigentlich ganz einfach ;)

Also wenn ich z.B. hier rein schaue
https://github.com/volkszaehler/vzlogger/blob/master/src/Options.cpp#L47
dann wird case sensitive verglichen.
Andererseits wird hier
https://github.com/volkszaehler/vzlogger/blob/19eb2d0ebfdd58352731f4ba8008e2cdb634d7a6/src/Channel.cpp#L65
mit strcasecmp() verglichen- also case insensitive.

Mindestens für Aggmode spielt case also keine Rolle, die Settings sind aber
nicht konsistent so implementiert.

Viele Grüße,
Andreas

andig, 2016-01-03:
@bgewehr könntest Du das Issue aufräumen? Diskussionen bitte auf der ML, konkrete Fehler hier.

bgewehr, 2016-01-03:
Viel konnte ich nicht aufräumen. Reicht das?

andig, 2016-01-03:
Nicht wirklich. Ich verstehe nicht welcher Fehler zu beheben ist. Der:

> eine Zeile "host": "", verhindert das D0-Logging.

?

Falls nein: bitte 1 Issue je Fehler, keine Textaufgaben ;)

bgewehr, 2016-01-09:
OK, fangen wir damit an: Wenn ich meine Konfig in das Textfeld unten paste und update drücke, dann passiert nichts.

````
/**
 * vzlogger configuration
 *
 * use proper encoded JSON with javascript comments
 *
 * take a look at the wiki for detailed information:
 * http://wiki.volkszaehler.org/software/controller/vzlogger#configuration
 */
{
  "retry": 30,
  "daemon": true,
  "verbosity": 0,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 10,
    "buffer": 5
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": 5,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "635481a0-6fcd-11e2-8587-eda25228f8bf",
          "identifier": "1-0:1.7.1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "ebb9c9b0-7058-11e2-b5ed-d12ec87a3a7f",
          "identifier": "1-0:1.8.1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "MAX",
          "duplicates": 0
        }
      ],
      "protocol": "sml",
      "device": "/dev/ttyAMA0",
      "baudrate": 9600,
      "parity": "8n1"
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 60,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "8102dbc0-6fcd-11e2-a8b7-a1307aa80770",
          "identifier": "Impulse",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "SUM",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 17,
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "send_zero": true,
      "resolution": 1000,
      "debounce_delay": 10000,
      "nonblocking_delay": 100000
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 30,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "89a19f80-6fcd-11e2-be46-d7713e888ef4",
          "identifier": "Impulse",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "SUM",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "device": "",
      "gpio": 18,
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "send_zero": true,
      "resolution": 100,
      "debounce_delay": 30,
      "nonblocking_delay": 100000
    },
    {
      "enabled": true,
      "allowskip": false,
      "aggtime": 60,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "fc73fdb0-831f-11e2-ab63-050a66acca61",
          "identifier": "28-000004858252",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "3a513760-89c9-11e2-8948-81bd1f67b0d0",
          "identifier": "28-000004854a62",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "27fd1e00-89c9-11e2-bc92-c922d3bff0dc",
          "identifier": "28-0000048553f3",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "3c391a00-8a11-11e2-92ec-0d9f1daff99d",
          "identifier": "28-00000484d527",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "2dd2cb00-8a11-11e2-b8a9-214e007465a4",
          "identifier": "28-000004851d7e",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "16a5a410-8a11-11e2-bfb5-55e9d027a1f8",
          "identifier": "28-00000484c3e1",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "1e6e25a0-8a11-11e2-96be-cd4b755a40c6",
          "identifier": "28-000004857e63",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        },
        {
          "uuid": "2414a2e0-8a11-11e2-8319-eb6370d8f760",
          "identifier": "28-000004856747",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "type": "device",
          "scaler": 1,
          "aggmode": "AVG",
          "duplicates": 0
        }
      ],
      "protocol": "w1therm"
    }
  ]
}
````

bgewehr, 2016-01-09:
Das liegt an den Kommentarzeilen am Anfang. Entferne ich diese, wird der JSON String verarbeitet.
Vielleicht ein wichtiger Hinweis für die Nutzer oder eine Default conf ohne Kommentare ausliefern.

bgewehr, 2016-01-09:
[image]
wirkt aber falsch.

bgewehr, 2016-01-09:
geht noch weiter... später.

andig, 2016-01-09:
Sollten wir den Link aus der default conf von Wiki auf github markdown umbiegen? Dann können bei jedem Commit auch gleich die Parameterdoku aktualisiert werden.

Kommentare als Problem sind leider logisch- json erlaubt keine Kommentare.

andig, 2016-03-26:
Tasks:

   - [] make config editor accept json comments

andig, 2016-06-25:
VZ meeting: aggmode is case insensitive. If config setting is encountered that is case sensitive after testing please open separate issue.

closed 2016-06-25 by andig