volkszaehler/vzlogger #197: uuid-dev and other packages not checked in dependencies

bgewehr, 2015-08-13:
When using install.sh for building vzlogger, some dependencies are checked.

I had some errors on a new raspian so I had to manually install

uuid-dev
dh-autoconf
cmake

which could be checked and installed by the script too.

Or are there any reasons for not doing so?

bgewehr, 2015-08-13:
Same for libsasl2-dev

andig, 2015-08-14:
> Same for libsasl2-dev

Wofür brauchen wir die:

> As the SASL library is a 'glue layer' between many different parts of the authentication system

andig, 2015-08-14:
@bgewehr mit cmake solltest Du kein Problem haben, ebenso dh-autoreconf. Wo ist das Problem?

	deps=( grep pidof git cmake pkg-config autoreconf )
	for binary in "${deps[@]}"; do
		if binpath="$(which $binary)" ; then
			echo -n " $binary"
		else
			echo
			echo " $binary: not found. Please install to use this script (e.g. sudo apt-get install $binary)."
			exit 1
		fi
	done
	echo

bgewehr, 2015-08-14:
Also mein Ablauf ist einfach:
Neuer Raspi, neues Raspian
Clone vzlogger
Install.sh
Fehler:
- kein cmake, direkt abgebrochen
- kein dh-autoconf, abgebrochen, falscher Korrekturvorschlag (für Raspian) "install autoconf"
- kein uuid-dev, abgebrochen "file not found"
- kein libsasl2-dev, nach 100% compile beim Link abgebrochen

Musste ich manuell aus dem errorlog lesen, verstehen, Pakete finden, installieren, neu versuchen.

Geht, aber schöner wär anders...

Daher mein Vorschlag:
Erst alles prüfen, dann vorschlagen, die fehlenden Elemente zu installieren und einfach weiter laufen.

andig, 2015-08-14:
> Musste ich manuell aus dem errorlog lesen, verstehen, Pakete finden, installieren, neu versuchen.

Versteh ich nicht. Das install script bricht ab und sagt was ihm fehlt -> alles ok.

Was Dir fehlt ist wiki lesen und Vorbedingungen herstellen: http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version

> kein dh-autoconf, abgebrochen, falscher Korrekturvorschlag (für Raspian) "install autoconf"

Leider heißen package und tool anders :(

> Erst alles prüfen, dann vorschlagen, die fehlenden Elemente zu installieren und einfach weiter laufen.

Oder wiki lesen. Das funktioniert alles wie gedacht.

> kein libsasl2-dev, nach 100% compile beim Link abgebrochen

Das ist nicht dokumentiert und mir unklar. Da fehlt ggf wirklich etwas.

bgewehr, 2015-08-14:
Ok. Kann man im Wiki alles finden.

andig, 2015-08-14:
Die Vollständigkeitsprüfung ist nicht superschick, aber programmtechnisch ist das ziemlcih aufwändig run zu bekommen. Wie können ja in den Installer noch den Link zum Wiki mit einbauen?

bgewehr, 2015-08-18:
Ich habe - vielleicht fälschlicherweise - erwartet, dass das install Script ähnlich wie ein debian Paket aufgebaut ist und alle benötigten Abhängigkeiten mit installiert.

Grade fehlende uuid-Dev und lsasl2-Dev führen zu Fehlern beim kompilieren, die man erst mal verstehen muss.

Der Verweis auf das Wiki ist gültig, aber nicht elegant.

andig, 2015-08-18:
Fairer Punkt. Auf installierte libs lässt sich halbwegs prüfen, leider nicht auf installierte dev Pakete da die prinzipiell an verschiedenen Stellen liegen klnnten. Vielleicht sollten wir wenigstens warnen wenn wir nix finden.

schnello, 2015-08-18:
Hallo Andi.

Auf installierte libs lässt sich halbwegs prüfen, leider nicht auf
> installierte dev Pakete da die prinzipiell an verschiedenen Stellen liegen
> klnnten.

Und über das Paket System selbst? Zb dpkg-query -l

Grüße

andig, 2015-08-18:
> Und über das Paket System selbst? Zb dpkg-query -l

das ist dann halt und leider Debian-spezifisch...

bgewehr, 2015-08-18:
Wie wäre es mit
sudo find /|grep -w .*uuid.h
/usr/include/uuid/uuid.h

Dann hätte man eine OS-unabhängige Indikation, wenn auch keinen Beweis, dass was fehlt, wenn nichts kommt in kann den passenden Hinweis ausgeben...

andig, 2015-08-18:
Siehe http://serverfault.com/questions/54736/how-to-check-if-a-library-is-installed

Für libs:

    ldconfig -p | grep lsasl2

Für dev Pakete:

    find /usr/include/ -iname uuid.h

Jetzt fehlt nur noch ein schicker pr...

bgewehr, 2015-08-18:
pi@volkszaehler ~ $ ldconfig -p | grep lsasl2
pi@volkszaehler ~ $

Wenn das hilft? Ich bekomme kein Ergebnis!

schnello, 2015-08-18:
Da fehlt doch nur das lib.

 ldconfig -p | grep libsasl2

bgewehr, 2015-08-18:
Stimmt!
pi@volkszaehler ~ $ ldconfig -p | grep libsasl2
        libsasl2.so.2 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libsasl2.so.2
        libsasl2.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libsasl2.so
pi@volkszaehler ~ $

andig, 2016-06-25:
VZ meeting: enhance error message by adding wiki link.