volkszaehler/vzlogger #198: wrong meter measurements

tulamidan, 2015-08-18:
The setup is an LDR connected to GPIO17 on Raspbian (supplied image from Volkszähler)
 The readings in the vzlogger log file look like this:

[Aug 17 23:13:02][chn0] Adding reading to queue (value=3.00 ts=1439845982188)
[Aug 17 23:13:08][chn0] Adding reading to queue (value=3.00 ts=1439845988188)
...
[Aug 17 23:19:24][chn0] Adding reading to queue (value=3.00 ts=1439846364196)
[Aug 17 23:19:31][chn0] Adding reading to queue (value=3.00 ts=1439846371196)
[Aug 17 23:19:38][chn0] Adding reading to queue (value=1.00 ts=1439846378196)
[Aug 17 23:19:39][chn0] Adding reading to queue (value=2.00 ts=1439846379196)
[Aug 17 23:19:46][chn0] Adding reading to queue (value=3.00 ts=1439846386196)

it is unclear why "value" is equal to 3 but sometimes also just 1 or 2. The calculated consumption appears to be threefold higher than it really is.

mbehr1, 2015-08-19:
Version? (vzlogger -V)

> The setup is an LDR connected to GPIO17 on Raspbian (supplied image from Volkszähler)
> The readings in the vzlogger log file look like this:
>
> [Aug 17 23:13:02][chn0] Adding reading to queue [...]

Gruß

Matthias

tulamidan, 2015-08-19:
Sorry 0.4.4 git checkout two days ago.
I am currently trailing the issue with a higher verbosity. It might actually be a config issue:
It may happen, that vzlogger tries to read the gpio value multiple times. Any clue why this might happen? (detailed log at the end)

...
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:34][s0]   Reading S0 - returning 2 readings (n=3 n_neg = 0)
[Aug 19 08:46:34][mtr0] Got 2 new readings from meter:
[Aug 19 08:46:34][mtr0] Reading: id=Power/StringItentifier: value=599.96 ts=1439966794396
[Aug 19 08:46:34][mtr0] Reading: id=Impulse/StringItentifier: value=3.00 ts=1439966794396
[Aug 19 08:46:34][chn0] Adding reading to queue (value=3.00 ts=1439966794396)
...

---full log for one reading---
[Aug 19 08:46:31]       Opened logfile /var/log/vzlogger.log
[Aug 19 08:46:31][push] No pushDataServer defined.
[Aug 19 08:46:31][]     ===> Start meters
[Aug 19 08:46:31][s0]   counter_thread created
[Aug 19 08:46:31][mtr0] Meter connection established
[Aug 19 08:46:31][mtr0] Meter thread started
[Aug 19 08:46:31][mtr0] Meter is opened. Starting channels.
[Aug 19 08:46:31][chn0] Logging thread started
[Aug 19 08:46:31][]     Startup done.
[Aug 19 08:46:31][mtr0] Number of readers: 4
[Aug 19 08:46:31][mtr0] Config.daemon: 1
[Aug 19 08:46:31][mtr0] Config.local: 0
[Aug 19 08:46:31][chn0] Start logging thread for volkszaehler-api. Running as daemon: yes
[Aug 19 08:46:31][s0]   Counter thread started with blocking hwif
[Aug 19 08:46:31][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:31][chn0] Using default volkszaehler api.
[Aug 19 08:46:32][s0]   Reading S0 - returning 2 readings (n=1 n_neg = 0)
[Aug 19 08:46:32][mtr0] Got 2 new readings from meter:
[Aug 19 08:46:32][mtr0] Reading: id=Power/StringItentifier: value=3599.55 ts=1439966792396
[Aug 19 08:46:32][mtr0] Reading: id=Impulse/StringItentifier: value=1.00 ts=1439966792396
[Aug 19 08:46:32][chn0] Adding reading to queue (value=1.00 ts=1439966792396)
[Aug 19 08:46:32][chn0] Buffer dump (size=1): {1.0000,}
[Aug 19 08:46:32][chn0] ==> number of tuples: 1
[Aug 19 08:46:32][chn0] compare: 0 1439966792396
[Aug 19 08:46:32][chn0] JSON request body: [ [ 1439966792396, 1 ] ]
[Aug 19 08:46:32][chn0] CURL: About to connect() to localhost port 80 (#0)
[Aug 19 08:46:32][chn0] CURL:   Trying 127.0.0.1...
[Aug 19 08:46:32][chn0] CURL: connected
[Aug 19 08:46:32][chn0] CURL: Connected to localhost (127.0.0.1) port 80 (#0)
[Aug 19 08:46:32][chn0] CURL: Sent 24 bytes.. 
[Aug 19 08:46:32][chn0] CURL: Sent '[ [ 1439966792396, 1 ] ]' bytes
[Aug 19 08:46:32][chn0] CURL: upload completely sent off: 24 out of 24 bytes
[Aug 19 08:46:32][chn0] CURL: additional stuff not fine transfer.c:1037: 0 0
[Aug 19 08:46:32][chn0] CURL: HTTP 1.0, assume close after body
[Aug 19 08:46:32][chn0] CURL: Received 26 bytes
[Aug 19 08:46:32][chn0] CURL: Received '{"version":"0.3","rows":1}' bytes
[Aug 19 08:46:32][chn0] CURL: Closing connection #0
[Aug 19 08:46:32][chn0] CURL Request succeeded with code: 200
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:33][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Aug 19 08:46:34][s0]   Reading S0 - returning 2 readings (n=3 n_neg = 0)
[Aug 19 08:46:34][mtr0] Got 2 new readings from meter:
[Aug 19 08:46:34][mtr0] Reading: id=Power/StringItentifier: value=599.96 ts=1439966794396
[Aug 19 08:46:34][mtr0] Reading: id=Impulse/StringItentifier: value=3.00 ts=1439966794396
[Aug 19 08:46:34][chn0] Adding reading to queue (value=3.00 ts=1439966794396)
[Aug 19 08:46:34][chn0] Buffer dump (size=1): {3.0000,}
[Aug 19 08:46:34][chn0] ==> number of tuples: 1
[Aug 19 08:46:34][chn0] compare: 1439966792396 1439966794396
[Aug 19 08:46:34][chn0] JSON request body: [ [ 1439966794396, 3 ] ]
[Aug 19 08:46:34][chn0] CURL: About to connect() to localhost port 80 (#0)
[Aug 19 08:46:34][chn0] CURL:   Trying 127.0.0.1...
[Aug 19 08:46:34][chn0] CURL: connected
[Aug 19 08:46:34][chn0] CURL: Connected to localhost (127.0.0.1) port 80 (#0)
[Aug 19 08:46:34][chn0] CURL: Sent 24 bytes.. 
[Aug 19 08:46:34][chn0] CURL: Sent '[ [ 1439966794396, 3 ] ]' bytes
[Aug 19 08:46:34][chn0] CURL: upload completely sent off: 24 out of 24 bytes
[Aug 19 08:46:34][chn0] CURL: additional stuff not fine transfer.c:1037: 0 0
[Aug 19 08:46:34][chn0] CURL: HTTP 1.0, assume close after body
[Aug 19 08:46:34][chn0] CURL: Received 26 bytes
[Aug 19 08:46:34][chn0] CURL: Received '{"version":"0.3","rows":1}' bytes
[Aug 19 08:46:34][chn0] CURL: Closing connection #0
[Aug 19 08:46:34][chn0] CURL Request succeeded with code: 200

tulamidan, 2015-08-19:
And I fail at using the comment correctly and accidentally closed the issue :\

mbehr1, 2015-08-19:
You should consider a proper denounce value (based on max frequency and electrical characteristics). 
The newer version returns impulses per second so a value of 3 could be ok but it might as well be a bouncing issue. 

Gruß
Matthias

Sent from a mobile device.

mbehr1, 2015-09-06:
Did you try debouncing? Any update?

tulamidan, 2015-09-06:
In fact I tried quite a lot of different debounce options. Yet I was unable to find a suitable set of parameters. 
Eventually I got to the point where I endet up with just one value per masurement e.g.

 [Aug 17 23:19:24][chn0] Adding reading to queue (value=1.00 ts=1439846364196)

Instead of  [Aug 17 23:19:24][chn0] Adding reading to queue (value=3.00 ts=1439846364196)

But then I got two measurements within a second at frequent intervals.

 [Aug 17 23:19:24][chn0] Adding reading to queue (value=1.00 ts=1439846364196)
 [Aug 17 23:19:24][chn0] Adding reading to queue (value=1.00 ts=1439846365196)

The range of debounce intervals was from 10ms to more than 1s. I gave up at some point and switched to a small tool (gpio-watch) that could handle the debouncing very well. This little tool then simply calls a bash script with the CURL call to VZ an my metering works fine now.

I'm unsure if I should close the issue as I took a different route to avoid it. 
Though I fear that vzlogger may handle the debouncing in an incorrect matter - I may have been unlucky to find the right set of parameters myself. 
If I failed - there could be a more clever way to determine the correct debouncing value(s). If vzlogger fails there is the need for improvement.

andig, 2015-09-07:
Which result did you get with which config?

Viele Grüße,
Andreas

mbehr1, 2015-11-01:
What's the frequency expected? Why do you think that getting two impulses per second is wrong?

andig, 2015-11-10:
closing as dead topic

closed 2015-11-10 by andig