volkszaehler/vzlogger #201: Gpio Settings made by vzlogger

bgewehr, 2015-08-21:
I am using vzlogger with S0-gpio logging on 4 gpio pins on a raspi 2B.

Although I set the internal pull up resistor to up in rc.local, this setting seems to be reset by vzlogger - the pin state is always zero when no pulse is given.

My Sensors are water flow hall sensors so the low state is to count and the high state is not.

When I switch the magnetic valves by triggering the relays with other gpio pins, vzlogger recognizes pulses on each of the four input pins even when no water is flowing thus no pulses are generates by the sensors.

How do you think I should set this up?

mbehr1, 2015-08-21:
Do you use configureGPIO ?
If not set it defaults to true. If you don’t want vzlogger to set gpio options set it to false.

Regarding the relays: this sounds like an electrical interference. Sadly debounce doesn’t help here.
The edge 0 -> 1 is used as a trigger/counter condition. Debounce only avoids other impulses within that timeframe. But it doesn’t ignore peaks if they are really short. We should probably change that. Opinions?

> I am using vzlogger with S0-gpio logging on 4 gpio pins on a raspi 2B.
>
> Although I set the internal pull up resistor to up in rc.local, this setting seems to be reset by vzlogger - the pin state is [...]

Gruß

Matthias

bgewehr, 2015-08-22:
I found the relays as the source of the wrong pulses. When I connected the relais power connector separately to the 5V power supply, no more ghost pulses appeared.

closed 2015-08-22 by bgewehr