volkszaehler/vzlogger #202: local httpd mit s0

gitka, 2015-09-29:
s0-Protokoll reagieren anders als d0 mit aktiviertem local httpd und NULL-api:
1. Beim Parameter uuid erfolgt eine strenge Prüfung auf Volkszähler-middleware-Notation, obwohl NULL-api aktiviert ist. Bitte keine strenge Prüfung auf uuid durchführen. (siehe funktionierende Prüfung in 

d0-protocol mit NULL-api)
2. Der buffer-Parameter unter local wird anscheinend nicht berücksichtigt und läuft über. Siehe logfile: buffer dump (size=586): ... trotz buffer=5 nach ca. 9 Minuten vzlogger-Laufzeit
3. Sollen über local-httpd nur Impulse oder Leistungswerte übergeben werden?

./vzlogger -V
0.4.4
 based on git version: heads/master-0-gee3c88b401
 last commit date: Sun, 6 Sep 2015 21:16:51 +0200

vzlogger.conf: https://gist.github.com/gitka/889b8b7204691feece4c
logfile: https://gist.github.com/gitka/5d587fd7e6b7b4218ee9

Gruß
gitka

mbehr1, 2015-09-29:
Das sollten lieber 3 separate Issues sein.

Zu Punkt 1: Die uuid wird immer gleich gebarest (Funktion config_validate_uuid(const char *uuid) ). Bitte mal ein Beispiel schicken, wo keine strenge Prüfung erfolgt. Bitte als separates Issue.

Punkt 2 ist kein Fehler im local http, sondern in der Null API. Die send() Funktion sollte die Daten aus dem Channel löschen. Das tut sie aber nicht. Daher läuft der Channel Buffer voll (als wenn die Daten nicht an die Middleware gesendet werden konnten). Das muss dringend gefixt werden!
@andig Null Api ist von dir, oder? Was ist die Intention davon? Soll ich bei vz::api::Null::send() Code zum Löschen (mark as deleted) der Channel Daten hinzufügen?

Zu Punkt 3: local httpd kann für beides genutzt werden. Je nachdem, wie die Channel konfiguriert sind. Kein Issue, oder?

mbehr1, 2015-09-29:
Siehe PR #203 mit einem möglichen Fix für Punkt 2.

mbehr1, 2015-09-30:
@gitka kannst du prüfen, ob der Fix in Version 0.4.5 das Problem 2 behebt?

gitka, 2015-10-04:
Zu Punkt 1 u. 3 mache ich gern eigene issues auf.

Zu Punkt 2:
Ich sehe keine Änderung.
Buffer dump steigt weiterhin stetig. Ich hätte erwartet, dass bei buffer=5 die Größe nach 5 Sekunden bei gleicher Leistung nicht weiter ansteigt, .
Ausserdem zeigt der httpd nun keine tuples mehr an:
{ "version": "0.4.5", "generator": "vzlogger", "data": [ { "uuid": "7836dd20-00d5-11e0-bab1-856ed5f95918", "last": 0, "interval": -1, "protocol": "s0" }, { "uuid": "7836dd20-00d5-11e0-bab1-856ed5f95919", "last": 1443941296859, "interval": -1, "protocol": "s0" } ] }

root@raspberrypi:/home/pi# ./vzloggers0.sh
[Oct 04 08:21:28][main] vzlogger v0.4.5 based on heads/master-0-gceb9fb2b89 from Wed, 30 Sep 2015 06:35:02 +0200 started.
[Oct 04 08:21:28][mtr0] Creating new meter with protocol s0.

logfile: https://gist.githubusercontent.com/gitka/5d587fd7e6b7b4218ee9/raw/1960fce1638ac07461d24f8c78de8c6e44110b98/vzloggers0_045.log

mbehr1, 2015-10-04:
Schaue ich mir heute Abend gleich an.
Aber wie gesagt: buffer=5 hat mit „Buffer dump (size=xy)“ nichts zu tun. Das eine ist local httpd das andere ist Null-Api Handler.

> Zu Punkt 1 u. 3 mache ich gern eigene issues auf.
>
> Zu Punkt 2:
> Ich sehe keine Änderung.
> Buffer dump steigt weiterhin stetig. Ich hätte erwartet, dass bei buffer=5 die Größe nach 5 Sekunden bei [...]

Gruß

Matthias

gitka, 2015-10-04:
Hab noch einmal getestet:
- mit negativen local>buffer Werten:
Ausgabe über local-httpd ist OK. buffer dump size im log-file steigt 
ständig.
- erneuter Test mit positiven local>buffer Werten:
Die local-httpd Ausgabe passt hier nicht. Die alten Werte bleiben in der 
Ausgabe und wachsen daher stetig an.
Die buffer dump size im log file steigt auch stetig.

Dann haben wir wohl zwei Probleme.

Gruß
Karlheinz

------- Original Nachricht --------
Betreff: *** GMX Spamverdacht *** Re: [vzlogger] local httpd mit s0 (#202)

mbehr1, 2015-10-04:
Hast du 

„daemon“: true

in der conf gesetzt? 

Damit sollte das Wachsen des Buffers nicht (mehr ab 0.4.5) auftreten.

> 
> Hab noch einmal getestet:
> - mit negativen local>buffer Werten:
> Ausgabe über local-httpd ist OK. buffer dump size im log-file steigt 
> ständig.
> - erneuter Test mit positiven local>buffer Werten:
> Die local-httpd Ausgabe passt hier nicht. Die alten Werte bleiben in der 
> Ausgabe und wachsen daher stetig an.
> Die buffer dump size im log file steigt auch stetig.
> 
> Dann haben wir wohl zwei Probleme.
> 
> Gruß
> Karlheinz
> 
> ------- Original Nachricht --------
> Betreff: *** GMX Spamverdacht *** Re: [vzlogger] local httpd mit s0 (#202)
> 
> > Schaue ich mir heute Abend gleich an.
> > Aber wie gesagt: buffer=5 hat mit „Buffer dump (size=xy)“ nichts zu 
> > tun. Das eine ist local httpd das andere ist Null-Api Handler.
> >
> >
> > >
> > > Zu Punkt 1 u. 3 mache ich gern eigene issues auf.
> > >
> > > Zu Punkt 2:
> > > Ich sehe keine Änderung.
> > > Buffer dump steigt weiterhin stetig. Ich hätte erwartet, dass bei 
> > buffer=5 die Größe nach 5 Sekunden bei gleicher Leistung nicht weiter 
> > ansteigt, .
> > > Ausserdem zeigt der httpd nun keine tuples mehr an:
> > > { "version": "0.4.5", "generator": "vzlogger", "data": [ { "uuid": 
> > "7836dd20-00d5-11e0-bab1-856ed5f95918", "last": 0, "interval": -1, 
> > "protocol": "s0" }, { "uuid": "7836dd20-00d5-11e0-bab1-856ed5f95919", 
> > "last": 1443941296859, "interval": -1, "protocol": "s0" } ] }
> > >
> > > root@raspberrypi:/home/pi# ./vzloggers0.sh
> > > [Oct 04 08:21:28][main] vzlogger v0.4.5 based on 
> > heads/master-0-gceb9fb2b89 from Wed, 30 Sep 2015 06:35:02 +0200 started.
> > > [Oct 04 08:21:28][mtr0] Creating new meter with protocol s0.
> > >
> > > logfile: 
> > https://gist.githubusercontent.com/gitka/5d587fd7e6b7b4218ee9/raw/1960fce1638ac07461d24f8c78de8c6e44110b98/vzloggers0_045.log 
> > <https://gist.githubusercontent.com/gitka/5d587fd7e6b7b4218ee9/raw/1960fce1638ac07461d24f8c78de8c6e44110b98/vzloggers0_045.log>
> > >
> >
> > Gruß
> >
> > Matthias

Gruß

Matthias

gitka, 2015-10-05:
Nein, daemon war auf false. Ich werde den buffer noch beobachten.
Wird der daemon Parameter denn noch benötigt? vzlogger startet bei mir 
immer mit "Daemonize process..."

------- Original Nachricht --------
Betreff: *** GMX Spamverdacht *** Re: [vzlogger] local httpd mit s0 (#202)

mbehr1, 2015-10-05:
Wir müssen die daemon, local, logging Logik mal aufräumen. Bei daemon = false und local=enabled wird auch das „daemonize“ vorgenommen, aber es gibt intern paar Unterschiede (z.B. wird logging nicht aktiviert, daher die Daten nie aus dem Buffer entfernt,…)
Wäre gut, wenn du das mal testen kannst und auch Feedback geben, ob damit das andere Problem (local->buffer) auch behoben ist.

> 
> Nein, daemon war auf false. Ich werde den buffer noch beobachten.
> Wird der daemon Parameter denn noch benötigt? vzlogger startet bei mir 
> immer mit "Daemonize process..."
> 
> ------- Original Nachricht --------
> Betreff: *** GMX Spamverdacht *** Re: [vzlogger] local httpd mit s0 (#202)
> 
> > Hast du
> >
> > „daemon“: true
> >
> > in der conf gesetzt?
> >
> > Damit sollte das Wachsen des Buffers nicht (mehr ab 0.4.5) auftreten.
> >
> >
> >
> > >
> > > Hab noch einmal getestet:
> > > - mit negativen local>buffer Werten:
> > > Ausgabe über local-httpd ist OK. buffer dump size im log-file steigt
> > > ständig.
> > > - erneuter Test mit positiven local>buffer Werten:
> > > Die local-httpd Ausgabe passt hier nicht. Die alten Werte bleiben in 
> > der
> > > Ausgabe und wachsen daher stetig an.
> > > Die buffer dump size im log file steigt auch stetig.
> > >
> > > Dann haben wir wohl zwei Probleme.
> > >
> > > Gruß
> > > Karlheinz
> > >
> > > ------- Original Nachricht --------
> > > Betreff: *** GMX Spamverdacht *** Re: [vzlogger] local httpd mit s0 
> > (#202)
> > >
> > > > Schaue ich mir heute Abend gleich an.
> > > > Aber wie gesagt: buffer=5 hat mit „Buffer dump (size=xy)“ nichts zu
> > > > tun. Das eine ist local httpd das andere ist Null-Api Handler.
> > > >
> > > >
> > > > >
> > > > > Zu Punkt 1 u. 3 mache ich gern eigene issues auf.
> > > > >
> > > > > Zu Punkt 2:
> > > > > Ich sehe keine Änderung.
> > > > > Buffer dump steigt weiterhin stetig. Ich hätte erwartet, dass bei
> > > > buffer=5 die Größe nach 5 Sekunden bei gleicher Leistung nicht weiter
> > > > ansteigt, .
> > > > > Ausserdem zeigt der httpd nun keine tuples mehr an:
> > > > > { "version": "0.4.5", "generator": "vzlogger", "data": [ { "uuid":
> > > > "7836dd20-00d5-11e0-bab1-856ed5f95918", "last": 0, "interval": -1,
> > > > "protocol": "s0" }, { "uuid": "7836dd20-00d5-11e0-bab1-856ed5f95919",
> > > > "last": 1443941296859, "interval": -1, "protocol": "s0" } ] }
> > > > >
> > > > > root@raspberrypi:/home/pi# ./vzloggers0.sh
> > > > > [Oct 04 08:21:28][main] vzlogger v0.4.5 based on
> > > > heads/master-0-gceb9fb2b89 from Wed, 30 Sep 2015 06:35:02 +0200 
> > started.
> > > > > [Oct 04 08:21:28][mtr0] Creating new meter with protocol s0.
> > > > >
> > > > > logfile:
> > > > 
> > https://gist.githubusercontent.com/gitka/5d587fd7e6b7b4218ee9/raw/1960fce1638ac07461d24f8c78de8c6e44110b98/vzloggers0_045.log 
> >
> > > > 
> > <https://gist.githubusercontent.com/gitka/5d587fd7e6b7b4218ee9/raw/1960fce1638ac07461d24f8c78de8c6e44110b98/vzloggers0_045.log>
> > > > 
> > > > >
> > > >
> > > > Gruß
> > > >
> > > > Matthias
> > > >
> > > > 
> > > >
> > >
> > >
> >
> > Gruß
> >
> > Matthias

Gruß

Matthias

gitka, 2015-10-05:
Mit daemon=true kann ich keine zweite vzlogger-Testinstanz starten. Der 
lokale httpd startet deshalb wohl auch nicht obwohl noch im log steht:
[http] Starting local interface HTTPd on port 8081
Das Testen wird daher etwas schwierig.

------- Original Nachricht --------
Betreff: Re: [vzlogger] local httpd mit s0 (#202)

mbehr1, 2015-10-06:
I added a 2nd fix so that you can keep "daemon=false". Please test #204 .(version should be 0.4.6).

mbehr1, 2015-10-10:
@gitka kannst du noch mal testen?

gitka, 2015-10-13:
Testdurchlauf:
0.4.6  based on git version: heads/master-0-gf5e1c4158d

daemon
	local>enabled 	local>buffer
	Ergebnis: log: buffer dump
	httpd-Ausgabe 	CPU nach 1 h
false 	true 	positiv (240) 	OK (immer 1) 	OK (4 Min rollierend)
	OK (wie zu Beginn unter 2%)
false 	true 	negativ (-10)
	OK
	OK (10 Werte)
	OK

Meine Tests waren also alle erfolgreich. :-) DANKE Matthias
Auch die falsche httpd-Ausgabe bei positiven Werten ist behoben.
Wäre es möglich den "identifier"-Parameter (Impulse/Power) noch in die 
Ausgabe des httpd aufzunehmen? Das würde mir sehr weiterhelfen.

Was mir noch aufgefallen ist: Die berechneten Power-Werte schwanken sehr 
stark und geben damit nicht unbedingt den realen Wert von zur Zeit 1499 
kW wieder. Eigentlich werden meist nur zwei Werte ausgegeben:  1799 bei 
einer Sekunde und 899 bei zwei-Sekunden-LeseAbstand. Werden keine 
Millisekunden berücksichtigt?
s. https://gist.github.com/gitka/0cfc59ce364024abd3b4

gitka, 2015-10-23:
Für eine höhere Genauigkeit u. für die httpd Erweiterung gibt es neue issues. 202 kann geschlossen werden.

closed 2015-10-23 by gitka