volkszaehler/vzlogger #206: GPIO interface seconds resolution is too low

andig, 2015-10-19:
GPIO, the replacement of s0vz, basically implements `aggtime=1s`:

	[1445155302635,1800],
	[1445155304635,1800],
	[1445155305635,3600],
	[1445155307635,1800],
	[1445155309635,1800],
	[1445155311635,1800],
	[1445155312635,3600],

Raw values:

	[1445155286634,1],
	[1445155288634,1],
	[1445155290634,1],
	[1445155291634,1],
	[1445155293634,1],

We need however millisecond resolution for smooth frontend display. Two approaches are possible:

  a. Send GPIO impulses without batching for 1 second
  b. similar to aggregation batch for 1 seconds but set timestamp to timestamp of last impulse received

Pinging @mbehr1 for expert support.

mbehr1, 2015-10-25:
Hmm. That is bad/tricky. The current implementation is tweaked for high(er) frequency inputs. We discussed this a lot.
Let me check whether we can treat low frequency signals in the same way as high frequency (>>10 Hz) in the same way.

andig, 2015-10-25:
Option b) would be totally sufficient.

mbehr1, 2015-10-25:
Yes, but measuring the time is quite cpu expensive (system call plus the needed locks as time is no atomic datatype) and I want to avoid that for high frequency signals. On the other hand I want to avoid another config option.

andig, 2015-10-25:
Mhhm. Could we do it if gpio mode is used only? This would help 99% of the users? As-is its worse than s0vz?

mbehr1, 2015-10-25:
ok. I'll implement that (once per sec but return time from last impulse) for the blocking interfaces (tty or gpio non-mmap)

mbehr1, 2015-10-25:
see #211. Please test.

andig, 2015-10-27:
> Mhhm. Could we do it if gpio mode is used only? This would help 99% of the users? As-is its worse than s0vz?

> ok. I'll implement that (once per sec but return time from last impulse) for the blocking interfaces (tty or gpio non-mmap)

Thinking about it I believe this might actually be more than a 95% solution. If I imagine `mmap` would really be used for high frequencies then the last timestamp per second would always be close to the second boundary plus the number of impulses per second would be high. I.e. at 1kHz the error would already be 1 promille.

mbehr1, 2015-10-27:
yes, exactly. Should fit fine for our needs.

closed 2015-10-31 by andig