volkszaehler/vzlogger #207: Erweiterung httpd Ausgabe

gitka, 2015-10-23:
Wäre es möglich die Parameter "identifier" (Impulse/Power), "resolution" u. "aggmode" falls vorhanden in die Ausgabe des httpd aufzunehmen? Das würde mir sehr weiterhelfen.
Ausgabe heute:
{ "version": "0.4.6", "generator": "vzlogger", "data": [ { "uuid": "180", "last": 1445512340000, "interval": 60, "protocol": "d0", "tuples":  ...
Ausgabevorschlag:
{ "version": "0.4.6", "generator": "vzlogger", "data": [ { "uuid": "180", "identifier": "Impulse", "resolution": "1000", "aggmode": "none", "last": 1445512340000, "interval": 60, "protocol": "d0", "tuples":  ...

Danke 
Karlheinz

andig, 2015-10-23:
Aus meiner Sicht nicht. Der httpd ist vorgesehen um Daten auszugeben, nicht die ganze Konfiguration zu reproduzieren.

Wofür brauchst Du das?

gitka, 2015-10-23:
Hallo Andreas,

wie im Forum schon erwähnt, verwende ich keine VZ-middleware oder DB, da 
ich damit ständig Probleme auf meinem Raspi hatte, sondern logge mit 
FHEM die local-HTTPD Ausgabe.
Einige Zusatzinfos werden schon ausgegeben (protocol, intervall), die in 
FHEM ausgewertet werden. Die weiteren Felder ersparen die doppelte 
Parameter-Eingabe in FHEM.
Drei Felder mehr in der Ausgabe ist doch keine Reproduktion der ganzen 
Konfiguration. Und würde den FHEM-Usern helfen :-) .

------- Original Nachricht --------
Betreff: Re: [vzlogger] Erweiterung httpd Ausgabe (#207)

saerdnaer, 2016-02-21:
@gitka Kannst du kurz deine FHEM Konfiguration zeigen, bzw. den Beitrag im Forum verlinken?

gitka, 2016-02-21:
Modul und Konfiguration unter FHEM:
http://forum.fhem.de/index.php/topic,40766.0.html

Das Attribut VZLOGGER_modus u. VZLOGGER_resolution könnte entfallen, wenn die vzlogger-httpd-Ausgabe erweitert wird.

andig, 2016-06-25:
VZ meeting: add settings to output.

mbehr1, 2018-08-05:
@gitka wird das noch benötigt? Dann würde ich das als nächstes hinzufügen.