volkszaehler/vzlogger #209: Unterstützung der bidirektionalen Info-Schnittstelle beim Easymeter Q3CA

UdoSchake, 2015-10-25:
Diese Schnittstelle arbeitet mit 9600bd, 8N1 im SML-Format und möchte, um die Daten auszugeben, mit dieser Sequenz aufgefordert werden:

1b 1b 1b 1b 01 01 01 01 76 03 30 30 62 00 62 00 72 65 00 00 01 00 77 01 01 09 31 31 33 31 31 38 36 32 01 01 01 01 63 03 36 00 76 03 30 31 62 00 62 00 72 65 00 00 07 00 75 01 01 01 01 01 63 14 cb 00 76 03 30 32 62 00 62 00 72 65 00 00 02 00 71 01 63 75 6d 00 00 00 1b 1b 1b 1b 1a 02 72 41

Ist das mit der jetzigen 'pullseq'-Funktion in der vzlogger.conf schon möglich oder ist die Anzahl der Zeichen in der Funktion begrenzt?

Gruß
Udo

mbehr1, 2015-10-25:
pullseq ist in der Länge nicht begrenzt.  Allerdings bin ich mir nicht sicher, ob die 00 korrekt verarbeitet werden.
Achte mal, ob die Ausgabe
sending pullsequenz send (len:%d is:%d).
(nur bei verbose 15) 

korrekte Länge anzeigt.

> Diese Schnittstelle arbeitet mit 9600bd, 8N1 im SML-Format und möchte, um die Daten auszugeben, mit dieser Sequenz aufgefordert werden:
>
> 1b 1b 1b 1b 01 01 01 01 76 03 30 30 62 00 62 00 72 65 00 00 [...]

Gruß

Matthias

UdoSchake, 2015-10-25:
Scheinbar nicht:
[Oct 25 15:55:55][sml]  sending pullsequenz send (len:96 is:-1).

Gruß
Udo

mbehr1, 2015-10-25:
Hmm. Das -1 zeigt auf einen Fehler hin. Der liegt aber nicht an den Daten, sondern vermutlich Rechte o.ä.
Passen die 96?

> 
> Scheinbar nicht:
> [Oct 25 15:55:55][sml] sending pullsequenz send (len:96 is:-1).
> 
> Gruß
> Udo

Gruß

Matthias

UdoSchake, 2015-10-25:
Die 96 passen.

Gruß
Udo

mbehr1, 2015-10-25:
schick mal nen kompletten Log.

> 
> Die 96 passen.
> 
> Gruß
> Udo

Gruß

Matthias

UdoSchake, 2015-10-25:
pi@raspberrypi ~ $ cat /var/log/vzlogger.log
[Oct 25 15:55:55]       Opened logfile /var/log/vzlogger.log
[Oct 25 15:55:55][push] No pushDataServer defined.
[Oct 25 15:55:55][]     ===> Start meters
[Oct 25 15:55:55][mtr0] Meter connection established
[Oct 25 15:55:55][mtr0] Meter thread started
[Oct 25 15:55:55][mtr0] Meter is opened. Starting channels.
[Oct 25 15:55:55][]     Startup done.
[Oct 25 15:55:55][mtr0] Number of readers: 32
[Oct 25 15:55:55][mtr0] Config.daemon: 1
[Oct 25 15:55:55][mtr0] Config.local: 0
[Oct 25 15:55:55][sml]  sending pullsequenz send (len:96 is:-1).
[Oct 25 16:09:53]       terminating on signal 15.
[Oct 25 16:09:53]       Closing connections to terminate
[Oct 25 16:09:53][]     Server stopped.
[Oct 25 16:09:53][]     Trying to delete curlSessionProvider...
[Oct 25 16:09:53][]     deleted curlSessionProvider
[Oct 25 17:04:47]       Opened logfile /var/log/vzlogger.log
[Oct 25 17:04:47][push] No pushDataServer defined.
[Oct 25 17:04:47][]     ===> Start meters
[Oct 25 17:04:47][mtr0] Meter connection established
[Oct 25 17:04:47][mtr0] Meter thread started
[Oct 25 17:04:47][mtr0] Meter is opened. Starting channels.
[Oct 25 17:04:47][]     Startup done.
[Oct 25 17:04:47][mtr0] Number of readers: 32
[Oct 25 17:04:47][mtr0] Config.daemon: 1
[Oct 25 17:04:47][mtr0] Config.local: 0
[Oct 25 17:04:47][sml]  sending pullsequenz send (len:96 is:-1).
[Oct 25 17:08:38]       terminating on signal 15.
[Oct 25 17:08:38]       Closing connections to terminate
[Oct 25 17:08:38][]     Server stopped.
[Oct 25 17:08:38][]     Trying to delete curlSessionProvider...
[Oct 25 17:08:38][]     deleted curlSessionProvider
[Oct 25 17:08:48]       Opened logfile /var/log/vzlogger.log
[Oct 25 17:08:48][push] No pushDataServer defined.
[Oct 25 17:08:48][]     ===> Start meters
[Oct 25 17:08:48][mtr0] Meter connection established
[Oct 25 17:08:48][mtr0] Meter thread started
[Oct 25 17:08:48][mtr0] Meter is opened. Starting channels.
[Oct 25 17:08:48][]     Startup done.
[Oct 25 17:08:48][mtr0] Number of readers: 32
[Oct 25 17:08:48][mtr0] Config.daemon: 1
[Oct 25 17:08:48][mtr0] Config.local: 0
[Oct 25 17:08:48][sml]  sending pullsequenz send (len:96 is:-1).

vzlogger.conf:
{
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "push": [],
  "local": {
    "enabled": false,
    "port": 8080,
    "index": false,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [],
      "protocol": "sml",
      "host": "",
      "device": "/dev/ttyUSB0",
      "pullseq": "1b1b1b1b010101017603303062006200726500000100770101093131333131383632010101016303360076033031620062007265000007007501010101016314cb007603303262006200726500000200710163756d0000001b1b1b1b1a027241",
      "baudrate": 9600,
      "parity": "8n1"
    }
  ]
}

Gruß
Udo

mbehr1, 2015-10-31:
Hmm. Komisch.

Kannst du zum Debuggen mal hinter:

if (_pull.size()) {
		int wlen = write(_fd,_pull.c_str(),_pull.size());
		print(log_debug,"sending pullsequenz send (len:%d is:%d).", name().c_str(), _pull.size(), wlen);

ein:

		if (-1 == wlen) print(log_error, „write returned error %d“, name().c_str(), errno );

hängen?

> pi@raspberrypi ~ $ cat /var/log/vzlogger.log
> [Oct 25 15:55:55] Opened logfile /var/log/vzlogger.log
> [Oct 25 15:55:55][push] No pushDataServer defined.
> [Oct 25 15:55:55][] ===> Start meters
> [Oct 25 [...]

Gruß

Matthias

UdoSchake, 2015-10-31:
In welcher Datei?

Gruß
Udo

closed 2015-10-31 by UdoSchake

mbehr1, 2015-11-01:
src/protocols/MeterSML.cpp

Gruß
Matthias 

Sent from a mobile device.

mbehr1, 2015-11-01:
erledigt oder aus Versehen geschlossen?

Gruß
Matthias 

Sent from a mobile device.

UdoSchake, 2015-11-01:
Hallo Matthias,

sollte erledigt sein. Ich hatte wie gesagt den vzlogger ohne 'sudo' aufgerufen. Dadurch bekam er beim write eine Rechte-Problem.

Gruß
Udo