volkszaehler/vzlogger #210: sudo service vzlogger restart hängt

andig, 2015-10-25:
Using Udo's latest image, this is the log file, only one s0 USB meter configured:

	[Oct 25 19:50:55][]     ===> Start meters
	[Oct 25 19:50:55][s0]   counter_thread created
	[Oct 25 19:50:55][mtr0] Meter connection established
	[Oct 25 19:50:55][mtr0] Meter thread started
	[Oct 25 19:50:55][mtr0] Meter is opened. Starting channels.
	[Oct 25 19:50:55][chn0] Logging thread started
	[Oct 25 19:50:55][]     Startup done.
	[Oct 25 19:50:55][mtr0] Number of readers: 4
	[Oct 25 19:50:55][mtr0] Config.daemon: 1
	[Oct 25 19:50:55][mtr0] Config.local: 0
	[Oct 25 19:50:55][s0]   Counter thread started with blocking hwif
	[Oct 25 19:50:55][chn0] Start logging thread for volkszaehler-api. Running as daemon: yes
	[Oct 25 19:50:55][chn0] Using default volkszaehler api.
	[Oct 25 19:51:01]       terminating on signal 15.
	[Oct 25 19:51:01]       Closing connections to terminate

vzlogger is still in the process list at this point and fails to terminate.

andig, 2015-10-25:
I'm confused- isn't there a vzlogger service in the latest image?

	pi@raspberrypi ~ $ service --status-all
	 [ + ]  alsa-utils
	 [ + ]  apache2
	 [ + ]  avahi-daemon
	 [ - ]  bootlogs
	 [ - ]  bootmisc.sh
	 [ - ]  checkfs.sh
	 [ - ]  checkroot-bootclean.sh
	 [ - ]  checkroot.sh
	 [ + ]  console-setup
	 [ + ]  cron
	 [ + ]  dbus
	 [ + ]  dhcpcd
	 [ + ]  dphys-swapfile
	 [ + ]  fake-hwclock
	 [ + ]  hdparm
	 [ - ]  hostname.sh
	 [ - ]  hwclock.sh
	 [ + ]  ifplugd
	 [ + ]  kbd
	 [ + ]  keyboard-setup
	 [ - ]  killprocs
	 [ + ]  kmod
	 [ + ]  lightdm
	 [ - ]  motd
	 [ - ]  mountall-bootclean.sh
	 [ - ]  mountall.sh
	 [ - ]  mountdevsubfs.sh
	 [ - ]  mountkernfs.sh
	 [ - ]  mountnfs-bootclean.sh
	 [ - ]  mountnfs.sh
	 [ + ]  mysql
	 [ + ]  networking
	 [ - ]  nfs-common
	 [ + ]  ntp
	 [ - ]  plymouth
	 [ - ]  plymouth-log
	 [ + ]  procps
	 [ + ]  raspi-config
	 [ - ]  rc.local
	 [ - ]  rmnologin
	 [ - ]  rpcbind
	 [ - ]  rsync
	 [ + ]  rsyslog
	 [ - ]  sendsigs
	 [ + ]  ssh
	 [ - ]  sudo
	 [ + ]  triggerhappy
	 [ + ]  udev
	 [ + ]  udev-finish
	 [ - ]  umountfs
	 [ - ]  umountnfs.sh
	 [ - ]  umountroot
	 [ + ]  urandom
	 [ - ]  uuidd
	 [ - ]  x11-common

schnello, 2015-10-25:
Hello.

Try:
systemctl status all

UdoSchake, 2015-10-25:
pi@raspberrypi ~ $ systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled)
   Active: active (running) since Sun 2015-10-25 17:08:44 CET; 3h 1min ago
 Main PID: 471 (vzlogger)
   CGroup: /system.slice/vzlogger.service
           └─471 /usr/local/bin/vzlogger -d

Entsprechend:
http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version#systemd_start_script

Gruß
Udo

andig, 2015-10-25:
Ok, das tut:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Oct 25 19:59:01 2015 from hp20029987.fritz.box
pi@raspberrypi ~ $ systemctl status all
● all.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
pi@raspberrypi ~ $ systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled)
   Active: active (running) since Sun 2015-10-25 19:51:58 CET; 57min ago
 Main PID: 1944 (vzlogger)
   CGroup: /system.slice/vzlogger.service
           └─1944 /usr/local/bin/vzlogger -d
pi@raspberrypi ~ $ 

Er scheint sich allerdings per service... start starten zu lassen? Gibts da sowas wie einen wrapper?

mbehr1, 2015-10-25:
aber kannst du ihn per 
sudo systemctl stop vzlogger
beenden?

Ich habe immer befürchtet, dass blocking S0 GPIO ohne Impulse sich nicht beenden lässt. (siehe auch todo im Code ;-)

> Ok, das tut:
>
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in [...]

Gruß

Matthias

andig, 2015-10-25:
Nein, er hängt leider- obwohl ich kein gpio nutze.

UdoSchake, 2015-10-25:
Geht hier aber:

pi@raspberrypi ~ $ sudo systemctl stop vzlogger
pi@raspberrypi ~ $ systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled)
   Active: inactive (dead) since Sun 2015-10-25 21:52:12 CET; 36s ago
  Process: 471 ExecStart=/usr/local/bin/vzlogger -d (code=exited, status=0/SUCCESS)
 Main PID: 471 (code=exited, status=0/SUCCESS)

Gruß
Udo

andig, 2015-10-25:
Bei mir ist kein gpio dran @UdoSchake- bei Dir ?

mbehr1, 2015-10-25:
Mit welcher Config?

> 
> Nein, er hängt leider- obwohl ich kein gpio nutze.

Gruß

Matthias

UdoSchake, 2015-10-25:
Hier auch nicht.
Was ist eigentlich das Problem?

Gruß
Udo

andig, 2015-10-26:
> Was ist eigentlich das Problem?

vzlogger scheint beim stoppen und restarten zu hängen- der Vorgang dauert im Minutenbereich. Ist mir beim testen der Config aufgefallen. Angeschlossen ist ein Ferrariszähler mit Udo's USB Kopf.

Das ist vielleicht kein Showstopper, aber sehr verwirrend zumal das Verhalten früher nicht da war. Hängts hiermit zusammen:

    [Oct 25 19:50:55][s0] Counter thread started with blocking hwif

Config:

	{
	  "retry": 0,
	  "daemon": true,
	  "verbosity": 15,
	  "log": "/var/log/vzlogger.log",
	  "local": {
	    "enabled": false,
	    "port": 8080,
	    "index": false,
	    "timeout": 0,
	    "buffer": 0
	  },
	  "meters": [
	    {
	      "enabled": true,
	      "allowskip": false,
	      "interval": -1,
	      "aggtime": -1,
	      "aggfixedinterval": false,
	      "channels": [
	        {
	          "uuid": "1541c0a0-xxxxxxxxxxxxxxxx",
	          "identifier": "Impulse",
	          "api": "volkszaehler",
	          "middleware": "http://127.0.0.1/middleware.php",
	          "aggmode": "none",
	          "duplicates": 0
	        }
	      ],
	      "protocol": "s0",
	      "device": "/dev/ttyUSB0",
	      "resolution": 75
	    }
	  ]
	}

mbehr1, 2015-11-01:
Ok. I can reproduce it. And it's exactly the blocking HWIFs if no signal/impulse arrives. And it only get's stopped after 1min because systemd finally uses a kill -9...
I'll try to find a nicer solution/fix.

andig, 2015-11-01:
Excellent. If it was only gpio I'd say ok, but for s0 via serial its a regression?

mbehr1, 2015-11-01:
See fixes with #215.

mbehr1, 2016-01-02:
@andig Was fehlt hier noch? Kann geschlossen werden mit Integration von #215?

andig, 2016-01-03:
Yep. Viele Dank für die endlosen Fixes!

closed 2016-01-03 by andig