volkszaehler/vzlogger #213: vzlogger unempfindlicher gegen Fehler in der vzlogger.conf machen

UdoSchake, 2015-10-27:
Hallo,

besteht die Möglichkeit vzlogger unempfindlicher gegenüber Fehler in der vzlogger.conf, wie zusätzliche Leerzeichen am Ende der conf etc. zu machen?

Gruß
Udo

andig, 2015-10-27:
Zusätzliche Leerzeichen sollten kein Problem sein. Ist m.e. immer noch json. Machen lässt sich nicht viel da wir einen Bibliothek nutzen. User können ihre conf auf jsonlint.com validieren.

UdoSchake, 2015-10-27:
Sorry, ich meinte nicht Leerzeichen nach der letzten geschweiften Klammer, sondern Leerzeilen.

Gruß
Udo

mbehr1, 2015-10-27:
Falls das zu Problemen führt, bitte mit einem Bsp. ein Issue einstellen. Sollten wir fixen.

> 
> Sorry, ich meinte nicht Leerzeichen nach der letzten geschweiften Klammer, sondern Leerzeilen.
> 
> Gruß
> Udo

Gruß

Matthias

UdoSchake, 2015-10-27:
Beispiel:
In eine funktionsfähige vzlogger.conf am Ende eine Leerzeile eingefügt. Beim Starten von vzlogger kommt dann:
pi@raspberrypi /etc $ sudo vzlogger -c /etc/vzlogger.conf
[Oct 27 20:32:26][main] vzlogger v0.4.5 based on heads/master-0-gceb9fb2b89 from Wed, 30 Sep 2015 06:35:02 +0200 started.
[Oct 27 20:32:26]       extra data after end of configuration in /etc/vzlogger.conf:200
[Oct 27 20:32:26]       Failed to parse configuration due to: extra data after end of configuration

Gruß
Udo

mbehr1, 2016-01-03:
Siehe #226 für eine erste Implementierung. Whitespaces (Leerzeichen Tabs) am Ende werden ignoriert. Und auch single-line style Kommentare (// ...), jedoch mit einer Compiler Einschränkung (nur mit gcc>=4.9).
Ok?

UdoSchake, 2016-01-03:
>  jedoch mit einer Compiler Einschränkung (nur mit gcc>=4.9).
mmh, das führt doch wieder nur zu Problemen beim update, dann lass sie lieber ihre Fehler suchen.

Gruß
Udo

mbehr1, 2016-01-03:
Bei altem Compiler gehen Leerzeichen, Leerzeilen, Tabs,… Nur keine Kommentare.

> 
> jedoch mit einer Compiler Einschränkung (nur mit gcc>=4.9).
> mmh, das führt doch wieder nur zu Problemen beim update, dann lass sie lieber ihre Fehler suchen.
> 
> Gruß
> Udo

Gruß

Matthias

mbehr1, 2016-01-03:
Noch mal zur Sicherheit:
gcc < 4.9:
- innerhalb der Config: wie bisher (Leerzeichen, Kommentare single-line // und multi-line /* */,...)
- nach der Config: Leerzeichen, Tab, Leerzeilen

gcc >=4.9 oder clang:
- innerhalb der Config: wie bisher (s.o.)
- nach der Config: Leerzeichen, Tab, Leerzeilen und single-line Kommentare (//)

Der Code wird nur aufgerufen, wenn es hinter der Config noch Zeilen/Zeichen gibt.
Kompilierung mit altem gcc funktioniert wie vorher. Keine Anforderungen an neueren GCC.

closed 2016-01-03 by andig

UdoSchake, 2016-01-03:
Ja, ok, wenn es keine Probleme beim Update mit alten gcc's gibt.

Danke.
Gruß
Udo