volkszaehler/vzlogger #222: vzlogger erzeugt 2ten Impuls nach Ablauf von debounce_delay bei Verwendung von GPIO

Rainer-Zufall, 2016-01-01:
wenn mehrere schnelle Impulse auf den GPIO erfolgen erzeugt vzlogger 2 Impulse in der Datenbank. Den ersten sofort und den zweiten nach ablauf des debounce_delay

zum Szenario:

Ich habe vzlogger mit einer Testkonfig laufen lassen (09:26:58), ich schicke die im Anhang mit.
Interessanter Weise sendet der vzlogger auch beim Start einen Datensatz, obwohl kein Impuls vorhanden ist. Ist das normal?

Anschließend habe ich über ein Script den GPIO 5 mal ein und ausgeschaltet (09:27:07), das Resultat sind 2 gesendete Datensätze, der erste zum Zeitpunkt des Schaltens, der zweite nach Ablauf des delays.
Erneutes starten des Scriptes um 09:27:23 mit dem gleichen Ergebnis.

Den Log-output und die Konfig hänge ich an, damit es nicht unübersichtlich wird.

root@raspberrypi:/home/pi# vzlogger -V
0.4.8
 based on git version: heads/master-0-g296b8aee41
 last commit date: Mon, 21 Dec 2015 22:56:20 +0100

Ich hoffe das hilft bei der Fehlereingrenzung.

Gruß Marius
[vzlogger_verbose.txt](https://github.com/volkszaehler/vzlogger/files/75917/vzlogger_verbose.txt)
[vzlogger_conf.txt](https://github.com/volkszaehler/vzlogger/files/75918/vzlogger_conf.txt)

andig, 2016-01-01:
> wenn mehrere schnelle Impulse auf den GPIO erfolgen erzeugt vzlogger 2 Impulse in der Datenbank.

D.h. Bei einem einzelnen Impuls ist das nicht nachvollziehbar?

> Anschließend habe ich über ein Script den GPIO 5 mal ein und ausgeschaltet (09:27:07), das Resultat sind 2 gesendete Datensätze, der erste zum Zeitpunkt des Schaltens, der zweite nach Ablauf des delays.

Wie? Durch einen anderen GPIO der mit gpio5 verbunden ist? Lässt sich das extern in Hardware, z.b. Durch Schalter, Monoflop etc auch nachvollziehen?

Rainer-Zufall, 2016-01-01:
> D.h. Bei einem einzelnen Impuls ist das nicht nachvollziehbar?

Nein

>Lässt sich das extern in Hardware, z.b. Durch Schalter, Monoflop etc auch nachvollziehen?
Ist auch durch manuelles schalten eines GPIO an +3V nachstellbar.

>Könntest Du bitte auch mal versuchen was passiert wenn Du gem. ccbc8f6 noch ein mmap: rpi2 in der Konfiguration ergänzt? Gleiche Anzahl (fehlerhafter) Impulse bei Start und debounce_delay?

grade ist beides nicht aufgetreten.
Laut Udo ist das aber falsch in der Konfig.

winfriedp, 2016-01-01:
Ich kann das Verhalten bestätigen (ein Datensatz bei vzlogger-Neustart und doppelte Datensätze bei einem Impuls). vzlogger 0.4.8 läuft bei mir auf Udo's YPORT+ unter OpenWrt. Ich habe diverse Test gefahren. In meinen Tests schließe ich einen Reedkontakt, der direkt an der S0-Schnittstelle (GPIO 20) der YPORT+_Erweiterung angeschlossen ist. Die Erweiterung hat nach Udo's Angaben eine HW-Entprellung von 20ms. Öffnen wird nicht über die S0-Schnittstelle registriert.
Ein Kontakt (= ein Impuls) war immer kleiner 0.5s und hat in meiner Konfiguration immer zwei Datensätze im zeitlichen Abstand von debounce_delay gesendet. 
Das lässt sich auch im Log nachvollziehen. Hier ein Auszug (zwei Datensätze im Abstand von einer Sekunde bei einem Impuls):
[Dec 31 11:29:31][chn0] Adding reading to queue (value=2.00 ts=1451557771445)
[Dec 31 11:29:31][chn0] Buffer dump (size=1): {2.0000,}
[Dec 31 11:29:32][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Dec 31 11:29:32][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Dec 31 11:29:32][chn0] Adding reading to queue (value=2.00 ts=1451557772126)
[Dec 31 11:29:32][chn0] Buffer dump (size=1): {2.0000,}

Hier meine vzlogger.conf:
{
  "retry": 0,
  "daemon": true,
  "verbosity": 10,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": true,
    "port": 8080,
    "index": true,
    "timeout": 0,
    "buffer": 60
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "120",
          "identifier": "Impulse",
          "api": "null",
          "aggmode": "none"
   //       "duplicates": 0
        }
      ],
      "protocol": "s0",
      "gpio": 20,
      "configureGPIO": true,
      "send_zero": false,
      "debounce_delay": 1000
    }
  ]
}

Viele Grüße

andig, 2016-01-01:
@winfriedp ist das bei Dir mit nur einem einzelnen Impuls nachvollziehbar oder Bedarf es immer mehrerer?

Ping @mbehr1 für weitere Ideen.

winfriedp, 2016-01-01:
Alle meine Tests wurden immer mit einem einzigen Impuls (händisches Auslösen des eines Reed-Kontaktes mit einem Magneten) durchgeführt.

Viele Grüße

mbehr1, 2016-01-01:
Hallo,

ich glaube, ich verstehe die Ursache des Problems (erster Impuls während debouncing Zeit wird nach debouncing Zeit gemeldet).

Das Problem dürfte nur mit dem HWIF_GPIO auftreten.
vzlogger wartet zwar die Debouncing Zeit ab, aber das Linux GPIO Interface meldet maximal ein anstehendes Event einfach weiter (woher sollte es auch wissen, dass ich das in dem Fall nicht möchte ;-)

Ich muss mir hier eine Lösung für das GPIO Interface überlegen.

Wäre gut, wenn jemand testen/bestätigen könnte, dass es nicht mit HWIF_MMAP und HWIF_UART auftritt.

> 
> @winfriedp <https://github.com/winfriedp> ist das bei Dir mit nur einem einzelnen Impuls nachvollziehbar oder Bedarf es immer mehrerer?
> 
> Ping @mbehr1 <https://github.com/mbehr1> für weitere Ideen.

Gruß

Matthias

Rainer-Zufall, 2016-01-01:
>Wäre gut, wenn jemand testen/bestätigen könnte, dass es nicht mit HWIF_MMAP und HWIF_UART auftritt.

Wie mache ich das?

mbehr1, 2016-01-01:
Für HWIF_MMAP einfach die Option „mmap“ : „rpi1“ (oder „rpi2“) hinzufügen. Z.B. unter der „gpio“ : … Option.
MMAP geht aktuell nur mit Raspberry Pi 1 oder 2.

> 
> Wäre gut, wenn jemand testen/bestätigen könnte, dass es nicht mit HWIF_MMAP und HWIF_UART auftritt.
> 
> Wie mache ich das?

Gruß

Matthias

andig, 2016-01-01:
@Matthias: warum glaubst Du dass da ein zweiter Impuls in der Pipe ist?

mbehr1, 2016-01-01:
siehe ersten Fix (falls meine Annahme richtig ist) mit #224.
@andig Weil ich mir sonst nicht vorstellen kann, warum nicht konstant mit min. 1 Hz zurückgeliefert wird. Der Effekt tritt ja nur innerhalb der Debounce Zeit auf, oder?

andig, 2016-01-01:
@mbehr1 fix verstanden, aber nicht hierfür- das würde bedeuten es gäbe einen weiteren Impuls trotz entprellten Kontaktes?
Könnte natürlich ein Fehler im Aufbau sein?

Viele Grüße,
Andreas

mbehr1, 2016-01-01:
Ja. Das ist meine Annahme. Habe aus den Beschreibungen nicht herausgelesen, dass die Kontakte sicher entprellt sind. 
Könnte ein Test mit mmap zeigen. Und Test mit Debounce 0.

Gruß
Matthias 

Sent from a mobile device.

Rainer-Zufall, 2016-01-01:
Also meine Kontakte sind sicher nicht entprellt, das sollte ja über die Software geschehen.

mit mmap tritt das Problem anscheinend nicht auf, mit debounce 0 werden die impulse sofort in die DB geschrieben.

mbehr1, 2016-01-01:
Danke! Kannst du den Fix/die Änderung mal testen? mmap ignoriert prinzipbedingt Spikes < 1e5 ns. Daher beantwortet das die Entprellungsfrage als Ursache noch nicht.

Rainer-Zufall, 2016-01-02:
Gerne. Wie bekomme ich den fix? Neu kompilieren?

mbehr1, 2016-01-02:
```
git fetch origin pull/222/head:pr222
git checkout pr222
```
dann neu kompilieren (sollte bei der Änderung ein ```make```reichen.

(Achtung: danach wieder per ```git checkout master```auf den normalen Branch wechseln.

Rainer-Zufall, 2016-01-02:
habs irgendwie geschafft zu updaten :D sorry, hab die ganze Linux Sache noch nicht wirklich drauf. Daher die eine oder andere vielleicht blöde Frage.

Es sieht gut aus. im Test grade gab es nur noch einen Impuls, allerdings auch weiterhin einen beim Start des vzloggers, das ist ebenfalls nicht der Fall mit mmap=rpi2.

Ich werde das jetzt mal beobachten!

Rainer-Zufall, 2016-01-02:
Ja.  Ist gefixt. Soll für den Impuls beim Start ein separates issue aufgemacht werden?

mbehr1, 2016-01-02:
Ja, mach neues Issue dafür auf. Danke!

> 
> Ja. Ist gefixt. Soll für den Impuls beim Start ein separates issue aufgemacht werden?

Gruß

Matthias

mbehr1, 2016-01-02:
@andig merge ok? Du hast die Lösung oben so in Frage gestellt.

Rainer-Zufall, 2016-01-02:
Ich habe zu danken.

Marius

andig, 2016-01-02:
Macht imho auf jeden Fall Sinn, hätte ihn nur nicht als Fix für dieses Problem erwartet (das hätte ja auch mit einem Impuls auftreten müssen bei @Rainer-Zufall)

closed 2016-01-02 by andig

winfriedp, 2016-01-02:
Ich bin begeistert, wie schnell der Fehler hier korrigiert wurde. Vielen Dank. Ich bin natürlich auch an den Fix interessiert. Ich bin kein github-Experte und bin wahrscheiinlich auf ein Firmware-Update meines YPORT+ angewiesen. Wie ist der weitere Ablauf? Gibt es eine neue vzlogger-Version, mit der Udo dann eine neue Firmware bauen kann?
Viele Grüße

Rainer-Zufall, 2016-01-02:
Hallo,  nein.
Du brauchst keine neue Erweiterung.
Du musst nur deine vzlogger Version aktualisieren.

Wenn der fix schon "gemerged " wurde brauchst du nur die Installation
nochmal durchführen wie auf der Installation Seite vom vzlogger
beschrieben.

Also quasi "bash install.sh vzlogger" ausführen.  Vorher habe ich die alten
Daten gelöscht. Und dann vzlogger neu starten.
Mit "vzlogger - V"  (großes V)  kannst du die Versionen vorher und nachher
vergleichen. Bin mir nicht sicher ob es wirklich einen versionshub gab,
aber die vorletzte Version  ist vom 21.12, die aktuelle ht von heute...

allerdings bin ich kein Linux experte und kenne auch open wrt nicht
wirklich.
Wenn du noch fragen hast meld dich nochmal,  bin Grade nur am Handy.

Und ja,  ging wirklich fix mit dem fix

Mfg, Marius

mbehr1, 2016-01-02:
Hallo,

Version ist mit dem Fix auf 0.4.9 erhöht.

> 
> Hallo, nein.
> Du brauchst keine neue Erweiterung.
> Du musst nur deine vzlogger Version aktualisieren.
> 
> Wenn der fix schon "gemerged " wurde brauchst du nur die Installation
> nochmal durchführen wie auf der Installation Seite vom vzlogger
> beschrieben.
> 
> Also quasi "bash install.sh vzlogger" ausführen. Vorher habe ich die alten
> Daten gelöscht. Und dann vzlogger neu starten.
> Mit "vzlogger - V" (großes V) kannst du die Versionen vorher und nachher
> vergleichen. Bin mir nicht sicher ob es wirklich einen versionshub gab,
> aber die vorletzte Version ist vom 21.12, die aktuelle ht von heute...
> 
> allerdings bin ich kein Linux experte und kenne auch open wrt nicht
> wirklich.
> Wenn du noch fragen hast meld dich nochmal, bin Grade nur am Handy.
> 
> Und ja, ging wirklich fix mit dem fix
> 
> Mfg, Marius
> 
> > Ich bin begeistert, wie schnell der Fehler hier korrigiert wurde. Vielen
> > Dank. Ich bin natürlich auch an den Fix interessiert. Ich bin kein
> > github-Experte und bin wahrscheiinlich auf ein Firmware-Update meines
> > YPORT+ angewiesen. Wie ist der weitere Ablauf? Gibt es eine neue
> > vzlogger-Version, mit der Udo dann eine neue Firmware bauen kann?
> > Viele Grüße

Gruß

Matthias

winfriedp, 2016-01-02:
Danke für den super schnellen Service und die Hinweise. Auf dem YPORT+ kann ich keine manuelle vzlogger-Installation durchführen, Ich werde mich an Udo wenden.
Viele Grüße