volkszaehler/vzlogger #223: vzlogger crashed mit "api": "NULL"

winfriedp, 2016-01-01:
Wird in der vzlogger.conf "api": "NULL" anstatt "api": "null" angegeben, stürzt vzlogger nach dem Start ab.

Im Log-Auszug vom Start sieht noch alles gut aus:
[Jan 01 17:09:19][]     ===> Start meters
[Jan 01 17:09:19][s0]   counter_thread created
[Jan 01 17:09:19][mtr0] Meter connection established
[Jan 01 17:09:19][s0]   Counter thread started with blocking hwif
[Jan 01 17:09:19][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Jan 01 17:09:19][mtr0] Number of readers: 4
[Jan 01 17:09:19][mtr0] Config.daemon: 1
[Jan 01 17:09:19][mtr0] Config.local: 1
[Jan 01 17:09:19][mtr0] Meter thread started
[Jan 01 17:09:19][mtr0] Meter is opened. Starting channels.
[Jan 01 17:09:19][chn0] Logging thread started
[Jan 01 17:09:19][http] Starting local interface HTTPd on port 8080
[Jan 01 17:09:19][]     Startup done.
[Jan 01 17:09:19][chn0] Start logging thread for NULL-api. Running as daemon: yes
----------------------------------------------------

Hier meine conf:
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": true,
    "port": 8080,
    "index": true,
    "timeout": 0,
    "buffer": 60
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "120",
          "identifier": "Impulse",
          "api": "NULL",
          "aggmode": "none"
        }
      ],
      "protocol": "s0",
      "gpio": 20,
      "configureGPIO": true,
      "send_zero": false,
      "debounce_delay": 1000
    }
  ]
}

Viele Grüße

mbehr1, 2016-01-02:
Siehe Fix in PR #225.
Ist kein Crash, sondern eigentlich eine gut behandelte Exception. Aber ohne Debug Ausgabe nicht verständlich.
Ursache:
Wenn api nicht bekannt ist, ("null" oder "mysmartgrid") wird "volkszaehler" als Api gewählt. Volkszähler erwartet aber Parameter "middleware". Der fehlt in o.g. Config und es gibt eine Exception.
Habe folgende Änderungen gemacht:
- Parameter "api" ist nun case-insensitive
- Fehlerausgabe, wenn wegen unbekannter Api auf Volkszähler zurückgefallen wird
- Fehlerausgabe bei Volkszähler Api für fehlende Middleware
- unit Test, der auf die Exception bei fehlender Middleware prüft.

winfriedp, 2016-01-02:
Die Änderungen werden die vzlogger.conf robuster gegen Fehleingaben machen. Ich hoffe, dann funktioniert aber Api HPPTd noch, das mit "null" zu konfigurieren ist und bisher auch funktioniert hat. 
Viele Grüße

mbehr1, 2016-01-02:
Alte funktionierende Configs sollten weiterhin funktionieren. Auch configs, die zufällig ok waren, sollten weiterhin gehen aber eine Fehlermeldung im Log haben. 

Gruß
Matthias 

Sent from a mobile device.

winfriedp, 2016-01-02:
ok, dann kann aus meiner Sicht das Issue geschlossen werden.
Viele Grüße

andig, 2016-01-03:
Kannst Du selber wenn Du mit der Lösung zufrieden bist ;)

andig, 2016-01-03:
Reopening- could we add the same for aggmode and other text-based parameters?

mbehr1, 2016-01-03:
case-insensitive sowohl für Keywords ("aggmode", "Aggmode",...) als auch für die Werte ("sum", "SUM")?
Für Keywords könnte ich mir eine ganz generische, einfache Lösung vorstellen. Für Werte geht das leider nicht generisch (ich darf ja nicht alle Parameter einfach in klein/groß umwandeln). Hier bitte explizit angeben, für welche Keyword-Parameter das gewünscht ist.

andig, 2016-01-03:
> case-insensitive sowohl für Keywords ("aggmode", "Aggmode",...) als auch für die Werte

M.E. bitte nicht für keywords- die sind heute alle lowercase und sollten so bleiben?

mbehr1, 2016-01-03:
Ok. Dann nur für Parameter.
aggmode
Welche noch?

> 
> case-insensitive sowohl für Keywords ("aggmode", "Aggmode",...) als auch für die Werte
> 
> M.E. bitte nicht für keywords- die sind heute alle lowercase und sollten so bleiben?

Gruß

Matthias

mbehr1, 2016-01-03:
Siehe #228. Damit sollten alle wichtigen Parameter case insensitive sein.

closed 2016-01-03 by andig