volkszaehler/vzlogger #229: vzlogger generiert Impuls bei Neustart

winfriedp, 2016-01-03:
In meiner S0-Impulse Konfiguration generiert vzlogger beim Neustart immer ein Reading vom Typ Impulse und value 1, ohne dass ein externer Impuls ausgelöst wurde. Dieser Fehler wurde in #222 erwähnt.

Output httpd-api:
{ "version": "0.4.8", "generator": "vzlogger", "data": [ { "uuid": "120", "last": 1451849762307, "interval": -1, "protocol": "s0", "tuples": [ [ 1451849762307, 1 ] ] } ] }

vzlogger.log:
[Jan 03 20:36:02][]     ===> Start meters
[Jan 03 20:36:02][s0]   counter_thread created
[Jan 03 20:36:02][mtr0] Meter connection established
[Jan 03 20:36:02][mtr0] Meter thread started
[Jan 03 20:36:02][mtr0] Meter is opened. Starting channels.
[Jan 03 20:36:02][chn0] Logging thread started
[Jan 03 20:36:02][http] Starting local interface HTTPd on port 8080
[Jan 03 20:36:02][]     Startup done.
[Jan 03 20:36:02][mtr0] Number of readers: 4
[Jan 03 20:36:02][mtr0] Config.daemon: 1
[Jan 03 20:36:02][mtr0] Config.local: 1
[Jan 03 20:36:02][s0]   Counter thread started with blocking hwif
[Jan 03 20:36:02][S0]   MeterS0:HWIF_GPIO:first poll returned 1
[Jan 03 20:36:02][chn0] Start logging thread for null-api. Running as daemon: yes
[Jan 03 20:36:02][chn0] Using null api- meter data available via local httpd if enabled.
[Jan 03 20:36:03][s0]   Reading S0 - returning 1 readings (n=1 n_neg = 0)
[Jan 03 20:36:03][mtr0] Got 1 new readings from meter:
[Jan 03 20:36:03][mtr0] Reading: id=Impulse/StringItentifier: value=1.00 ts=1451849762307
[Jan 03 20:36:03][chn0] Adding reading to queue (value=1.00 ts=1451849762307)
[Jan 03 20:36:03][chn0] Buffer dump (size=0): {}
[Jan 03 20:36:03][S0]   MeterS0:HWIF_GPIO:first poll returned 0
[Jan 03 20:36:04][S0]   MeterS0:HWIF_GPIO:first poll returned 0
[Jan 03 20:36:05][S0]   MeterS0:HWIF_GPIO:first poll returned 0

vzlogger.conf:
{
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": true,
    "port": 8080,
    "index": true,
    "timeout": 0,
    "buffer": 60
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "120",
          "identifier": "Impulse",
          "api": "null",
          "aggmode": "none"
        }
      ],
      "protocol": "s0",
      "gpio": 20,
      "configureGPIO": true,
      "send_zero": false,
      "debounce_delay": 0
    }
  ]
}

Viele Grüße

mbehr1, 2016-01-04:
Siehe Fix in PR #230.

winfriedp, 2016-01-04:
Ich kann den Fix leider nicht testen. Daher überlasse ich anderen dieses Issue zu schliessen.
Viele Grüße

andig, 2016-01-05:
Is drin, git pull genügt.

closed 2016-01-05 by andig