volkszaehler/vzlogger #231: aggtime wird ignoriert wenn aggfixedinterval nicht gesetzt ist

andig, 2016-01-11:
Aus der Mailinglsite. Zu testen sind:

  - funktioniert `aggtime` wenn `send_zero` aktiviert ist?
  - log auf lvl15 bereitstellen

andig, 2016-05-06:
ping @justinotherguy for testing:

  - aggtime only
  - aggtime plus aggfixedinterval
  - aggtime plus sendzero

justinotherguy, 2016-05-06:
alright:
input signal: 100 Hz switched on for 10s and then off for 10s, start over
basic config: http://pastie.org/10827079

1. 
//aggtime
//aggfixedinterval
//sendzero
result:
# tail -f /var/log/vzlogger.log | grep "chn0.*JSON"
[May 06 15:12:55][chn0] JSON request body: [ [ 1462540375394, 99 ] ]
[May 06 15:12:56][chn0] JSON request body: [ [ 1462540376397, 99 ] ]
[May 06 15:12:57][chn0] JSON request body: [ [ 1462540377399, 100 ] ]
[May 06 15:12:58][chn0] JSON request body: [ [ 1462540377580, 99.999999999999986 ] ]
[May 06 15:13:08][chn0] JSON request body: [ [ 1462540388392, 18 ] ]
[May 06 15:13:09][chn0] JSON request body: [ [ 1462540389394, 81 ] ]
[May 06 15:13:10][chn0] JSON request body: [ [ 1462540390397, 100 ] ]
[May 06 15:13:11][chn0] JSON request body: [ [ 1462540391399, 100 ] ]
[May 06 15:13:12][chn0] JSON request body: [ [ 1462540392392, 100 ] ]
[May 06 15:13:13][chn0] JSON request body: [ [ 1462540393394, 99 ] ]
[May 06 15:13:14][chn0] JSON request body: [ [ 1462540394397, 100 ] ]
[May 06 15:13:15][chn0] JSON request body: [ [ 1462540395400, 100 ] ]
[May 06 15:13:16][chn0] JSON request body: [ [ 1462540396392, 100 ] ]
[May 06 15:13:17][chn0] JSON request body: [ [ 1462540397395, 99 ] ]
[May 06 15:13:18][chn0] JSON request body: [ [ 1462540397605, 100 ] ]
[May 06 15:13:28][chn0] JSON request body: [ [ 1462540408397, 21 ] ]
[May 06 15:13:29][chn0] JSON request body: [ [ 1462540409400, 79 ] ]
[May 06 15:13:30][chn0] JSON request body: [ [ 1462540410402, 100 ] ]
[May 06 15:13:31][chn0] JSON request body: [ [ 1462540411395, 100 ] ]
[May 06 15:13:32][chn0] JSON request body: [ [ 1462540412398, 99 ] ]
[May 06 15:13:33][chn0] JSON request body: [ [ 1462540413400, 100 ] ]
[May 06 15:13:34][chn0] JSON request body: [ [ 1462540414403, 100 ] ]

-> until 15:12:58: every second one request is logged
-> then a pause of 10s (until 15:13:08)
-> start over
-> no aggregation, logging every second, no logging of zeroes
-> everything as expected

2.
aggtime: 2
//aggfixedinterval
//sendzero
result:
[May 06 15:24:29][chn0] JSON request body: [ [ 1462541069806, 100 ] ]
[May 06 15:24:30][chn0] JSON request body: [ [ 1462541070808, 100 ] ]
[May 06 15:24:31][chn0] JSON request body: [ [ 1462541071811, 100 ] ]
[May 06 15:24:32][chn0] JSON request body: [ [ 1462541072803, 99 ] ]
[May 06 15:24:33][chn0] JSON request body: [ [ 1462541073806, 100 ] ]
[May 06 15:24:34][chn0] JSON request body: [ [ 1462541074809, 100 ] ]
[May 06 15:24:35][chn0] JSON request body: [ [ 1462541075811, 100 ] ]
[May 06 15:24:36][chn0] JSON request body: [ [ 1462541076804, 99 ] ]
[May 06 15:24:38][chn0] JSON request body: [ [ 1462541078478, 167 ] ]
[May 06 15:24:48][chn0] JSON request body: [ [ 1462541088809, 33 ] ]
[May 06 15:24:49][chn0] JSON request body: [ [ 1462541089811, 100 ] ]
[May 06 15:24:50][chn0] JSON request body: [ [ 1462541090804, 99 ] ]
[May 06 15:24:51][chn0] JSON request body: [ [ 1462541091806, 100 ] ]
[May 06 15:24:52][chn0] JSON request body: [ [ 1462541092809, 100 ] ]
[May 06 15:24:53][chn0] JSON request body: [ [ 1462541093812, 100 ] ]
[May 06 15:24:54][chn0] JSON request body: [ [ 1462541094804, 90 ] ]
[May 06 15:24:55][chn0] JSON request body: [ [ 1462541095807, 100 ] ]
[May 06 15:24:56][chn0] JSON request body: [ [ 1462541096809, 100 ] ]
[May 06 15:24:58][chn0] JSON request body: [ [ 1462541098504, 169 ] ]
[May 06 15:25:08][chn0] JSON request body: [ [ 1462541108804, 30 ] ]
[May 06 15:25:09][chn0] JSON request body: [ [ 1462541109807, 100 ] ]
[May 06 15:25:10][chn0] JSON request body: [ [ 1462541110809, 100 ] ]
[May 06 15:25:11][chn0] JSON request body: [ [ 1462541111812, 97 ] ]
[May 06 15:25:12][chn0] JSON request body: [ [ 1462541112805, 99 ] ]
[May 06 15:25:13][chn0] JSON request body: [ [ 1462541113807, 100 ] ]
[May 06 15:25:14][chn0] JSON request body: [ [ 1462541114810, 100 ] ]
[... 130 lines cut ...]
[May 06 15:38:24][chn0] JSON request body: [ [ 1462541902000, 0 ] ]
[May 06 15:38:26][chn0] JSON request body: [ [ 1462541904000, 0 ] ]
[May 06 15:38:28][chn0] JSON request body: [ [ 1462541906000, 0 ] ]
[May 06 15:38:30][chn0] JSON request body: [ [ 1462541910000, 72 ] ]
[May 06 15:38:32][chn0] JSON request body: [ [ 1462541912000, 199 ] ]
[May 06 15:38:34][chn0] JSON request body: [ [ 1462541914000, 199 ] ]
[May 06 15:38:36][chn0] JSON request body: [ [ 1462541916000, 199 ] ]
[May 06 15:38:38][chn0] JSON request body: [ [ 1462541918000, 200 ] ]
[May 06 15:38:40][chn0] JSON request body is null. Nothing to send now.
[May 06 15:38:42][chn0] JSON request body: [ [ 1462541920000, 0 ] ]
[May 06 15:38:44][chn0] JSON request body: [ [ 1462541922000, 0 ] ]
[May 06 15:38:46][chn0] JSON request body: [ [ 1462541924000, 0 ] ]
[May 06 15:38:48][chn0] JSON request body: [ [ 1462541926000, 0 ] ]
[May 06 15:38:50][chn0] JSON request body: [ [ 1462541930000, 70 ] ]
[May 06 15:38:52][chn0] JSON request body: [ [ 1462541932000, 199 ] ]
[May 06 15:38:54][chn0] JSON request body: [ [ 1462541934000, 200 ] ]
[May 06 15:38:56][chn0] JSON request body: [ [ 1462541936000, 199 ] ]

-> aggregation works fine, again
-> send_zero works fine
-> here the "JSON request body is null. Nothing to send now" lines come back (just to be sure, it's not just my impatience - I ran test 4 again and left it run for 2 mins. - none of these line appear; they seem to be related to aggfixedinterval

bottom line:
- aggregation does not seem to need aggfixedinteval
- aggregation seems to need send_zero to work

justinotherguy, 2016-05-06:
oh:
vzlogger -V
0.5.2
 based on git version: heads/master-0-gd98c44860c
 last commit date: Wed, 4 May 2016 07:19:44 +0200

andig, 2016-05-07:
@justinotherguy warum wird bei 1) jede Sekunde geloggt? Wo kommt der Wert her wenn nicht aus aggtime?

andig, 2016-05-07:
@mbehr1 könntest Du Dir das mal anschauen? 

  1. Vmtl. seit https://github.com/volkszaehler/vzlogger/commit/3b48c8b33c5bd87b14a8adf31f1a76a0c5fd7716 funktioniert `aggtime` nicht mehr wenn `send_zero` nicht gesetzt ist.

  2. Etwas unklarer- ab und an taucht

      ````[chn0] JSON request body is null. Nothing to send now.````

  im Logfiel auf, anscheinend nur wenn `aggfixedinterval` gesetzt ist. Das mag daran liegen das zum Zeitpunkt der abgelaufenen Intervalls keine Daten vorliegen?

mbehr1, 2016-05-07:
> Vmtl. seit 3b48c8b funktioniert aggtime nicht mehr wenn send_zero nicht gesetzt ist.

schau ich mir an. 

Gruß
Matthias 

Sent from a mobile device.

justinotherguy, 2016-05-07:
> @justinotherguy warum wird bei 1) jede Sekunde geloggt? Wo kommt der Wert her wenn nicht aus aggtime?

das ist eine gute Frage. Beantworten kann ich sie nicht :)
Wo kommt der Sekundentakt überhaupt her? Mit aggtime: 10, //"aggfixedinterval", "send_zero": true und "aggmode": "sum" sieht das (grep -E "Impulse/StringIdentifier|JSON") so aus:

[May 07 17:55:39][mtr0] Reading: id=Impulse/StringIdentifier: value=42.00 ts=1462636539434
[May 07 17:55:40][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636540434
[May 07 17:55:41][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636541434
[May 07 17:55:42][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636542434
[May 07 17:55:43][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636543435
[May 07 17:55:44][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636544435
[May 07 17:55:45][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636545435
[May 07 17:55:46][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636546435
[May 07 17:55:47][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636547435
[May 07 17:55:48][mtr0] Reading: id=Impulse/StringIdentifier: value=0.00 ts=1462636548435
[May 07 17:55:48][chn0] JSON request body: [ [ 1462636548435, 42 ] ]
[May 07 17:55:49][mtr0] Reading: id=Impulse/StringIdentifier: value=8.00 ts=1462636549594
[May 07 17:55:50][mtr0] Reading: id=Impulse/StringIdentifier: value=49.00 ts=1462636550576
[May 07 17:55:51][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636551577
[May 07 17:55:52][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636552581
[May 07 17:55:53][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636553579
[May 07 17:55:54][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636554581
[May 07 17:55:55][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636555582
[May 07 17:55:56][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636556583
[May 07 17:55:57][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636557585
[May 07 17:55:58][mtr0] Reading: id=Impulse/StringIdentifier: value=50.00 ts=1462636558586
[May 07 17:55:58][chn0] JSON request body: [ [ 1462636558586, 457 ] ]

-> die Zeilen "JSON request body" sind klar - Requests alle 10s wg. Aggregation; aber: warum kommen die anderen Zeilen im Sekundenabstand? Vielleicht beantwortet die Antwort auf die Frage auch andigs Frage?

mbehr1, 2016-05-07:
MeterS0 loggt im 1s Takt, falls Signale anliegen oder SendZero=true ist.
(Habe ich so implementiert. ;-)

> @justinotherguy <https://github.com/justinotherguy> warum wird bei 1) jede Sekunde geloggt? Wo kommt der Wert her wenn nicht aus aggtime?
>
> das ist eine gute Frage. Beantworten kann ich sie nicht :) [...]

Gruß

Matthias

andig, 2016-06-25:
VZ Meeting:

  - [ ] Justin retest with/without aggmode, aggtime=10, send_zero=off
  - [ ] add warning when aggtime set but aggmode empty

andig, 2016-08-05:
ping @justinotherguy could you run the missing retest? This bug is ugly :/

mbehr1, 2018-08-05:
Bzgl: add warning when agg time set but aggmode empty.
Habe mir das grad angeschaut. Das ist per-se eine gültige Konfig, da aggmode per Channel, aggtime aber über alle Channel ist. D.h. die Warnung, macht nur Sinn, wenn aggmode in allen Channel fehlt, oder? (ein default für aggmode macht ja auch kein Sinn, da max/avg,... nicht zu raten ist.)