volkszaehler/vzlogger #232: vzlogger crashes

andig, 2016-01-11:
Auf der User-Liste sind jetzt zwei Meldungen aufgelaufen, die von einem Absturz von vzlogger berichten:

    Erstmal lief er einige Stunden brav und stabil.
    Leider stirbt dieser nach einer bisher unbestimmten Zeit mit folgenden Meldungen anscheinend mehrere bevor der Prozess endgültig stirbt.

    Jan  3 22:26:15 raspberrypi kernel: [  178.023390] ------------[ cut here ]------------
    Jan  3 22:26:15 raspberrypi kernel: [  178.023466] WARNING: CPU: 0 PID: 437 at fs/sysfs/dir.c:31 sysfs_warn_dup+0x68/0x88()
    Jan  3 22:26:15 raspberrypi kernel: [  178.023519] sysfs: cannot create duplicate filename '/bus/w1/devices/28-000003c3a47a'
    Jan  3 22:26:15 raspberrypi kernel: [  178.023533] Modules linked in: rtc_ds1307 cfg80211 rfkill ftdi_sio usbserial spi_bcm2708 snd_soc_bcm2708_i2s regmap_mmio snd_soc_pcm512x_i2c snd_soc_wm8804_i2c snd_soc_pcm512x snd_soc_tas5713 snd_soc_wm8804 snd_soc_core snd_compress snd_pcm_dmaengine regmap_i2c w1_therm ds2482 wire cn i2c_dev i2c_bcm2708 snd_bcm2835 snd_pcm snd_timer snd fuse ipv6
    Jan  3 22:26:15 raspberrypi kernel: [  178.023685] CPU: 0 PID: 437 Comm: w1_bus_master2 Tainted: G        W       4.1.13+ #826
    Jan  3 22:26:15 raspberrypi kernel: [  178.023699] Hardware name: BCM2708
    Jan  3 22:26:15 raspberrypi kernel: [  178.023764] [<c0016d84>] (unwind_backtrace) from [<c0013b44>] (show_stack+0x20/0x24)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023808] [<c0013b44>] (show_stack) from [<c0587d94>] (dump_stack+0x20/0x28)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023851] [<c0587d94>] (dump_stack) from [<c00241d8>] (warn_slowpath_common+0x8c/0xc4)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023884] [<c00241d8>] (warn_slowpath_common) from [<c0024250>] (warn_slowpath_fmt+0x40/0x48)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023918] [<c0024250>] (warn_slowpath_fmt) from [<c01b4128>] (sysfs_warn_dup+0x68/0x88)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023957] [<c01b4128>] (sysfs_warn_dup) from [<c01b44a0>] (sysfs_do_create_link_sd+0xf0/0xf4)
    Jan  3 22:26:15 raspberrypi kernel: [  178.023989] [<c01b44a0>] (sysfs_do_create_link_sd) from [<c01b44d8>] (sysfs_create_link+0x34/0x44)
    Jan  3 22:26:15 raspberrypi kernel: [  178.024026] [<c01b44d8>] (sysfs_create_link) from [<c038e8c0>] (bus_add_device+0x10c/0x194)
    Jan  3 22:26:15 raspberrypi kernel: [  178.024061] [<c038e8c0>] (bus_add_device) from [<c038c990>] (device_add+0x364/0x540)
    Jan  3 22:26:15 raspberrypi kernel: [  178.024090] [<c038c990>] (device_add) from [<c038cb90>] (device_register+0x24/0x28)
    Jan  3 22:26:15 raspberrypi kernel: [  178.024181] [<c038cb90>] (device_register) from [<bf0f3b88>] (w1_attach_slave_device+0x1ec/0x3dc [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.024294] [<bf0f3b88>] (w1_attach_slave_device [wire]) from [<bf0f3ee0>] (w1_slave_found+0xc8/0xcc [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.024419] [<bf0f3ee0>] (w1_slave_found [wire]) from [<bf0f4720>] (w1_search+0x27c/0x290 [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.024689] [<bf0f4720>] (w1_search [wire]) from [<bf0f6c64>] (w1_search_devices+0x54/0x58 [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.024876] [<bf0f6c64>] (w1_search_devices [wire]) from [<bf0f47c4>] (w1_search_process_cb+0x90/0x134 [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.025000] [<bf0f47c4>] (w1_search_process_cb [wire]) from [<bf0f49d8>] (w1_process+0xf8/0x158 [wire])
    Jan  3 22:26:15 raspberrypi kernel: [  178.025077] [<bf0f49d8>] (w1_process [wire]) from [<c0041714>] (kthread+0xe0/0xfc)
    Jan  3 22:26:15 raspberrypi kernel: [  178.025118] [<c0041714>] (kthread) from [<c000f7d8>] (ret_from_fork+0x14/0x3c)
    Jan  3 22:26:15 raspberrypi kernel: [  178.025138] ---[ end trace 56f1a57bf8bab665 ]---

ich hab den "send_zero" versuchsweise auf true gesetzt. Leider bringt das keine Besserung, der vzlogger verabschiedet sich dann auch irgendwann bei mir. Habt dann wieder rausgenommen.

Kann es sein, dass sich da ein Bug eingeschlichen hat?

mbehr1, 2016-01-11:
Das sieht eher nach einem Kernel/Treiber-Problem aus (Racecondition im wire/w1_therm,… Treiber). Der Kernel meldet da ein richtiges Problem.
Ich könnte mir vorstellen, dass das auftritt, wenn die Spannungsversorgung einknickt und dann die W1 Sensoren kurzfristig ab-/angemeldet werden.
Zur Vermeidung könnte mal mal die Spannungsversorgung der Sensoren überprüfen (kleinerer Widerstand notwendig bei parasitäter Versorgung,…).
Wenn wir das reproduzieren könnten, sollten wir das an als Kernel/Treiber Bug melden.

> Auf der User-Liste sind jetzt zwei Meldungen aufgelaufen, die von einem Absturz von vzlogger berichten:
>
> Erstmal lief er einige Stunden brav und stabil.
> Leider stirbt dieser nach einer bisher [...]

Gruß

Matthias

andig, 2016-05-06:
> Das sieht eher nach einem Kernel/Treiber-Problem aus (Racecondition im wire/w1_therm,… Treiber). Der Kernel meldet da ein richtiges Problem. Ich könnte mir vorstellen, dass das auftritt, wenn die [...]

Closing due to missing feedback.

closed 2016-05-06 by andig