volkszaehler/vzlogger #236: S0 Impulse am GPIO werden bei Kontaktprellern regelmäßig falsch gewertet.

Clocker, 2016-01-26:
Das aktuell implementierte Debouncing in der MeterS0.cpp ist wirkungslos.
(siehe auch http://www.photovoltaikforum.com/volkszaehler-org-f131/wie-stromzaehler-mit-75-u-kwh-korrekt-einrichten-t110897-s40.html)

Ich habe den Fehler bei mir auf dem Raspberry korrigiert und würde jetzt gerne die korrigierte Version hier zur Verfügung stellen und mit Euch diskutieren.

Da ich bislang noch nicht mit GIT Hub gearbeitet habe (und auch noch nicht richtig firm mit Linux bin), wäre es mir recht, wenn mir jemand kurz unter die Arme greifen könnte.

Was muß ich auf dem Raspberry eingeben, um einen
1. Branch anzulegen (vorzugsweise bei gleicher Dateistruktur, wie im aktuellen Image)
2. Ich ändere die MeterS0.cpp
3. Wie bekomme ich die nach Github?
4. Wie gebe ich die Diskussion frei?
5. Merge into Master (Wer macht das?)

Ich würde nur ungerne durch Spielversuche ausversehen irgendwas zerschießen.

mbehr1, 2016-01-26:
Hallo,

was kaputt machen kannst du eigentlich nicht (außer auf deinem lokalen Branch ;-)

Branch anlegen:
git checkout -b <branchname> 
(ohne <>)
dann Änderung committen:
git commit -a -m "message"

dann pushen nach github. Das ist schwieriger. Ich glaube du musst daher erst bei github auf deinen github Account das Projekt "forken",... 
(müsste ich aus github Doku raussuchen. vielleicht hat es hier grad jemand parat)

Gruß
Matthias 

Sent from a mobile device.

andig, 2016-01-26:
> Was muß ich auf dem Raspberry eingeben, um einen

Minimale git doku lesen https://git-scm.com/book/en/v1/Getting-Started

mbehr1, 2016-02-05:
Mit merge von #237 erledigt, oder?

closed 2016-03-20 by andig