volkszaehler/vzlogger #238: Use MQTT for subscribing to external data sources and for providing data

bgewehr, 2016-01-29:
With MQTT as a meter type it would be possible to make vzlogger subscribe to sources of data which another device pushes to a MQTT broker. Taken this way, vzlogger would "read" the subscribed data like from any other input channel.

Other way round, vzlogger could publish the data from other meters to configured MQTT topics for sharing them with the rest of the world. Maybe the whole communication to the middleware could be changed to MQTT too. If done so it would be very very easy to connect Volkszaehler to other IOT devices and solutions.

bgewehr, 2016-01-29:
Most of the home automation solutions have MQTT bindings so there are lots of use cases out there.

andig, 2016-01-29:
There are MQTT libraries for PHP- as long as we have a middleware, this could be integrated with the push server.

bgewehr, 2016-01-29:
Yes, good idea, the frontend could be another message consumer of the live publishing vzlogger push server. One technology that works for many purposes...perfect!

tkuehne71, 2016-03-19:
Nice idea and it would be really good to have!

andig, 2016-03-19:
Quite simple:

  - install node-red
  - create flow with http endpoint
  - add endpoint as push server in vzlogger.conf
  - add conversion step and send via mqtt to flow
  - deploy and run flow
  - add flow ui if desired

bgewehr, 2016-03-19:
To connect to Outside MQTT by using a web socket listener that publishes to an MQTT broker would work for the one direction but not for the other. I'd prefer MQTT as a meter (subscribe to channel abc and log that) AND as kind of a middleware target for push data.

andig, 2016-03-19:
> To connect to Outside MQTT by using a web socket listener that publishes to an MQTT broker would work for the one direction but not for the other

Why? Read up on Mqtt?

andig, 2016-05-05:
See https://github.com/volkszaehler/volkszaehler.org/pull/430 for an approach how to do this and https://www.mail-archive.com/volkszaehler-users%40demo.volkszaehler.org/msg04924.html for the related discussion.

Can we close this issue then?

andig, 2016-05-15:
ping @bgewehr

andig, 2016-05-20:
Closed via https://github.com/volkszaehler/volkszaehler.org/pull/430 and https://github.com/Sineos/Push-Volkszaehler-Readings-to-Influxdb-via-MQTT

closed 2016-05-20 by andig

bgewehr, 2016-05-20:
Sorry, habe den Ping verpasst. Ich schau mir die Node Red Sache an, bin aber immer noch der Meinung, dass ein Meter Type MQTT in vzlogger interessant wäre. Ganz einfach subscriben und kommende Werte loggen. Wäre dann aber ein anderes Repo...

andig, 2016-05-20:
> Ganz einfach subscriben und kommende Werte loggen

@bgewehr so funktioniert es m.W. einfach nicht weil Du dafür ohnehin einen MQTT Server brauchst. Der vzlogger hat die Rolle Client, nicht Server! Und wenn der eh aufgesetzt werden muss dann ist das bißchen node-red auch schnell gemacht. Im Zweifel könnten wir das ja auch im Image mit ausliefern.

bgewehr, 2016-05-24:
Ja, klar braucht es einen MQTT Server, aber apt-get install mosquitto reicht dafür aus. Ich hab schon mehrere Dinge auf sehr angenehm einfache Weise mit MQTT umsetzen können.

Der nächste Schritt wäre für mich aber folgender:

Ich nutze die Volkszaehler Daten für diverse Zwecke, bin aber aktuell auf 10s scheduled pull von fhem angewiesen, was unnötige Systemlast verursacht.

Ein push to MQTT on changed data würde es mir ersparen, tausende Nullen von der Middleware pro Tag abzufragen...

Dann würde z.B. dieses Display nur bei tatsächlichen Wertänderungen aufgefrischt:

[image]

andig, 2016-05-24:
> Ein push to MQTT on changed data würde es mir ersparen, tausende Nullen von der Middleware pro Tag abzufragen...

Alle Komponenten dafür sind da. Du must einzig die Websocket Nachricht noch in eine MQTT Nachricht verpacken. node-red ist Den Freund!

bgewehr, 2016-05-24:
Ich nutze nicht nur die Impulse oder direkt vom Meter gelieferte Momentanwerte, sondern auch den durch die Middleware berechneten aktuellen Durchfluss und den Verbrauch der letzten 60 Minuten. Kann ich auch diese Daten per Node Red generieren?

andig, 2016-05-24:
> Ich nutze nicht nur die Impulse oder direkt vom Meter gelieferte Momentanwerte, sondern auch den durch die Middleware berechneten aktuellen Durchfluss und den Verbrauch der letzten 60 Minuten. Kann ich auch diese Daten per Node Red generieren?

Wie in der im PR verlinkten Doku geschrieben - ja! Der Pushserver liefert _immer_ konvertierte Werte wie sie auch das Frontend versteht.