volkszaehler/vzlogger #239: Kein 'oms'-Protocol in 0.5.1?

UdoSchake, 2016-01-30:
In der letzten Version von vzlogger scheint das OMS-Protocol nicht vorhanden zu sein:

root@minibian:/etc# vzlogger
[Jan 30 12:04:49][main] vzlogger v0.5.1 based on heads/master-0-gf4a1f98bbf from Tue, 5 Jan 2016 08:10:50 +0100 started.
[Jan 30 12:04:49][mtr0] Creating new meter with protocol oms.
[Jan 30 12:04:49][mtr0] Missing protocol or invalid type (Protocol not found.)
[Jan 30 12:04:49][]     parse configuration failed due to:
[Jan 30 12:04:49]       Failed to parse configuration due to: Protocol not found.

Zugehörige vzlogger.conf:
{
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "local": {
    "enabled": false,
    "port": 8080,
    "index": true,
    "timeout": 0,
    "buffer": 0
  },
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "deine UUID",
          "identifier": "1.8.0",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "oms",
      "device": "/dev/ttyUSB0",
      "baudrate": 9600,
      "key": "dein Key",
      "mbus_debug": false
    }
  ]
}

Eine vzlogger.log wird nicht erstellt.

Gruß
Udo

schnello, 2016-01-30:
Hallo.

Hast du oms gezielt mitgebaut?

>  It requires openssl and libmbus >=0.8.0 from e.g. https://github.com/rscada/libmbus.git 
> (installation is easy:
> 
> git clone https://github.com/rscada/libmbus.git
> cd libmbus
> ./build.sh
> make install

Grüße

UdoSchake, 2016-01-30:
Öhm, nein.

Kann das nachträglich installiert werden oder muss danach vzlogger neu compiliert werden?

Gruß
Udo

andig, 2016-01-30:
Ist ENABLE_OMS beim compilieren gesetzt?

andig, 2016-01-30:
Muss neu compiliert werden.

UdoSchake, 2016-01-30:
Wo setzt man ENABLE_OMS?

UdoSchake, 2016-01-30:
ok, vzlogger muss neu compiliert werden.
ENABLE_OMS ist nicht notwendig.
Ich habe das Wiki mal aktualisiert.

Gruß
Udo

closed 2016-01-30 by UdoSchake