volkszaehler/vzlogger #240: Logfile Parsing

hg-smart, 2016-02-06:
Is it possible to read out from a log file using vzlogger with the follwing structure :

```
06 Feb 2016 07:51:38 +0000,AA,TEMP22.1
06 Feb 2016 07:51:48 +0000,AA,RHUM43.2
06 Feb 2016 07:51:48 +0000,AA,TEMP22.1
06 Feb 2016 07:51:58 +0000,AA,RHUM43.2
06 Feb 2016 07:51:58 +0000,AA,TEMP22.1
06 Feb 2016 07:52:08 +0000,AA,RHUM43.2
06 Feb 2016 07:52:08 +0000,AA,TEMP22.1
06 Feb 2016 07:52:18 +0000,AA,RHUM43.2
06 Feb 2016 07:52:18 +0000,AA,TEMP22.1
06 Feb 2016 07:52:18 +0000,AA,SLEEPING
06 Feb 2016 07:52:28 +0000,AA,RHUM43.2
06 Feb 2016 07:52:28 +0000,AA,TEMP22.1
06 Feb 2016 07:52:38 +0000,AA,TEMP22.1
06 Feb 2016 07:52:47 +0000,AA,TEMP22.1
06 Feb 2016 07:52:57 +0000,AA,RHUM43.2
06 Feb 2016 07:52:57 +0000,AA,TEMP22.1
```

The Logfile simply appends and writes a Temperature for sensor AA out, for temperature and humidity...

I am Stuck with this format, described in the config file
```
"format": "$i $v $t",
// $i => identifier, $v => value, $t => timestamp
```

Can anybody help me ?

Thx in Advance

andig, 2016-02-06:
I've honestly not had any success use the file parser. It is likely broken.

Sorry,
Andreas

hg-smart, 2016-02-06:
well i used grep and awk to arrange the format to ...
`AA   22.1`

just a one-liner... and got the following logs

```
[Feb 06 23:46:20][]     MeterFile::read: 32, 32
[Feb 06 23:46:20][]     MeterFile::read: 'AA 24.0'
[Feb 06 23:46:20][]     MeterFile::read: 24.000000, AA, -1.000000
[Feb 06 23:46:20][mtr0] Got 1 new readings from meter:
[Feb 06 23:46:20][mtr0] Reading: id=AA/StringItentifier: value=24.00 ts=1454798780867
[Feb 06 23:46:20][chn0] Buffer dump (size=0): {}
[Feb 06 23:46:20][mtr0] Next reading in 2 seconds
[Feb 06 23:46:22][]     MeterFile::read: 32, 32
[Feb 06 23:46:22][]     MeterFile::read: 'AA 24.0'
[Feb 06 23:46:22][]     MeterFile::read: 24.000000, AA, -1.000000
[Feb 06 23:46:22][mtr0] Got 1 new readings from meter:
[Feb 06 23:46:22][mtr0] Reading: id=AA/StringItentifier: value=24.00 ts=1454798782868
[Feb 06 23:46:22][chn0] Buffer dump (size=0): {}
[Feb 06 23:46:22][mtr0] Next reading in 2 seconds
```

but i do not see anything in the web interface (channel is set...)...  ???

andig, 2016-02-07:
See wiki/howto/debug? Whats your .conf file and logfile at level 15?

hg-smart, 2016-02-08:
well i switched to pimatic's log file parser, does not crash...

closed 2016-02-08 by hg-smart