volkszaehler/vzlogger #241: Problem mit S0-Impulse in vzlogger

UdoSchake, 2016-02-06:
In letzter Zeit ist mir aufgefallen, dass es vermehrt Probleme beim Loggen von S0-Impulsen gibt.
Zuletzt mit dem Betriebsstundenzähler von sollner.
Ich habe heute mal meinen Frequenzgenerator mit einer Frequenz von 1Hz direkt an GPIO 17 von einem Raspi2 angeschlossen.
Die vzlogger.conf sieht so aus:
 {
      "enabled": true,
      "allowskip": true,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "ed97a180-ba3d-11e5-a655-85d821f2a6a0",
          "identifier": "Impulse",
          "api": "volkszaehler",
          "middleware": "http://127.0.0.1/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "s0",
      "gpio": 17,
      "resolution": 1000,
      "configureGPIO": true,
      "debounce_delay": 0
    },
vzlogger ist die letzte Git-Version:
root@minibian:~# vzlogger -V
0.5.1
 based on git version: heads/master-0-g7e79f4c700
 last commit date: Fri, 5 Feb 2016 18:03:28 +0100

Aber das Problem bestand auch in vorhergehenden Versionen.

Im Frontend tauchen immer wieder Einbrüche und Überhöhungen auf:
[image]
Genau das berichten auch andere User.

Gruß
Udo

bgewehr, 2016-02-06:
Hast Du mal an den besonderen Stellen die Tinestamps verglichen? Der Abstand müsste ja immer eine Sekubde sein, oder?

Oder fehlen Impulse?

andig, 2016-02-06:
@UdoSchake kannst Du im git mal ein paar commits zurück gehen und rausfinden mit welchem commit der Fehler begann?

andig, 2016-02-06:
Der Graph sagt nixhts aus. Wir holen so ca 400 Werte ab, Du zeigst bei 1Hz eine Zeitspanne von ca. 4000 Werten an. Bitte mal auch etwa 5min rein zoomen.

UdoSchake, 2016-02-06:
Was mir auffällt ist ein Zeitsprung wenn der 'Fehler' auftritt:
[image]
[image]

UdoSchake, 2016-02-06:
[image]

UdoSchake, 2016-02-06:
[image]

andig, 2016-02-07:
Also: 

  - mal fix auf kleinen fehlerhaften Bereich gezoomt
  - Daten als csv exportiert und angeschaut:

	# source:;volkszaehler.org
	# version:;0.3

	# uuid:;ed97a180-ba3d-11e5-a655-85d821f2a6a0
	# title:;S0/2
	# from:;2016-02-07 10:03:48
	# to:;2016-02-07 10:04:08
	# min:;2016-02-07 10:03:57; => ;3157.895
	# max:;2016-02-07 10:04:03; => ;4260.355
	# average:;3628.301
	# consumption:;20
	# rows:;21
	2016-02-07 10:03:49;3600;1
	2016-02-07 10:03:50;3600;1
	2016-02-07 10:03:51;3600;1
	2016-02-07 10:03:52;3600;1
	2016-02-07 10:03:53;3603.604;1
	2016-02-07 10:03:54;3596.404;1
	2016-02-07 10:03:55;3600;1
	2016-02-07 10:03:56;3600;1
	2016-02-07 10:03:57;3157.895;1
	2016-02-07 10:03:58;4186.047;1
	2016-02-07 10:03:59;3600;1
	2016-02-07 10:04:00;3603.604;1
	2016-02-07 10:04:01;3600;1
	2016-02-07 10:04:02;3600;1
	2016-02-07 10:04:03;4260.355;1
	2016-02-07 10:04:04;3600;1
	2016-02-07 10:04:05;3600;1
	2016-02-07 10:04:06;3600;1
	2016-02-07 10:04:07;3600;1
	2016-02-07 10:04:08;3600;1

Offensichtlich gibts trotz anscheinendem Sekundenabstand Schwankungen in der Wertberechnung. Also Timestamps anschauen.

  - `/middleware.php/data.json?from=1454835829712&to=1454835847777&uuid[]=ed97a180-ba3d-11e5-a655-85d821f2a6a0&options=raw`

	{"version":"0.3","data":[{"tuples":[[1454835829344,1,1],
	[1454835830344,1,1],
	[1454835831344,1,1],
	[1454835832344,1,1],
	[1454835833343,1,1],
	[1454835834344,1,1],
	[1454835835344,1,1],
	[1454835836344,1,1],
	[1454835837484,1,1],
	[1454835838344,1,1],
	[1454835839344,1,1],
	[1454835840343,1,1],
	[1454835841343,1,1],
	[1454835842343,1,1],
	[1454835843188,1,1],
	[1454835844188,1,1],
	[1454835845188,1,1],
	[1454835846188,1,1],
	[1454835847188,1,1],
	[1454835848188,1,1]],"uuid":"ed97a180-ba3d-11e5-a655-85d821f2a6a0","from":1454835828344,"to":1454835848188,"average":0,"consumption":0,"rows":21}]}

Und siehe da: die timestamps haben nicht den erwarteten Sekundenabstand! Jetzt wäre die Frage also warum. 

@mbehr1 schickt vzlogger den Timestamp mit oder wird der von der MW erst ermittelt?

UdoSchake, 2016-02-07:
> Und siehe da: die timestamps haben nicht den erwarteten Sekundenabstand! Jetzt wäre die Frage
> also warum. 
Der von mir verwendete Signalgenerator dürfte da außen vor sein:
http://www.batronix.com/shop/waveform-generator/Siglent-SDG1010.html

Gruß
Udo

mbehr1, 2016-02-07:
Hallo Udo,

Raspberry Pi? Schreibst du Daten auf die SD-Karte? (der Treiber sperrt recht lange Interrupts)

Falls ja: versuch mal nicht auf die SD-Karte zu schreiben (zB ramdisk oder MySQL auf anderem Rechner,...)

Gruß
Matthias 

Sent from a mobile device.

UdoSchake, 2016-02-07:
> Raspberry Pi? Schreibst du Daten auf die SD-Karte?

Ich vermute ja.
Wie stelle ich das an: auf eine ramdisk zu schreiben?

Gruß
Udo

bgewehr, 2016-02-07:
> Am 07.02.2016 um 12:36 schrieb Matthias Behr <notifications@github.com>:
> 
> Hallo Udo,
> 
> Raspberry Pi? Schreibst du Daten auf die SD-Karte? (der Treiber sperrt recht lange Interrupts)
> 
> Falls ja: versuch mal nicht auf die SD-Karte zu schreiben (zB ramdisk oder MySQL auf anderem Rechner,...)
> 
> Gruß
> Matthias 
> 
Wie sieht es mit USB-HDD aus? Gleiches Thema?

Gruß
Bernd

UdoSchake, 2016-02-07:
Fällt mir noch zu ein: Bei einer Ramdisk sind die Daten nach einem Reboot weg.

Gruß
Udo

andig, 2016-02-07:
@mbehr1 schickt vzlogger den Timestamp mit oder wird der von der MW erst ermittelt?

mbehr1, 2016-02-07:
timestamp wird mitgeschickt.

> 
> @mbehr1 <https://github.com/mbehr1> schickt vzlogger den Timestamp mit oder wird der von der MW erst ermittelt?

Gruß

Matthias

andig, 2016-02-07:
> timestamp wird mitgeschickt.

Ok. In dem Fall wird der Impuls auf der Erfassungsseite zu spät erkannt. Das sollte eigentlich gehen auch wenn die MW noch mit speichern beschäftigt ist, oder blockt der CURL Transfer die Erfassung?

UdoSchake, 2016-02-07:
Ich vermute, wie Matthias schon gesagt hat, dass die Interrupts durch die SD-Karten-Zugriffe zu lange blockiert sind.
Habe jetzt 'verbosity' auf 0 gesetzt und siehe da der Graph sieht vernünftig aus:
[image]

ich würde ja gerne das Verzeichnis /var/log in tmpfs schieben, aber wenn ich das mache, funktioniert das Frontend nicht mehr und in Top gibt es keinen apache2 mehr.

Gruß
Udo

andig, 2016-02-07:
Fein- dann können wir das issue schließen?

UdoSchake, 2016-02-07:
Nicht ganz: irgendwie brauchen wir eine Lösung um die Zugriffe durch den log-file auf die SD-Karte zu verhindern.

Gruß
Udo

UdoSchake, 2016-02-07:
ok, ich habe eine Lösung:
das /tmp-Verzeichnis wird in tmpfs geschoben:
Dazu in der /etc/fstab folgende Zeile einfügen:
`tmpfs /tmp     tmpfs nodev,nosuid,mode=1777,size=30M              0 0`
In der vzlogger.conf den Pfad für die vzlogger.log auf
`/tmp/vzlogger.log`
setzen.
Danach reboot.

Gruß
Udo

closed 2016-02-07 by UdoSchake