volkszaehler/vzlogger #244: Unterschiede bei S0-Power und S0-Impuls

gitka, 2016-03-21:
in der HttpD-Ausgabe ist mir aufgefallen, dass es Unterschiede im Ausgabe-Ergebnis des selben S0-Eingangs gibt. Dazu lasse ich den selben  gpio-S0-Eingang über zwei vzlogger-channels, einmal mit "identifier" : "Impulse" und einmal mit "identifier" : "Power"  ausgeben. Die Impulse u. der dazugehörige Zeitstempel wird sofort ausgegeben. Der Leistungswert ("Power") erst einen Zeitstempel später.
Beispiel: Der errechnete Leistungswert 2595,15570 aus einem Impuls-Tupel (1458290656634) erscheint erst zum nächsten Zeitpunkt (1458290658381) im Power-Tupel.
[http://article.gmane.org/gmane.network.volkszaehler.user/6389](url)

Ob der Bug auch die MW betrifft, kann ich leider nicht beurteilen, da ich nur httpd einsetze.
Gruß
Karlheinz

andig, 2016-03-23:
Ping @mbehr ich weiss icht mal wo ich im Quelltext schauen soll....

mbehr1, 2016-03-23:
ich schau mir das an. 

Gruß
Matthias

Sent from a mobile device.

mbehr1, 2016-03-25:
hier der Code Auszug:
```
if (!_first_impulse) {
			double value = (3600000 / ((t2-t1) * _resolution)) * t_imp;
			rds[ret].identifier(new StringIdentifier("Power"));
			rds[ret].time(req);
			rds[ret].value(value);
			++ret;
		}
		rds[ret].identifier(new StringIdentifier("Impulse"));
		rds[ret].time(req);
		rds[ret].value(t_imp);
		++ret;
```
Die Logik ist:
erst beim 2. Impuls wird aus dem zeitl. Abstand die Leistung berechnet. Beim ersten Impuls fehlt ja die zeitl. Distanz/Angabe.
Aus meiner Sicht ist das beobachtete Verhalten kein Fehler, sondern so Planstand/gewollt.

andig, 2016-03-26:
> Die Logik ist:

Korrekt. Der Screenshot (Link gefixt) zeigt aber dass der richtige Leistungswert beim httpd dem falsche Timestamp zugeordnet ist. Oben ist es die Requestzeit (was ja richtig wäre).

 Gibts da noch irgendwo eine weitere Logik?

mbehr1, 2016-03-26:
Hmm. Kann ich mal config (Annahme: send-zero false, blocking hwif) und Log Auszug bekommen?

gitka, 2016-03-26:
Unter https://gist.github.com/gitka/889b8b7204691feece4c
habe ich meine vzlogger.conf gespeichert. Unter UUID e2302 lese ich den "Impulse" und unter e23 den selben Port als "Power".
Hier die httpd-Ausgabe passend zum Screenshot im Forum:
https://gist.github.com/gitka/0cfc59ce364024abd3b4
Log habe ich zu dem Zeitpunkt nicht aufgezeichnet. Könnte ich natürlich nachholen wenn du das noch brauchst.
vzlogger läuft dabei schon länger - es ist also nicht das Thema mit dem ersten Impuls.

andig, 2016-03-29:
Könntest Du das Log noch mit Lvl 15 erzeugen? Max. 3 Ausgabewerte sollten ausreichen um das Phänomen zu erkennen.

andig, 2016-03-29:
Ps.: die Conf passt auch nicht zur Ausgabe- oder habe ich die uuid 180 in der conf übersehen?

mbehr1, 2016-04-01:
Hast du bei e23 mal das aggmode AVG abgeschaltet?
Das macht zumindest die Analyse ein wenig komplizierter.

Gruß
Matthias

   "aggtime": 10,    // aggregate all signals for x seconds 18.11.15 22:38
//    "aggfixedinterval": true, // round all timestamps to middleware to nearest aggtime
    "device": "/sys/class/gpio/gpio17/value",
    "resolution": 800,
    "channel": {
	"api" : "NULL" ,
	"uuid" : "e23",     // nur Buchstaben von a - f u. Zahlen erlaubt
	"type": "device",
	"scaler": 1,        // neu
        "identifier" : "Power", /* Power oder Impulse bei S0 */
//        "aggmode": "none",  // 18.11.15 22:38
	"aggmode": "AVG",     // SUM, AVG; MAX

> 
> Ps.: die Conf passt auch nicht zur Ausgabe- oder habe ich die uuid 180 in der conf übersehen?
> 
> > 
> > Unter https://gist.github.com/gitka/889b8b7204691feece4c
> > habe ich meine vzlogger.conf gespeichert. Unter UUID e2302 lese ich den "Impulse" und unter e23 den selben Port als "Power".
> > Hier die httpd-Ausgabe passend zum Screenshot im Forum:
> > https://gist.github.com/gitka/0cfc59ce364024abd3b4
> > Log habe ich zu dem Zeitpunkt nicht aufgezeichnet. Könnte ich natürlich nachholen wenn du das noch brauchst.
> > vzlogger läuft dabei schon länger - es ist also nicht das Thema mit dem ersten Impuls.

Gruß

Matthias

mbehr1, 2016-04-01:
After having a look at the code for AVG I think it’s due to that. AVG seem to shift the values by one 1 timeslot. (Didn’t think about it yet whether thats ok/needed or not.)

> 
> Hast du bei e23 mal das aggmode AVG abgeschaltet?
> Das macht zumindest die Analyse ein wenig komplizierter.
> 
> Gruß
> Matthias
> 
>    "aggtime": 10,    // aggregate all signals for x seconds 18.11.15 22:38
> //    "aggfixedinterval": true, // round all timestamps to middleware to nearest aggtime
>     "device": "/sys/class/gpio/gpio17/value",
>     "resolution": 800,
>     "channel": {
> 	"api" : "NULL" ,
> 	"uuid" : "e23",     // nur Buchstaben von a - f u. Zahlen erlaubt
> 	"type": "device",
> 	"scaler": 1,        // neu
>         "identifier" : "Power", /* Power oder Impulse bei S0 */
> //        "aggmode": "none",  // 18.11.15 22:38
> 	"aggmode": "AVG",     // SUM, AVG; MAX
> 
> 
>> 
>> Ps.: die Conf passt auch nicht zur Ausgabe- oder habe ich die uuid 180 in der conf übersehen?
>> 
>> > 
>> > Unter https://gist.github.com/gitka/889b8b7204691feece4c <https://gist.github.com/gitka/889b8b7204691feece4c>
>> > habe ich meine vzlogger.conf gespeichert. Unter UUID e2302 lese ich den "Impulse" und unter e23 den selben Port als "Power".
>> > Hier die httpd-Ausgabe passend zum Screenshot im Forum:
>> > https://gist.github.com/gitka/0cfc59ce364024abd3b4 <https://gist.github.com/gitka/0cfc59ce364024abd3b4>
>> > Log habe ich zu dem Zeitpunkt nicht aufgezeichnet. Könnte ich natürlich nachholen wenn du das noch brauchst.
>> > vzlogger läuft dabei schon länger - es ist also nicht das Thema mit dem ersten Impuls.
>> > 
>> > 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

andig, 2016-04-01:
Then lets fix avg similar to what was done for power without avg?

mbehr1, 2016-04-01:
Let’s confirm this first and then let us consider/discuss whether this is a problem or needed behavior.

> 
> Then lets fix avg similar to what was done for power without avg?

Gruß

Matthias

gitka, 2016-04-01:
hab jetzt mit und ohne AVG getestet:
ohne AVG: Impuls und Power ist gleich
mit AVG: nicht gleich (sollte ja auch nicht sein, wegen 10 Sekunden-AVG), aber versetzt um einen Tupel wäre nicht OK
Wie Matthias schon vermutet hat, scheint beim AVG etwas nicht zu stimmen. 
AVG sollte aber im Prinzip schon funktionieren - oder?
Bei kleinen AVG-Werten stelle ich mir die Berechnung etwas schwierig vor.
Karlheinz

Unter https://gist.github.com/gitka/f0e8297fed3a0a716e9e noch ein vzlogger.log mit verbosity 15

mbehr1, 2016-04-03:
Habe mir AVG noch mal angeschaut. Und insbesondere die Unit Tests (siehe ut_buffer.cpp).

AVG habe ich damals bewusst so implementiert.

// 2nd case: prev. value (1.0 at 1s), just one new data (2.0 at 2s)-> return 1.0 as AVG (2.0 has no time yet!)

Hintergrund ist genau eine Implementierung für wenige Werte. D.h. die AV betrachtet, wie lange der letzte Wert angelegen hat. Das geht aber erst, wenn der nächste Wert vorliegt, sonst würde der AVG mit jeder Sekunde wieder schwanken und beim neuen Impuls wieder nach oben gehen.

Ich meine, wir hatten damals längere Diskussionen zur Implementierung von AVG.

—> Ich sehe hier keinen Fehler und würde die Implementierung so unverändert lassen.

> hab jetzt mit und ohne AVG getestet:
> ohne AVG: Impuls und Power ist gleich
> mit AVG: nicht gleich (sollte ja auch nicht sein, wegen 10 Sekunden-AVG), aber versetzt um einen Tupel wäre nicht OK
> Wie [...]

Gruß

Matthias

mbehr1, 2016-04-03:
Ist es ok, wenn ich den Punkt schließe?

gitka, 2016-04-03:
AVG mit 10s nutz ich, da bei hohen Leistungswerten (mehre S0 pro Zeiteinheit) das Ergebnis zu ungenau ist. Bei zu wenig Werten für AVG, kann ich das nicht nachvollziehen: "AVG seem to shift the values by one 1 timeslot.". Bedeutet das, dass bei Reduzierung von hoher Leistung auf geringe Leistung, der erste Wert verloren geht? 

Nach ca. 1-2 Tagen, läuft mit der jetzigen AVG-Berechnung ein errechner Zählerstand, zum echten Zählerstand auseinander. Ohne AVG bleiben die Zählerstände (nach ca. 2 Tagen) synchron. 
Das dürfte meiner Meinung nach nicht sein.

mbehr1, 2016-04-03:
Schick bitte mal Beispiel-Daten.
Auch bei AVG dürfte das nicht auseinander laufen, sondern der letzte gemessene Wert wird dann erst nach 10s mit einberechnet. Das führt aber zu keinem auseinander laufen.

> AVG mit 10s nutz ich, da bei hohen Leistungswerten (mehre S0 pro Zeiteinheit) das Ergebnis zu ungenau ist. Bei zu wenig Werten für AVG, kann ich das nicht nachvollziehen: "AVG seem to shift the [...]

Gruß

Matthias

gitka, 2016-04-03:
Tabelle und Formatierung ist in github nicht so einfach. 
Die AVG Logik verstehe ich noch nicht: Bei 10s hätte doch gar kein Durchschnitt berechnet werden dürfen, da die Zeit_diff nicht so groß war - oder?
Die 4 linken Spalten ohne AVG - rechte Spalten mit AVG. 
Auch die zusammengefassten Werte passen nicht. Z.B.: In der 4+5 Zeile wird mit AVG eine Leistung von 599,128 W ausgegeben (=8995907,92936 Wms
Wms). Es müssten aber 599,4006 W sein, damit die gleiche Arbeit von 4500000,07297 + 4500000,00688 = 9000000,07985 Wms rauskommt.

Am Schluss siehst du die nach meiner Rechnung die Differenz über 35031,71213 Wms.

**Zeitstempel e23a0 (ohne AVG)**|Power lt. Httpd |Zeit_diff in ms|Power*Zeit_diff in Wms||**Zeitstempel e23 mit AVG 10s**|Power lt. Httpd |Zeit_diff in ms|Power*Zeit_diff in Wms
-------------|-----------|-----|------------|---|-------------|-----------|-----|------------
1459541512912|598,8820916||#BEZUG!||1459541512913|603,0144611||
1459541520385|602,1678001|7473|4499999,97014||1459541520385|598,8024058|7472|4474251,57650
1459541527879|600,4803877|7494|4500000,02520||1459541527879|602,2483842|7494|4513249,39093
1459541535407|597,7683413|7528|4500000,07297|||||
1459541542894|601,0418067|7487|4500000,00688||1459541542894|599,1280672|15015|8995907,92936
1459541550394|600|7500|4500000,00000||1459541550394|601,0418067|7500|4507813,55037
1459541557922|597,7683223|7528|4499999,93045||1459541557922|600|7528|4516800,00000
1459541565422|600|7500|4500000,00000|||||
1459541572874|603,8647422|7452|4500000,05874||1459541572875|598,8806539|14953|8955062,41743
1459541580327|603,7837057|7453|4499999,95854||1459541580326|603,7837057|7451|4498792,39113
1459541587751|606,1422394|7424|4499999,98497||1459541587752|603,9457811|7426|4484901,37045
1459541595231|601,6042765|7480|4499999,98853|||||
1459541602752|598,3246981|7521|4500000,05478||1459541602753|603,8257014|15001|9057989,34739
1459541610259|599,4405246|7507|4500000,01829||1459541610259|598,2451417|7506|4490428,03323
1459541617774|598,8023869|7515|4499999,93718||1459541617774|599,5203994|7515|4505395,80137
1459541625295|598,3246981|7521|4500000,05478|||||
1459541632836|596,7378231|7541|4499999,92374||1459541632837|598,5632095|15063|9016157,62494
1459541640316|601,6042765|7480|4499999,98853||1459541640316|596,6587068|7479|4462410,46847
1459541647796|601,6042765|7480|4499999,98853||1459541647797|601,6847098|7481|4501203,31433
1459541655131|613,4969493|7335|4500000,12287|||||
1459541662536|607,6974845|7405|4499999,87250||1459541662536|607,5812691|14739|8955140,32461
1459541669926|608,9309988|7390|4500000,08130||1459541669926|607,6974845|7390|4490884,41023
1459541677351|606,06061|7425|4500000,02890||1459541677352|608,9309988|7426|4521921,59726
1459541684776|606,06061|7425|4500000,02890|||||
1459541692167|608,848597|7391|4499999,98026||1459541692167|606,0604294|14815|8978785,26145
1459541699599|605,4897609|7432|4499999,90299||1459541699598|608,8486166|7431|4524354,07008
1459541707010|607,2055042|7411|4499999,99189||1459541707010|605,571256|7412|4488494,14960
1459541714428|606,6325169|7418|4500000,01041|||||
1459541721832|607,779575|7404|4499999,97334||1459541721832|606,8782747|14822|8995149,78774
1459541729278|604,3513409|7446|4500000,08415||1459541729278|607,779575|7446|4525526,71549
1459541736723|604,4325105|7445|4500000,04035||1459541736724|604,3513409|7446|4500000,08415
1459541744150|605,8973976|7427|4499999,97226|||||
1459541751596|604,3513215|7446|4499999,94006||1459541751596|605,1662565|14872|9000032,56650
1459541759055|603,2980396|7459|4500000,07710||1459541759055|604,3513215|7459|4507856,50724
1459541766507|603,8647229|7452|4499999,91477||1459541766508|603,2980396|7453|4496380,28886
1459541773953|604,3513409|7446|4500000,08415|||||
1459541781392|604,9200056|7439|4499999,92154||1459541781392|604,1079773|14884|8991543,13418
1459541788871|601,684729|7479|4500000,08837||1459541788871|604,9200056|7479|4524196,72177
1459541796345|602,0872376|7474|4500000,01378||1459541796345|601,684729|7474|4496991,66472
1459541803811|602,7323804|7466|4499999,95172|||||
1459541811263|603,8647229|7452|4499999,91477||1459541811264|602,4095279|14919|8987347,74671
||||179999999,95860|||||179964968,24648
||||||**Differenz:**|35031,71213 Wms

andig, 2016-04-04:
Wenn ichs richtig lese dann erzeugt Dein Zähler alle 7,5s einen Impuls/Messwert. Jeder Impuls hat einen Wert von 4500Ws wie aus der Spalte ohne AVG ersichtlich ist.

Was mir auffällt:

  1. Zeile 1 enthält _überhaupt_ einen Power Wert der beim ersten Timestamp gar nicht existieren dürfte- ich nehme an das Log ist nicht ab Start?
  2. Die Leistungswerte bei AVG sind tatsächlich alle 1 Timestamp verschoben- das sollte nicht der Fall sein. Allerdings ist mir nicht klar ab welchem Zeitpunkt das AVG Intervall jeweils läuft. Nehmen wir mal an ab Start des Logs.
  3. Jetzt wirds spannend. Was passiert zum zeitpunkt 10 Sekunden wenn das AVG Intervall ausläuft? Zu diesem Zeitpunkt haben wir 2 Werte aus Timestamp 1 (0s) und Timestamp 2 (7,5s)? Die Middleware würde jetzt alle Impulse im Aggregationsintervall nehmen (also 1 Impuls bei 7,5s) und durch den gemessenen Zeitraum zwischen letztem Impuls der Vorperiode und letztem Impuls der aktuellen Periode dividieren. Das wären wie beim httpd dann 602W, nicht 599. Durch die Vorgehensweise Summierung Impulse dividiert durch Zeitintervall wird sichergestellt dass wirklich ein gewichteter Mittelwert für AVG gebildet wird.

@mbehr1 könntest Du nochmal einen Pointer geben wo der Durchschnitt berechnet wird? Power selbst ist erstmal prinzipiell richtig- was passiert bei AVG?

gitka, 2016-04-04:
------- Original Nachricht --------
Betreff: Re: [volkszaehler/vzlogger] Unterschiede bei S0-Power und 
S0-Impuls (#244)
Von: Andig

> Wenn ichs richtig lese dann erzeugt Dein Zähler alle 7,5s einen 
> Impuls/Messwert. Jeder Impuls hat einen Wert von 4500Ws wie aus der 
> Spalte ohne AVG ersichtlich ist.
Ich teste das ganze mit dem S0-Anschluss vom Haushaltszähler. Zufällig 
war die Leistung während des Ausschnitts sehr konstant bei ca. 600W.
>
> Was mir auffällt:
>
>  1. Zeile 1 enthält /überhaupt/ einen Power Wert der beim ersten
>     Timestamp gar nicht existieren dürfte- ich nehme an das Log ist
>     nicht ab Start?
Richtig, das war nicht ab Start.
>
>  1. Die Leistungswerte bei AVG sind tatsächlich alle 1 Timestamp
>     verschoben- das sollte nicht der Fall sein. Allerdings ist mir
>     nicht klar ab welchem Zeitpunkt das AVG Intervall jeweils läuft.
>     Nehmen wir mal an ab Start des Logs.
>  2. Jetzt wirds spannend. Was passiert zum zeitpunkt 10 Sekunden wenn
>     das AVG Intervall ausläuft? Zu diesem Zeitpunkt haben wir 2 Werte
>     aus Timestamp 1 (0s) und Timestamp 2 (7,5s)? Die Middleware würde
>     jetzt alle Impulse im Aggregationsintervall nehmen (also 1 Impuls
>     bei 7,5s) und durch den gemessenen Zeitraum zwischen letztem
>     Impuls der Vorperiode und letztem Impuls der aktuellen Periode
>     dividieren. Das wären wie beim httpd dann 602W, nicht 599. Durch
>     die Vorgehensweise Summierung Impulse dividiert durch
>     Zeitintervall wird sichergestellt dass wirklich ein gewichteter
>     Mittelwert für AVG gebildet wird.
Verständnisfrage: Bei einem AVG von 10 Sekunden, müsste es zwischen den 
Werten nicht mindestens 10S Abstand geben? (Die Statistik-Vorlesung über 
gleitenden Durchschnitt ist schon zu lange her :-) )

mbehr1, 2016-04-04:
Der AVG Code ist hier:
Buffer::aggregate (Buffer.cpp)

und wird aufgerufen von hier:
reading_thread (threads.cpp).

AVG kann nicht alle 10s liefern, sondern alle >=10s. Bei jeder Rückkehr von mtr->read(…) wird geprüft, ob die aggtime überschritten ist. Daher sind ja nach Dauer des read Calls (manchmal blocking) die Werte nicht äquidistant.

> 
> Wenn ichs richtig lese dann erzeugt Dein Zähler alle 7,5s einen Impuls/Messwert. Jeder Impuls hat einen Wert von 4500Ws wie aus der Spalte ohne AVG ersichtlich ist.
> 
> Was mir auffällt:
> 
> Zeile 1 enthält überhaupt einen Power Wert der beim ersten Timestamp gar nicht existieren dürfte- ich nehme an das Log ist nicht ab Start?
> Die Leistungswerte bei AVG sind tatsächlich alle 1 Timestamp verschoben- das sollte nicht der Fall sein. Allerdings ist mir nicht klar ab welchem Zeitpunkt das AVG Intervall jeweils läuft. Nehmen wir mal an ab Start des Logs.
> Jetzt wirds spannend. Was passiert zum zeitpunkt 10 Sekunden wenn das AVG Intervall ausläuft? Zu diesem Zeitpunkt haben wir 2 Werte aus Timestamp 1 (0s) und Timestamp 2 (7,5s)? Die Middleware würde jetzt alle Impulse im Aggregationsintervall nehmen (also 1 Impuls bei 7,5s) und durch den gemessenen Zeitraum zwischen letztem Impuls der Vorperiode und letztem Impuls der aktuellen Periode dividieren. Das wären wie beim httpd dann 602W, nicht 599. Durch die Vorgehensweise Summierung Impulse dividiert durch Zeitintervall wird sichergestellt dass wirklich ein gewichteter Mittelwert für AVG gebildet wird.
> @mbehr1 <https://github.com/mbehr1> könntest Du nochmal einen Pointer geben wo der Durchschnitt berechnet wird? Power selbst ist erstmal prinzipiell richtig- was passiert bei AVG?

Gruß

Matthias

andig, 2016-04-05:
> AVG kann nicht alle 10s liefern, sondern alle >=10s. Bei jeder Rückkehr von mtr->read(…) wird geprüft, ob die aggtime überschritten ist. Daher sind ja nach Dauer des read Calls (manchmal blocking) die Werte nicht äquidistant.

Verstanden. Damit hat sich auch mein Punkt 2) erledigt.

> Richtig, das war nicht ab Start.

@gitka Könntest Du das Log und die Tabelle nochmal- gerne bei schwankender Last- ab Start erzeugen? Ich finds extrem schwierig zu raten in welchem Zustand sich die Aggregation jeweils befindet. Vom Code her sieht das für mich erstmal logisch aus und eine Verschiebung der Timestamps ist nicht erkennbar.

Allerdings erscheint mir das Verhalten eindeutig falsch- der Durchschnitt aus z.B. 600 und 604 Watt kann sicher nicht 599 sein.

gitka, 2016-04-08:
Diesmal beginnen die Daten direkt ab Neustart. Zwischendurch ist der Wasserkocher gelaufen. Der erste Zeitstempel kommt mir seltsam vor.

*Zeitstempel e23 (ohne AVG)*|Power lt. Httpd |Zeit_diff in ms|Power*Zeit_diff in Wms||Zeitstempel e23a (mit AVG)|Power lt. Httpd |Zeit_diff in ms|Power*Zeit_diff in Wms
--------------|------------|-----|-------------- |---|--------------|------------|-----|-------------- |
1460123563006|261,2936959||#BEZUG!||1460123563005|261,2936959||
1460123580428|258,2941119|17422|4500000,01774||1460123580427|261,2936959|17422|4552258,76958
1460123586884|697,0260126|6456|4499999,93751||1460123586884|258,2941084|6457|1667805,05779
1460123589025|2101,821547|2141|4499999,93185||1460123589025|696,9180973|2141|1492101,64630
1460123591179|2089,136448|2154|4499999,90835||1460123591179|2101,821547|2154|4527323,61196
1460123593343|2079,482407|2164|4499999,92861|||||
1460123595504|2082,369261|2161|4499999,97220|||||
1460123597673|2074,688882|2169|4500000,18599||1460123597672|2083,66384|6493|13529229,31124
1460123599841|2075,645773|2168|4500000,03563|||||
1460123602016|2068,965563|2175|4500000,09866|||||
1460123604184|2075,645773|2168|4500000,03563|||||
1460123606360|2068,014593|2176|4499999,75545|||||
1460123608534|2069,91718|2174|4499999,94868||1460123608534|2072,783143|10862|22514570,50341
1460123610703|2074,688882|2169|4500000,18599|||||
1460123612871|2075,645773|2168|4500000,03563|||||
1460123615040|2074,688654|2169|4499999,69134|||||
1460123617221|2063,273733|2181|4500000,01181||1460123617221|2073,738968|8687|18014570,41640
1460123619397|2068,01482|2176|4500000,24850|||||
1460123621565|2075,645773|2168|4500000,03563|||||
1460123623740|2068,965563|2175|4500000,09866|||||
1460123625915|2068,965336|2175|4499999,60538|||||
1460123628077|2081,406162|2162|4500000,12307||1460123628076|2068,782374|10855|22456632,67450
1460123630259|2062,328213|2182|4500000,16128|||||
1460123632427|2075,645773|2168|4500000,03563|||||
1460123634595|2075,645773|2168|4500000,03563|||||
1460123636771|2068,014593|2176|4499999,75545||1460123636771|2073,777619|8695|18031496,39572
1460123638939|2075,645773|2168|4500000,03563|||||
1460123641114|2068,965563|2175|4500000,09866|||||
1460123643282|2075,645773|2168|4500000,03563|||||
1460123645458|2068,014593|2176|4499999,75545|||||
1460123647640|2062,328213|2182|4500000,16128||1460123647640|2071,259184|10869|22512516,07052
1460123649808|2075,645773|2168|4500000,03563|||||
1460123651990|2062,328213|2182|4500000,16128|||||
1460123654166|2068,014593|2176|4499999,75545|||||
1460123656341|2068,965563|2175|4500000,09866|||||
1460123658523|2062,328213|2182|4500000,16128||1460123658523|2067,465535|10883|22500227,41804
1460123660705|2062,327988|2182|4499999,66958|||||
1460123662887|2062,328213|2182|4500000,16128|||||
1460123665069|2062,328213|2182|4500000,16128|||||
1460123679855|304,3419448|14786|4499999,99594||1460123679855|2062,328034|21332|43993581,62110
1460123697311|257,7910161|17456|4499999,97689||1460123697311|304,3419448|17456|5312592,98857
1460123714686|258,9928058|17375|4500000,00000||1460123714685|257,7910196|17374|4478861,17474
1460123732060|259,0077116|17374|4499999,98123||1460123732059|259,0077116|17374|4499999,98123
||||189000000,02537|||||210083767,64109
||||||||
*|||Differenz:||-21083767,62 Wms|||*

andig, 2016-05-09:
Hallo @gitka ich hab mir das angeschaut und es macht nicht "klick":

  - Du hattest auf Unterschiede zwischen Power und Impulse hingewiesen- das kann ich mangels Darstellung der Impulse nicht nachvollziehen. Da Power errechnet wird wäre es sinnvoll bei der Analyse mit den Impulse anzufangen.
  - mir ist nicht klar wie und wann Du die httpd Werte ermittelst. Polls Du im ms-Abstand und stellst dann nur die aktualisierten dar?

Gibt es eine Möglichkeit (ich weiß, Arbeit...) die Impulse mal zusätzlich in die vz Middleware zu loggen? Genial wäre auch wenn Du die vom httpd abgeholten Werte dan gleich mit rein loggen würdest.

Das wäre dann der "golden master"- vz speichert immer die Originaldaten und auf der Basis ließen sich dann konsistente Analysen machen und sogar visualisieren.

Momentan hab ich keine bessere Idee- leider ist Deine httpd Nutzung der absolute Randfall. Sonst schau Dir doch auch gerne mal das hier an: https://github.com/volkszaehler/volkszaehler.org/pull/430 - damit bekommst Du die Nachrichten sogar gepusht statt per poll!

andig, 2016-06-25:
VZ meeting: works as implemented. Aggregation currently shifts timestamp by one. Note: differes with middleware behaviour. Reason clear, fix not a priority.

closed 2016-06-25 by andig