volkszaehler/vzlogger #249: "aggtime" for workinghoursmeter seems to produce wrong results

justinotherguy, 2016-04-19:
Servus,

I have been running the whm for a while and it works flawlessly. Trying to reduce the amount of data it produces, I tried to aggregate the values over a longer time period - at present I have set "aggtime" to 10, so I've tried changing it to 20.

While aggtime of 10 gives me readings in the range of 0-1 (which means 0 - 100% run time), an aggtime of 20 gives me values in the range of 0-2.

My assumptions might me wrong - especially for aggmode (I've set it to "sum" for the time being).

Changing aggmode to "max" gives me readings of .1, "avg" even less.

Any ideas on what to look at next?

I'll dig deeper into this within the next few days(TM) and set up a test whm.

Regards, J.

Here's my vzlogger.conf:

{
  "retry": 0,
  "daemon": true,
  "verbosity": 5,
  "log": "/var/log/vzlogger.log",
  "meters": [
    {
      "enabled": true,
      "interval": 1,
      "aggtime": 10, // aggregate all signals and give one update to middleware every <aggtime> seconds
      "aggfixedinterval": false,
      "send_zero": true, //bei Betriebsstundenzähler angebracht, damit Pausen auch als Pausen dargestellt werden
      "protocol": "s0",
      "mmap": "",
      "gpio_dir": -1,
      "configureGPIO": true,
      "resolution": 1, 
      "debounce_delay": 1,
      "nonblocking_delay": 100000,
      "channels": [
        {
          "uuid": <uuid>,
          "identifier": "Impulse", 
          "api": "volkszaehler",
          "middleware": "http://demo.volkszaehler.org/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1, //scaling factor to use. Only needed for MySmartGrid api
          "aggmode": "sum", //Mittelwert fuer Leistung, MAX fuer Zaehler, SUM fuer Counter
          "duplicates": 0
       }
      ]
    }
  ]
}

andig, 2016-04-19:
> I have been running the whm for a while and it works flawlessly.

VZ currently does not have a workinghoursmeter (https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L234). What implementation of EntityDefinition.json are you referring to?

mbehr1, 2016-04-19:
Hi,

sorry, I don’t get the problem.
Basically aggmode sum just sums up the single values that would usually get returned within that time.

Btw: I wouldn’t use „interval 1“ with s0. During that time (1s) the data does not get counted!
With send_zero = true you should get one data value (the amount of impulses per second).
With aggmode sum/aggtime 10 you should get the amount of impulses per 10s.

(How do you use s0 with „workinghoursmeter“? What are you actually counting? the amount of seconds the data is „on“? That’s not easy with „impulses“ vs. „high level“?)

> 
> Servus,
> 
> I have been running the whm for a while and it works flawlessly. Trying to reduce the amount of data it produces, I tried to aggregate the values over a longer time period - at present I have set "aggtime" to 10, so I've tried changing it to 20.
> 
> While aggtime of 10 gives me readings in the range of 0-1 (which means 0 - 100% run time), an aggtime of 20 gives me values in the range of 0-2.
> 
> My assumptions might me wrong - especially for aggmode (I've set it to "sum" for the time being).
> 
> Changing aggmode to "max" gives me readings of .1, "avg" even less.
> 
> Any ideas on what to look at next?
> 
> I'll dig deeper into this within the next few days(TM) and set up a test whm.
> 
> Regards, J.
> 
> Here's my vzlogger.conf:
> 
> {
> "retry": 0,
> "daemon": true,
> "verbosity": 5,
> "log": "/var/log/vzlogger.log",
> "meters": [
> {
> "enabled": true,
> "interval": 1,
> "aggtime": 10, // aggregate all signals and give one update to middleware every seconds
> "aggfixedinterval": false,
> "send_zero": true, //bei Betriebsstundenzähler angebracht, damit Pausen auch als Pausen dargestellt werden
> "protocol": "s0",
> "mmap": "",
> "gpio_dir": -1,
> "configureGPIO": true,
> "resolution": 1, 
> "debounce_delay": 1,
> "nonblocking_delay": 100000,
> "channels": [
> {
> "uuid": ,
> "identifier": "Impulse", 
> "api": "volkszaehler",
> "middleware": "http://demo.volkszaehler.org/middleware.php <http://demo.volkszaehler.org/middleware.php>",
> "secretKey": "",
> "type": "device",
> "scaler": 1, //scaling factor to use. Only needed for MySmartGrid api
> "aggmode": "sum", //Mittelwert fuer Leistung, MAX fuer Zaehler, SUM fuer Counter
> "duplicates": 0
> }
> ]
> }
> ]
> }

Gruß

Matthias

justinotherguy, 2016-04-19:
> > I have been running the whm for a while and it works flawlessly.
>
> 
VZ currently does not have a workinghoursmeter (https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L234). What implementation of EntityDefinition.json are you referring to?

yes, you are right (of course!) I'm referring to workinghourssensor:
https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L249

justinotherguy, 2016-04-19:
> Basically aggmode sum just sums up the single values that would usually get returned within that time. 

yes, that was, what I had expected. But:
The values that vzlogger logged were between 0-500 (0-1) for aggtime = 10 and bewteen 0-1000 (0-2) for aggtime = 20 with less entries per time. Hm - that sounds plausible. Maybe my expectation is simply wrong...

I would expect the reading for the workinghourssensor (sorry for consusing this...) to remain constant for different values of aggtime.

Now I wonder, if the consumption could constant for different values of aggtime; having higher readings per data point but less data points could turn out in constant consumptions (which is the value I'm really interested in). I guess I'll need to go for a test setup to dig into this...

> Btw: I wouldn’t use „interval 1“ with s0. During that time (1s) the data does not get counted! 

hm - the input frequency is 50 Hz; I can see in vzlogger.log that the readings reflect the number of counted 50 Hz pulses; if your statement is correct, that should not work, right?

> With send_zero = true you should get one data value (the amount of impulses per second). With aggmode sum/aggtime 10 you should get the amount of impulses per 10s. 

see above - vzlogger does work that way. Should I close this one and open an issue for the middleware?
How are these two components (s0 in vzlogger and workinghourssensor in MW) supposed to work together?

> (How do you use s0 with „workinghoursmeter“? 

see above - workinghourssensor is what I use in the middleware.

> What are you actually counting? 

50 Hz pulses whenever an engine is turned on

> the amount of seconds the data is „on“? 

yes and no - indirectly; the number of pulses in those intervals

Understand?

Regards, J.

andig, 2016-04-20:
> yes, you are right (of course!) I'm referring to workinghourssensor:
https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L249

Thats a "SensorInterpreter" e.g. for temperature. Meaning it records the current value of a phyisical entity (e.g. temp, power). To use this as a Working Hours Meter basically means integrating this value (0=off, 1=on) over time. Unitless integrated over time gives hours.

>> Basically aggmode sum just sums up the single values that would usually get returned within that time.
> yes, that was, what I had expected. But:

SensorInterpreter has aggmode avg in the middleware. Different values in vzlogger don't make sense. We could (idea) check if theres a conflict between vzlogger's aggmode and the middleware channel configuration.

> The values that vzlogger logged were between 0-500 (0-1) for aggtime = 10 and beteen 0-1000 (0-2) for aggtime = 20 with less entries per time. Hm - that sounds plausible. Maybe my expectation is simply wrong...

If you log any number of 1nes over an over the total status is still 1- meaning the device is active. Right? if you have 50% 0 and 50% 1 then it was only working 30minutes. So sum is wrong.

> I would expect the reading for the workinghourssensor (sorry for consusing this...) to remain constant for different values of aggtime.
> Now I wonder, if the consumption could constant for different [...]

That's why you need to use avg.

> hm - the input frequency is 50 Hz; I can see in vzlogger.log that the readings reflect the number of counted 50 Hz pulses; if your statement is correct, that should not work, right?

We still have an open and potentially ugly bug with aggmode thats waiting for feedback on the ML: https://github.com/volkszaehler/vzlogger/issues/231

> see above - vzlogger does work that way. Should I close this one and open an issue for the middleware?

For what- maybe move an open discussion to gitter or ml?

andig, 2016-04-20:
Btw, using a "MeterInterpreter" (https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L234) for working hours is totally different thing. Meters count "units of consumption", that is 1 impulse designates x seconds of runtime recorded. Then aggmode sum is of course correct.

Maybe you just need a different sensor type? Or other way round: how does your physical sensor work that you'd like to implement in vz?

mbehr1, 2016-04-20:
Hi,

> Basically aggmode sum just sums up the single values that would usually get returned within that time.
>
> yes, that was, what I had expected. But:
> The values that vzlogger logged were between 0-500 [...]

No. 1s will be correctly measured. Then you will be blind for 1s, then it will be measured correctly,...
> With send_zero = true you should get one data value (the amount of impulses per second). With aggmode sum/aggtime 10 you should get the amount of impulses per 10s.
>
> see above - vzlogger does work [...]
Ah. Funny idea! Yes, should work.
> the amount of seconds the data is „on“?
> 
> yes and no - indirectly; the number of pulses in those intervals
> 
> Understand?
> 
> Regards, J.

Gruß

Matthias

mbehr1, 2016-04-20:
> yes, you are right (of course!) I'm referring to workinghourssensor:
> https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Definition/EntityDefinition.json#L249 [...]

You could use both:
sum: over 10s -> value between 0 and 500 to be treated as 0-100%

avg: over 10s (or wouldn’t matter) -> value between 0-50 to be treated as 0-10%
> I would expect the reading for the workinghourssensor (sorry for consusing this...) to remain constant for different values of aggtime.
> Now I wonder, if the consumption could constant for different [...]
that shouldn’t matter. you values might be off by 1 reading. can be ignored for your case.
> see above - vzlogger does work that way. Should I close this one and open an issue for the middleware?
> 
> For what- maybe move an open discussion to gitter or ml?

Gruß

Matthias

andig, 2016-05-04:
@justinotherguy whats missing now? Can we close this one?

justinotherguy, 2016-05-04:
yes, we can. I'll continue on the ML

closed 2016-05-04 by justinotherguy