volkszaehler/vzlogger #253: Throw an error or warning, if protocol "s0" is used in combination with parameter "interval"

justinotherguy, 2016-05-05:
Apparantly protocol "s0" should not be used in combination with parameter "interval", see thread on ML:
https://demo.volkszaehler.org/pipermail/volkszaehler-dev/2016-May/004974.html

I see two possible solutions to take care of this while parsing vzlogger.conf:
1) throw an error and terminate vzlogger
2) display (and log?) a warning and continue vzlogger

To be honest, I would prefer option 1.

And: it would be brillant if our editor (https://volkszaehler.github.io/vzlogger/) would also avoid this combination 

Regards, J.

andig, 2016-05-06:
I'd prefer 2- let people who know what they're doing some freedom?

Viele Grüße,
Andreas

justinotherguy, 2016-05-07:
I'm happy with both.
BTW: a first step might be to check this during validation in the editor, right?
That would help already, I think.

andig, 2016-06-25:
VZ meeting: add warning and ignore setting.

mbehr1, 2016-06-26:
See fix with #265.

andig, 2016-06-26:
@mbehr1 ich hätte noch einen Zusatzvorschlag: wie wäre es eine Warnung im Logfile auszugeben (sonst nix) falls es sich um S0 handelt, aggtime gesetzt ist und der aggmode != sum (nur Warnung)?

aggtime ohne aggmode könnte zusätzlich noch bei jedem Meter eine Warnung wert sein. Ich mach aber auch gerne ein neues Issue auf.

mbehr1, 2016-06-26:
gern per sep. Issue.

> @mbehr1 <https://github.com/mbehr1> ich hätte noch einen Zusatzvorschlag: wie wäre es eine Warnung im Logfile auszugeben (sonst nix) falls es sich um S0 handelt, aggtime gesetzt ist und der aggmode [...]

Gruß

Matthias

closed 2016-06-27 by andig